astsam/rtl8812au #1: error build

cristi28, 2017-01-21:
`make ARCH=i386 CROSS_COMPILE= -C /lib/modules/4.8.0-kali2-686-pae/build M=/usr/src/rtl8812au-master  modules
make[1]: se entra en el directorio '/usr/src/linux-headers-4.8.0-kali2-686-pae'
  CC [M]  /usr/src/rtl8812au-master/core/rtw_cmd.o
  CC [M]  /usr/src/rtl8812au-master/core/rtw_security.o
  CC [M]  /usr/src/rtl8812au-master/core/rtw_debug.o
  CC [M]  /usr/src/rtl8812au-master/core/rtw_io.o
  CC [M]  /usr/src/rtl8812au-master/core/rtw_ioctl_query.o
  CC [M]  /usr/src/rtl8812au-master/core/rtw_ioctl_set.o
  CC [M]  /usr/src/rtl8812au-master/core/rtw_ieee80211.o
  CC [M]  /usr/src/rtl8812au-master/core/rtw_mlme.o
  CC [M]  /usr/src/rtl8812au-master/core/rtw_mlme_ext.o
  CC [M]  /usr/src/rtl8812au-master/core/rtw_wlan_util.o
  CC [M]  /usr/src/rtl8812au-master/core/rtw_vht.o
  CC [M]  /usr/src/rtl8812au-master/core/rtw_pwrctrl.o
  CC [M]  /usr/src/rtl8812au-master/core/rtw_rf.o
  CC [M]  /usr/src/rtl8812au-master/core/rtw_recv.o
  CC [M]  /usr/src/rtl8812au-master/core/rtw_sta_mgt.o
  CC [M]  /usr/src/rtl8812au-master/core/rtw_ap.o
  CC [M]  /usr/src/rtl8812au-master/core/rtw_xmit.o
  CC [M]  /usr/src/rtl8812au-master/core/rtw_p2p.o
  CC [M]  /usr/src/rtl8812au-master/core/rtw_tdls.o
  CC [M]  /usr/src/rtl8812au-master/core/rtw_br_ext.o
  CC [M]  /usr/src/rtl8812au-master/core/rtw_iol.o
  CC [M]  /usr/src/rtl8812au-master/core/rtw_sreset.o
  CC [M]  /usr/src/rtl8812au-master/core/rtw_btcoex.o
  CC [M]  /usr/src/rtl8812au-master/core/rtw_beamforming.o
  CC [M]  /usr/src/rtl8812au-master/core/rtw_odm.o
  CC [M]  /usr/src/rtl8812au-master/core/efuse/rtw_efuse.o
  CC [M]  /usr/src/rtl8812au-master/os_dep/osdep_service.o
  CC [M]  /usr/src/rtl8812au-master/os_dep/linux/os_intfs.o
  CC [M]  /usr/src/rtl8812au-master/os_dep/linux/usb_intf.o
  CC [M]  /usr/src/rtl8812au-master/os_dep/linux/usb_ops_linux.o
  CC [M]  /usr/src/rtl8812au-master/os_dep/linux/ioctl_linux.o
  CC [M]  /usr/src/rtl8812au-master/os_dep/linux/xmit_linux.o
  CC [M]  /usr/src/rtl8812au-master/os_dep/linux/mlme_linux.o
  CC [M]  /usr/src/rtl8812au-master/os_dep/linux/recv_linux.o
  CC [M]  /usr/src/rtl8812au-master/os_dep/linux/ioctl_cfg80211.o
/usr/src/rtl8812au-master/os_dep/linux/ioctl_cfg80211.c:92:12: error: ‘IEEE80211_BAND_2GHZ’ undeclared here (not in a function)
  .band   = IEEE80211_BAND_2GHZ,  \
            ^
/usr/src/rtl8812au-master/os_dep/linux/ioctl_cfg80211.c:150:2: note: in expansion of macro ‘CHAN2G’
  CHAN2G(1, 2412, 0),
  ^
/usr/src/rtl8812au-master/os_dep/linux/ioctl_cfg80211.c:101:12: error: ‘IEEE80211_BAND_5GHZ’ undeclared here (not in a function)
  .band   = IEEE80211_BAND_5GHZ,  \
            ^
/usr/src/rtl8812au-master/os_dep/linux/ioctl_cfg80211.c:167:2: note: in expansion of macro ‘CHAN5G’
  CHAN5G(34, 0),  CHAN5G(36, 0),
  ^
/usr/src/rtl8812au-master/os_dep/linux/ioctl_cfg80211.c:101:12: error: initializer element is not constant
  .band   = IEEE80211_BAND_5GHZ,  \
            ^
/usr/src/rtl8812au-master/os_dep/linux/ioctl_cfg80211.c:167:2: note: in expansion of macro ‘CHAN5G’
  CHAN5G(34, 0),  CHAN5G(36, 0),
  ^
/usr/src/rtl8812au-master/os_dep/linux/ioctl_cfg80211.c:101:12: note: (near initialization for ‘rtw_5ghz_a_channels[0].band’)
  .band   = IEEE80211_BAND_5GHZ,  \
            ^
/usr/src/rtl8812au-master/os_dep/linux/ioctl_cfg80211.c:167:2: note: in expansion of macro ‘CHAN5G’
  CHAN5G(34, 0),  CHAN5G(36, 0),
  ^
/usr/src/rtl8812au-master/os_dep/linux/ioctl_cfg80211.c:101:12: error: initializer element is not constant
  .band   = IEEE80211_BAND_5GHZ,  \
            ^
/usr/src/rtl8812au-master/os_dep/linux/ioctl_cfg80211.c:167:18: note: in expansion of macro ‘CHAN5G’
  CHAN5G(34, 0),  CHAN5G(36, 0),
                  ^
/usr/src/rtl8812au-master/os_dep/linux/ioctl_cfg80211.c:101:12: note: (near initialization for ‘rtw_5ghz_a_channels[1].band’)
  .band   = IEEE80211_BAND_5GHZ,  \
            ^
[... 537 lines cut ...]
/usr/src/rtl8812au-master/os_dep/linux/ioctl_cfg80211.c:6610:33: error: array subscript is not an integer
   rtw_spt_band_free(wiphy->bands[IEEE80211_BAND_5GHZ]);
                                 ^
/usr/src/rtl8812au-master/os_dep/linux/ioctl_cfg80211.c:6610:21: error: passing argument 1 of ‘rtw_spt_band_free’ from incompatible pointer type [-Werror=incompatible-pointer-types]
   rtw_spt_band_free(wiphy->bands[IEEE80211_BAND_5GHZ]);
                     ^
/usr/src/rtl8812au-master/os_dep/linux/ioctl_cfg80211.c:272:6: note: expected ‘struct ieee80211_supported_band *’ but argument is of type ‘struct ieee80211_channel *’
 void rtw_spt_band_free(struct ieee80211_supported_band *spt_band)
      ^
/usr/src/rtl8812au-master/os_dep/linux/ioctl_cfg80211.c:6611:15: error: array subscript is not an integer
   wiphy->bands[IEEE80211_BAND_5GHZ] = NULL;
               ^
cc1: some warnings being treated as errors
/usr/src/linux-headers-4.8.0-kali2-common/scripts/Makefile.build:294: fallo en las instrucciones para el objetivo '/usr/src/rtl8812au-master/os_dep/linux/ioctl_cfg80211.o'
make[4]: *** [/usr/src/rtl8812au-master/os_dep/linux/ioctl_cfg80211.o] Error 1
/usr/src/linux-headers-4.8.0-kali2-common/Makefile:1490: fallo en las instrucciones para el objetivo '_module_/usr/src/rtl8812au-master'
make[3]: *** [_module_/usr/src/rtl8812au-master] Error 2
Makefile:150: fallo en las instrucciones para el objetivo 'sub-make'
make[2]: *** [sub-make] Error 2
Makefile:8: fallo en las instrucciones para el objetivo 'all'
make[1]: *** [all] Error 2
make[1]: se sale del directorio '/usr/src/linux-headers-4.8.0-kali2-686-pae'
Makefile:1259: fallo en las instrucciones para el objetivo 'modules'
make: *** [modules] Error 2
`

kcdtv, 2017-01-21:
Hi!!
Very nice project
Unfortunately I have the same problem here with Ubuntu 16.10 
`Linux ubuntumuxo 4.8.0-32-generic #34-Ubuntu SMP Tue Dec 13 14:30:43 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux`
I don't kinow C so i don't know until which point this implicit declaration is the problem. 
This `_module_` directory looks like a little issue in some variable declaration. I couldn't get to the point parsing the makefile. 
I had a similar issue with kali rolling amd64 (same OS than cristi_28)
Thanks for your efforts, ask for any piece any information you need, i will be glad to provide it.

astsam, 2017-01-21:
Currently try to build it under Ubuntu 16.04.

kcdtv, 2017-01-21:
i didn't try with the older release...  Good luck!

astsam, 2017-01-21:
Will you try with the latest commit bd960d1.

kcdtv, 2017-01-21:
Great Job! :smiley_cat: 
It worked now in Ubuntu 16.10
They are some warnings", always for some `-Wmisleading-indentation` related to asome if condition.
But the driver compiled and installation worked! 
[image]
This are the little warnings that i got while compiling:

>  CC [M]  /home/kcdtv/rtl8812au/core/rtw_mlme_ext.o
/home/kcdtv/rtl8812au/core/rtw_mlme_ext.c: In function ‘rtw_delba_check’:
/home/kcdtv/rtl8812au/core/rtw_mlme_ext.c:12242:7: warning: this ‘else’ clause does not guard... [-Wmisleading-indentation]
       else
       ^~~~
/home/kcdtv/rtl8812au/core/rtw_mlme_ext.c:12244:8: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘else’
        psta->recvreorder_ctrl[i].enable = _FALSE;
        ^~~~
/home/kcdtv/rtl8812au/core/rtw_mlme_ext.c:12245:7: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
       if (ret != _FAIL)
       ^~
/home/kcdtv/rtl8812au/core/rtw_mlme_ext.c:12247:8: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘if’
        rtw_reset_continual_no_rx_packet(psta, i); 

> CC

 [M]  /home/kcdtv/rtl8812au/hal/rtl8812a/usb/rtl8812au_xmit.o
/home/kcdtv/rtl8812au/hal/rtl8812a/usb/rtl8812au_xmit.c: In function ‘rtl8812au_xmitframe_complete’:
/home/kcdtv/rtl8812au/hal/rtl8812a/usb/rtl8812au_xmit.c:618:2: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
  if (pbuf < bulkPtr)
  ^~
/home/kcdtv/rtl8812au/hal/rtl8812a/usb/rtl8812au_xmit.c:620:3: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘if’
   if (descCount == pHalData->UsbTxAggDescNum)
   ^~

I don't know if you want to debug that too.. some kind of syntax that it doesn't like...
thanks !

kcdtv, 2017-01-21:
Even better in Kali Linux!  :smiley_cat:
It compiled without a single warning, everything went perfectly well!  :smiley_cat:
sensibility is tremendous....
 
[image]

the issue can be closed as the problem is solved.

kcdtv, 2017-01-21:
very good injection : 
[image]

astsam, 2017-01-22:
Thank you for testing!

closed 2017-01-22 by astsam