astsam/rtl8812au #2: add Alfa Netwrok AWUS036ACH to the devices list

kcdtv, 2017-01-21:
for the file **usb_intf.c** a line like this could be added:
`{USB_DEVICE(0x0bda, 0x8812),.driver_info = RTL8812}, /* Alfa Network - AWUS036ACH */`
Here is the stdout of my `cat  /sys/kernel/debug/usb/devices` for you to check:

> T:  Bus=01 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#= 12 Spd=480  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=0bda ProdID=8812 Rev= 0.00
S:  Manufacturer=Realtek
S:  Product=802.11n NIC
S:  SerialNumber=123456
C:* #Ifs= 1 Cfg#= 1 Atr=80 MxPwr=500mA
I:* If#= 0 Alt= 0 #EPs= 5 Cls=ff(vend.) Sub=ff Prot=ff Driver=rtl8821au
E:  Ad=81(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=02(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=03(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=04(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=85(I) Atr=03(Int.) MxPS=  64 Ivl=125us

Cheers!

astsam, 2017-01-21:
Are you sure this device is really unsupported? It seems it uses Realtek VID/PID.

kcdtv, 2017-01-21:
it is supported! 
Maybe i got it wrong, i think this file was for supported devices.

closed 2017-01-22 by astsam