astsam/rtl8812au #3: txpower blocked at 12dBm

kcdtv, 2017-01-22:
Hi again!
I have a litlle issue with the Alfa AWUS036ACH which is realy reducing the performances of the device
Txpower is blocked at 12dbm and there is no way to rise this txpower.
As you can see on the folowing picture my AWUS036ACH (wlan2) stay at 12dBm while i can rise the power to 30dBm to my AWUS036H. 
It is even under my laptop card (16dBm) 

[image]

strange think is that the device shows 30dBm in both bands (it is erfectivly a high power device) as you can see from `ìw list`

> 			* 2412 MHz [1] (30.0 dBm)
			* 2417 MHz [2] (30.0 dBm)
			* 2422 MHz [3] (30.0 dBm)
			* 2427 MHz [4] (30.0 dBm)
			* 2432 MHz [5] (30.0 dBm)
			* 2437 MHz [6] (30.0 dBm)
			* 2442 MHz [7] (30.0 dBm)
			* 2447 MHz [8] (30.0 dBm)
			* 2452 MHz [9] (30.0 dBm)
			* 2457 MHz [10] (30.0 dBm)
			* 2462 MHz [11] (30.0 dBm)
			* 2467 MHz [12] (30.0 dBm)
			* 2472 MHz [13] (30.0 dBm)
			* 2484 MHz [14] (30.0 dBm)
	Band 2:
		Capabilities: 0x1862
			HT20/HT40
			Static SM Power Save
			RX HT20 SGI
			RX HT40 SGI
			No RX STBC
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT Max RX data rate: 300 Mbps
		HT TX/RX MCS rate indexes supported: 0-15
		VHT Capabilities (0x00000010):
			Max MPDU length: 3895
			Supported Channel Width: neither 160 nor 80+80
			RX LDPC
		VHT RX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: MCS 0-9
			4 streams: MCS 0-9
			5 streams: MCS 0-9
			6 streams: MCS 0-9
			7 streams: MCS 0-9
			8 streams: MCS 0-9
		VHT RX highest supported: 0 Mbps
		VHT TX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: MCS 0-9
			4 streams: MCS 0-9
			5 streams: MCS 0-9
			6 streams: MCS 0-9
			7 streams: MCS 0-9
			8 streams: MCS 0-9
		VHT TX highest supported: 0 Mbps
		Bitrates (non-HT):
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 5170 MHz [34] (30.0 dBm)
			* 5180 MHz [36] (30.0 dBm)
			* 5190 MHz [38] (30.0 dBm)
			* 5200 MHz [40] (30.0 dBm)
			* 5210 MHz [42] (30.0 dBm)
			* 5220 MHz [44] (30.0 dBm)
			* 5230 MHz [46] (30.0 dBm)
			* 5240 MHz [48] (30.0 dBm)
			* 5260 MHz [52] (30.0 dBm)
			* 5280 MHz [56] (30.0 dBm)
			* 5300 MHz [60] (30.0 dBm)
			* 5320 MHz [64] (30.0 dBm)
			* 5500 MHz [100] (30.0 dBm)
			* 5520 MHz [104] (30.0 dBm)
			* 5540 MHz [108] (30.0 dBm)
			* 5560 MHz [112] (30.0 dBm)
			* 5580 MHz [116] (30.0 dBm)
			* 5600 MHz [120] (30.0 dBm)
			* 5620 MHz [124] (30.0 dBm)
			* 5640 MHz [128] (30.0 dBm)
			* 5660 MHz [132] (30.0 dBm)
			* 5680 MHz [136] (30.0 dBm)
			* 5700 MHz [140] (30.0 dBm)
			* 5745 MHz [149] (30.0 dBm)
			* 5755 MHz [151] (30.0 dBm)
			* 5765 MHz [153] (30.0 dBm)
			* 5775 MHz [155] (30.0 dBm)
			* 5785 MHz [157] (30.0 dBm)
			* 5805 MHz [161] (30.0 dBm)
			* 5825 MHz [165] (30.0 dBm)
			* 5920 MHz [184] (30.0 dBm)
			* 5940 MHz [188] (30.0 dBm)
			* 5960 MHz [192] (30.0 dBm)
			* 5980 MHz [196] (30.0 dBm)
			* 6000 MHz [200] (30.0 dBm)
			* 6020 MHz [204] (30.0 dBm)
			* 6040 MHz [208] (30.0 dBm)
			* 6060 MHz [212] (30.0 dBm)
			* 6080 MHz [216] (30.0 dBm)

So it is not a problem with CRDA as my other Alfa is at 30dBm. 
What could it be? What could I try? 
Thanks again for your attention

astsam, 2017-01-24:
TX power control is not implemented. Driver always returns a hardcoded value (12 dBm).

mehmetgoksu, 2017-01-25:
@kcdtv why do you need to increase tx power ? is tx power normally not enough ?

cristi28, 2017-01-25:
Txpower is blocked at 12dbm(16 mW) @mehmetgoksu awus036ach have 1600mW(32dbm)

mehmetgoksu, 2017-01-25:
@cristi28 is it blocked after driver ? or it always 12 dbm ?

kcdtv, 2017-01-25:
Sorry guys, i couldn't answer before

> TX power control is not implemented. Driver always returns a hardcoded value (12 dBm).

Is there any plan to implement it? :smiley-cat:
  I am afraid this hardcoded value is the one in use... It is what i would say from my first impressions, although it is always hard to say
  PING fluctuations with lower performance than usual. 
 

> @kcdtv why do you need to increase tx power ? is tx power normally not enough ?

  I don't want to use it at full power all time. 
But yes, i need sometimes to slightly increase txpower for better performances.  
It is useful for the creation of a soft AP and to reach better AP with low sensibility (like the "livebox" or other "boxes" kind of AP that doesn't have external antenna and have poor wifi characteristic)

astsam, 2017-01-25:
> Is there any plan to implement it? :smiley_cat: cat:

This issue has a quite low priority. You may help if you check the power control in https://github.com/ulli-kroll/rtl8821au project and/or https://github.com/gnab/rtl8821au.

> I am afraid this hardcoded value is the one in use...

Driver just returns 12. This value has no relations with driver internals, phy state etc.

mehmetgoksu, 2017-01-25:
ı dont understand the value of 12 is stock or our driver affect it ?

cristi28, 2017-01-25:
take a look here maybe helps https://github.com/mgillespie/rtl8812au/blob/master/hal/OUTSRC/HalPhyRf.c

kcdtv, 2017-01-25:
@ cristie: astsam told us that the value is "not real", it is just an hardcoded value that doesn't reflect the reality.  
@ astan 
 1) I might have spoken too fast and the quality issue that i experimented may have been the results of temporary interferences. 
 I live in a very crowded area with a lot of routers and clients, at night when everybody sleeps things were going better, 
 I will test again late tonight
 2) I will see if i can get the txpower control playing with different branches
I will report here the results.
 Thanks again for your attention and for your efforts.

mehmetgoksu, 2017-01-25:
@kcdtv waiting yoyr resultss

kcdtv, 2017-01-26:
cristi_28 showed me this branch that seems to have power control:
[https://github.com/mgillespie/rtl8812au](url)
Unfortunately I was not able to install it my in my OS 
I add several criticals warnings-errors when i compiled it
I tried in Kali roling (debian sid derivate)  and Ubuntu 16.10
Similar to this issue  : #1 (but with much more warnings)

If we have a look closer to the branch we can see that it has two more folders than this one
  1  `rtl8812au/hal/OUTSRC-BTCoexist/`
  2  `rtl8812au/hal/OUTSRC/`
Cristi_28 found this file that is clearly mentioning taxpower control: [https://github.com/mgillespie/rtl8812au/blob/master/hal/OUTSRC/HalPhyRf.c](url)
We can see function such as _ConfigureTxpowerTrack_ etc.

That all i could try yesterday... I will find sometime tonight to do more search.
cheers

mehmetgoksu, 2017-01-26:
Did you test injection and aircrack module ? Are they working well without any bug?

kcdtv, 2017-01-26:
This questions are off topic, aren't they? 
Let's focus here on the txpower issue. :wink:

FightingForFun, 2017-01-27:
Good day :)
Today try Gnab and Grawp drivers. Tx-power not show and not change. Why ? Hmm...))

Astsam's drivers in managed mode work great.
But in monitor...

When select any channel in airodump-ng (-c)... I catch all channels.
And i dont know how write this (my bad language)... but airodump update update update (catch) info bssids, beacons, data and etc and after stop on some miliseconds (or one second). after update update update. stop...

```
 CH  3 ][ Elapsed: 2 mins ][ 2017-01-28 00:05 ][ fixed channel wlan0: -1                                         
 BSSID              PWR RXQ  Beacons    #Data, #/s  CH  MB   ENC  CIPHER AUTH ESSID
                                                                                                                                           
 CC:5D:4E:00:00:00  -39   0       45        1    0   6  54e  WPA2 CCMP   PSK  MYNET                                                          
 74:D0:2B:00:00:00  -53   0       71      155    0  11  54e  WPA2 CCMP   PSK  LITE3                                                        
 00:90:4C:00:00:00  -56   0       90        1    0   2  54e. WPA2 CCMP   PSK  Wi-Fi Access Point  
```

Wash from Reaver packet write: 

`[!] Found packet with bad FCS, skipping... `

When use (-C) for skip bad fcs, RSSI not recognized

```
BSSID                  Channel       RSSI       WPS Version       WPS Locked        ESSID
---------------------------------------------------------------------------------------------------------------
C0:4A:00:00:00:00       1            00        1.0               No                KA110000

```

Reaver my and any stations with good signal not hack. 

```
[+] Switching wlan0 to channel 1
[+] Waiting for beacon from C8:6C:87:00:00:00
[+] Associated with C8:6C:87:00:00:00 (ESSID: 1000)
[+] Starting Cracking Session. Pin count: 0, Max pin attempts: 11000
[+] Trying pin 12345670.
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[!] WARNING: Receive timeout occurred
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x02), re-trying last pin
```

Bully not try to brute wps...

```
[!] Bully v1.0-22 - WPS vulnerability assessment utility
[+] Switching interface 'wlan0' to channel '1'
[!] Using 'e0:3f:49:00:00:00' for the source MAC address
[+] Datalink type set to '127', radiotap headers present
[+] Scanning for beacon from 'c8:6c:87:00:00:00' on channel '1'
[!] Excessive (3) FCS failures while reading next packet
[X] Unable to get a beacon from the AP, possible causes are
[.]    an invalid --bssid or -essid was provided,
[.]    the access point isn't on channel '1',
[.]    you aren't close enough to the access point.
```

kcdtv, 2017-01-28:
Good day too
This is not the place to analyses the behavior with reaver and others hacking tools.
This issue is about txpower management, please stay focus  :smiley_cat:
 

> Today try Gnab and Grawp drivers. Tx-power not show and not change. Why ? Hmm...))

 They don't have power management neither.  
Could you tell us which OS did you use?
Could you try to install this drivers: [(I]( https://github.com/mgillespie/) 
I cannot install them in mine, it fails compiling. They have power management (at least they have something to do it)  
Cheers

astsam, 2017-01-28:
Unfortunately this driver (https://github.com/mgillespie/rtl8812au) has no power control too.

kcdtv, 2017-01-30:
Damned it! :smiley_cat:
Thanks for trying and keeping us informed... 
I will keep searching and trying.
cya

galexant, 2017-02-07:
TX power should be setting and getting via ioctl: _fg80211_rtw_set_txpower_  and _cfg80211_rtw_get_txpower_ . In every driver i found in _cfg80211_rtw_get_txpower_ next line `*dbm = (12);`

astsam, 2017-02-10:
Preliminary power control was implemented. Pull v4.3.21 branch, set tx power, and switch channel. Power setting will be applied on channel switch only.

kcdtv, 2017-02-10:
Hello! Just on time for the good news! Thanks maestro!

mehmetgoksu, 2017-02-13:
`Wiphy phy0
	max # scan SSIDs: 9
	max scan IEs length: 2304 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
	Available Antennas: TX 0 RX 0
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * monitor
		 * P2P-client
		 * P2P-GO
	Band 1:
		Capabilities: 0x19e3
			RX LDPC
			HT20/HT40
			Static SM Power Save
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT Max RX data rate: 300 Mbps
		HT TX/RX MCS rate indexes supported: 0-15
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps
			* 5.5 Mbps
			* 11.0 Mbps
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 2412 MHz [1] (20.0 dBm)
			* 2417 MHz [2] (20.0 dBm)
			* 2422 MHz [3] (20.0 dBm)
			* 2427 MHz [4] (20.0 dBm)
			* 2432 MHz [5] (20.0 dBm)
			* 2437 MHz [6] (20.0 dBm)
			* 2442 MHz [7] (20.0 dBm)
			* 2447 MHz [8] (20.0 dBm)
			* 2452 MHz [9] (20.0 dBm)
			* 2457 MHz [10] (20.0 dBm)
			* 2462 MHz [11] (20.0 dBm)
			* 2467 MHz [12] (20.0 dBm) (no IR)
			* 2472 MHz [13] (20.0 dBm) (no IR)
			* 2484 MHz [14] (20.0 dBm) (no IR)
	Band 2:
		Capabilities: 0x19e3
			RX LDPC
			HT20/HT40
			Static SM Power Save
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT Max RX data rate: 300 Mbps
		HT TX/RX MCS rate indexes supported: 0-15
		VHT Capabilities (0x00000010):
			Max MPDU length: 3895
			Supported Channel Width: neither 160 nor 80+80
			RX LDPC
		VHT RX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: MCS 0-9
			4 streams: MCS 0-9
			5 streams: MCS 0-9
			6 streams: MCS 0-9
			7 streams: MCS 0-9
			8 streams: MCS 0-9
		VHT RX highest supported: 0 Mbps
		VHT TX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: MCS 0-9
			4 streams: MCS 0-9
			5 streams: MCS 0-9
			6 streams: MCS 0-9
			7 streams: MCS 0-9
			8 streams: MCS 0-9
		VHT TX highest supported: 0 Mbps
		Bitrates (non-HT):
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 5170 MHz [34] (30.0 dBm) (no IR)
			* 5180 MHz [36] (30.0 dBm) (no IR)
			* 5190 MHz [38] (30.0 dBm) (no IR)
			* 5200 MHz [40] (30.0 dBm) (no IR)
			* 5210 MHz [42] (30.0 dBm) (no IR)
			* 5220 MHz [44] (30.0 dBm) (no IR)
			* 5230 MHz [46] (30.0 dBm) (no IR)
			* 5240 MHz [48] (30.0 dBm) (no IR)
			* 5260 MHz [52] (30.0 dBm) (no IR, radar detection)
			* 5280 MHz [56] (30.0 dBm) (no IR, radar detection)
			* 5300 MHz [60] (30.0 dBm) (no IR, radar detection)
			* 5320 MHz [64] (30.0 dBm) (no IR, radar detection)
			* 5500 MHz [100] (30.0 dBm) (no IR, radar detection)
			* 5520 MHz [104] (30.0 dBm) (no IR, radar detection)
			* 5540 MHz [108] (30.0 dBm) (no IR, radar detection)
			* 5560 MHz [112] (30.0 dBm) (no IR, radar detection)
			* 5580 MHz [116] (30.0 dBm) (no IR, radar detection)
			* 5600 MHz [120] (30.0 dBm) (no IR, radar detection)
			* 5620 MHz [124] (30.0 dBm) (no IR, radar detection)
			* 5640 MHz [128] (30.0 dBm) (no IR, radar detection)
			* 5660 MHz [132] (30.0 dBm) (no IR, radar detection)
			* 5680 MHz [136] (30.0 dBm) (no IR, radar detection)
			* 5700 MHz [140] (30.0 dBm) (no IR, radar detection)
			* 5745 MHz [149] (30.0 dBm) (no IR)
			* 5765 MHz [153] (30.0 dBm) (no IR)
			* 5785 MHz [157] (30.0 dBm) (no IR)
			* 5805 MHz [161] (30.0 dBm) (no IR)
			* 5825 MHz [165] (30.0 dBm) (no IR)
			* 5920 MHz [184] (30.0 dBm)
			* 5940 MHz [188] (30.0 dBm)
			* 5960 MHz [192] (30.0 dBm)
			* 5980 MHz [196] (30.0 dBm)
			* 6000 MHz [200] (30.0 dBm)
			* 6020 MHz [204] (30.0 dBm)
			* 6040 MHz [208] (30.0 dBm)
			* 6060 MHz [212] (30.0 dBm)
			* 6080 MHz [216] (30.0 dBm)
	Supported commands:
		 * new_interface
		 * set_interface
		 * new_key
		 * start_ap
		 * new_station
		 * set_bss
		 * join_ibss
		 * set_pmksa
		 * del_pmksa
		 * flush_pmksa
		 * remain_on_channel
		 * frame
		 * set_channel
		 * connect
		 * disconnect
	Supported TX frame types:
		 * IBSS: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP/VLAN: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * IBSS: 0xd0
		 * managed: 0x40 0xd0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
	WoWLAN support:
		 * wake up on anything (device continues operating normally)
	software interface modes (can always be added):
		 * monitor
	interface combinations are not supported
	Device supports scan flush.
`

I dont understand ı pull 4.3.21 and install but my iw list show 20 dbm in band1 ? why ?kcdv has 30 dbm?

cristi28, 2017-02-13:
iw reg set GY
iwconfig wlan0 txpower 30

mehmetgoksu, 2017-02-13:
@Cristi28 now it works normally thanks

astsam, 2017-02-13:
Ignore the iw list output. Just set any tx power level (0..38). The driver does not obey CRDA TX power limits. To function properly with an increased power level dongle should be attached to USB 3.0 port or externally powered USB hub.  

> Power setting will be applied on channel switch only.

Fixed.

closed 2017-02-28 by kcdtv

hrach9779, 2017-10-21:
i have the same problem :(( please i can't change the value of tx-power,pls ask me how you did this @kcdtv