astsam/rtl8812au #5: Raspberry Pi not supported?

osenbruggen, 2017-02-05:
Would it be possible to support the ARM RPI platform?

astsam, 2017-02-06:
I see no reason why it is not possible if you have:
- cross toolchains
- kernel sources

```
$make ARCH=arm KSRC=<path to kernel sources> CROSS_COMPILE=<cross compiler prefix>
```

osenbruggen, 2017-02-06:
Yes thank you that is what I ended op doing. This worked:

make ARCH="arm" CROSS_COMPILE=arm-linux-gnueabihf- KSRC=/usr/src/linux-headers-4.4.45+/

boredwookie, 2017-03-24:
I have tried this as well as the suggestion from ledjohnyy in https://github.com/astsam/rtl8812au/issues/10. In both cases I can see the module compiles and I can install it and see '8812au' in the lsmod output.

What I can't figure out is how to get the interface to show up in either ifconfig or iwconfig. My Raspberry Pi 3 does not show any additional interfaces after running modprobe or a reboot.
@osenbruggen Did you have to do anything extra to get your interface to show up in iw/ifconfig?

astsam, 2017-03-24:
Will you attach output of the `lsusb` and `ip link` commands.

boredwookie, 2017-03-24:
```
piclopse@host:~ $ lsusb
Bus 001 Device 005: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 004: ID 0bda:a811 Realtek Semiconductor Corp.
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
piclopse@host:~ $ ip link
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DEFAULT group default qlen 1000
    link/ether b8:27:eb:56:ac:18 brd ff:ff:ff:ff:ff:ff
3: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN mode DORMANT group default qlen 1000
    link/ether b8:27:eb:03:f9:4d brd ff:ff:ff:ff:ff:ff
```

And for good measure:
```
piclopse@host:~ $ lsmod | grep 8812au
8812au               1304521  0
cfg80211              428871  2 brcmfmac,8812au
```

astsam, 2017-03-25:
Plug adapter into USB 3.0 and attach dmesg output.

boredwookie, 2017-03-25:
[124453.777349] usb 1-1.2: USB disconnect, device number 4
[124455.292277] usb 1-1.3: new high-speed USB device number 6 using dwc_otg
[124455.393362] usb 1-1.3: New USB device found, idVendor=0bda, idProduct=a811
[124455.393382] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[124455.393396] usb 1-1.3: Product: 802.11ac WLAN Adapter
[124455.393408] usb 1-1.3: Manufacturer: Realtek
[124455.393420] usb 1-1.3: SerialNumber: 00e04c000001
[124455.612228] usb 1-1.3: reset high-speed USB device number 6 using dwc_otg

astsam, 2017-03-25:
> [124455.612228] usb 1-1.3: reset high-speed USB device number 6 using dwc_otg

The issue has no relations to the driver, something wrong with your USB subsystem.
https://superuser.com/questions/668911/what-are-usb-device-resets
https://www.raspberrypi.org/forums/viewtopic.php?t=47239&p=373974

closed 2017-03-25 by astsam

boredwookie, 2017-03-25:
Thanks, I'll take a look. The other rtl8812au drivers worked fine, which is why I commented here. :)

tangibilizing, 2017-09-12:
I am using raspi zero w, how can cross compile according "make ARCH="arm" CROSS_COMPILE=arm-linux-gnueabihf- KSRC=/usr/src/linux-headers-4.4.45+/"