astsam/rtl8812au #7: Make Error

mehmetgoksu, 2017-02-11:
before ı again install kali latest make and make install were working but today ı again fresh install and it gives me error
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.9.0-kali1-amd64/build M=/root/Desktop/rtl8812au-master  modules
make[1]: *** /lib/modules/4.9.0-kali1-amd64/build: No such file or directory.  Stop.
Makefile:1264: recipe for target 'modules' failed
make: *** [modules] Error 2

mehmetgoksu, 2017-02-11:
sudo apt-get install linux-headers-$(uname -r) build-essential   fixed my problem

closed 2017-02-11 by mehmetgoksu

Gentech57, 2017-02-11:
mehmet bende hata veriyo knk            Paket listeleri okunuyor... Bitti
Bağımlılık ağacı oluşturuluyor       
Durum bilgisi okunuyor... Bitti      
E: linux-headers-4.6.0-kali1-amd64 paketi bulunamadı
E: 'linux-headers-4.6.0-kali1-amd64' ifadesine eşleşen herhangi bir paket bulunamadı
E: 'linux-headers-4.6.0-kali1-amd64' düzenli ifadesini içeren herhangi bir paket bulunamadı

mehmetgoksu, 2017-02-11:
Live mi kullanıyorsun? Sources.list de kaynaklar düzgün gözüküyor mu?

Gentech57, 2017-02-11:
knk sisteme yükledim hard disk de yani canlı canlı

Gentech57, 2017-02-11:
sendekide awus036ac mi ?

mehmetgoksu, 2017-02-11:
Sistemi kurarken network mirror kullanabildin mi

mehmetgoksu, 2017-02-11:
Hayır ach

Gentech57, 2017-02-11:
ag baglantısını diyorsan evet wifi üzerinden baglandım

Gentech57, 2017-02-11:
monitör moduna geçiyormu peki seninki ? airmon-ng de ?

mehmetgoksu, 2017-02-11:
Etc/apt/sources.list de yazanları buraya atar mısın

Gentech57, 2017-02-11:
bash: Etc/apt/sources.list: Böyle bir dosya ya da dizin yok

mehmetgoksu, 2017-02-11:
Nano /etc/apt/sources.list

Gentech57, 2017-02-11:
dur kendim yerel olarak girdim atıyorum # 

# deb cdrom:[Debian GNU/Linux 2016.1 _Kali-rolling_ - Official Snapshot amd64 LIVE/INSTALL Binary 20160830-11:29]/ kali-rolling contrib main non-free

#deb cdrom:[Debian GNU/Linux 2016.1 _Kali-rolling_ - Official Snapshot amd64 LIVE/INSTALL Binary 20160830-11:29]/ kali-rolling contrib main non-free

deb http://http.kali.org/kali kali-rolling main non-free contrib
# deb-src http://http.kali.org/kali kali-rolling main non-free contrib

Gentech57, 2017-02-11:
knk bu driver monitör modunu destekliyormu sende ?

mehmetgoksu, 2017-02-11:
Destekliyor ama airmon ile başlatmayacaksın

Gentech57, 2017-02-11:
bu iwconfig bildigime göre sadece wep şifrelemede işe yarıyor ? wpa da etkili olabilirmi veya başka bir altarnatif ?

mehmetgoksu, 2017-02-11:
Derken?

Gentech57, 2017-02-11:
sen monitör modunu ne için kullanıyorsun ?

mehmetgoksu, 2017-02-11:
Driver i test amaçlı

Gentech57, 2017-02-11:
tmm

Gentech57, 2017-02-11:
sourceslist i attım sana ? şimdi ne yapmam gerekiyor

mehmetgoksu, 2017-02-11:
Sıkıntısız gözüküyor

Gentech57, 2017-02-12:
Mehmet birde sunu denesene https://github.com/Fredy72/rtl8821au. Airmon-ng kullan bir bak oluyormu diye

Bluewild93, 2017-10-09:
Hi mehmetgoksu. Are u in kali linux?
I try to fix the same problem. but nothing happen. I can't do make. :/