astsam/rtl8812au #8: Driver

Gentech57, 2017-02-11:
Awus036ac. Monitor mode don't work. Please fix. Airmon-ng

astsam, 2017-02-11:
Will you describe the issue in details.

Gentech57, 2017-02-11:
phy0 wlan0 rtl8812au Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac WLAN Adapter
cat: /sys/class/ieee80211/phy0/device/net/wlan0mon/type: Datei oder Verzeichnis nicht gefunden

Newly created monitor mode interface wlan0mon is NOT in monitor mode.
Removing non-monitor wlan0mon interface...

WARNING: unable to start monitor mode, please run "airmon-ng check kill"

mehmetgoksu, 2017-02-11:
You cant start monitor mode using airmon . You should do it manually

astsam, 2017-02-11:
The proper way to switch into the monitor is on the project first page. Airmon-ng script is incompatible with the driver. Do not use this script. Turn off NetworkManager and other tools those may control wireless cards.

closed 2017-02-11 by astsam