astsam/rtl8812au #9: installation error

Gentech57, 2017-02-11:
root@kali:~/rtl8812au# make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.6.0-kali1-amd64/build M=/root/rtl8812au  modules
make[1]: *** /lib/modules/4.6.0-kali1-amd64/build: Böyle bir dosya ya da dizin yok. Durdu.
Makefile:1264: recipe for target 'modules' failed
make: *** [modules] Error 2
root@kali:~/rtl8812au#

astsam, 2017-02-11:
1. Install kernel sources (headers).
2. Change to driver directory and run
`$ make KSRC=<path to kernel sources>`

closed 2017-02-11 by astsam

kcdtv, 2017-02-12:
Just to confirm that installation works without any problems in an updated and correctly configured Kali Linux thanks to  this commit: [https://github.com/astsam/rtl8812au/commit/bd960d186665cb5d611100a0152232ee584230d3](url)
[image]
You're system is **not up to odate** (kernel 4.6) 
I suggest you to start by updating your system. 
`sudo apt-get update && sudo apt-get upgrade && sudo apt-get dist-upgrade && sudo apt-get autoremove && sudo reboot`
And you will see that everything goes fine

j1nxd071r, 2017-06-03:
@astsam You saved my brain after 3 days of hardcore nervous breakdown. Thank you :)

ressac, 2017-08-06:
I'm getting same error, running on latest kali live usb

```
root@kali:~/rtl8812au# uname -a
Linux kali 4.9.0-kali3-amd64 #1 SMP Debian 4.9.18-1kali1 (2017-04-04) x86_64 GNU/Linux
root@kali:~/rtl8812au# make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.9.0-kali3-amd64/build M=/root/rtl8812au  modules
make[1]: *** /lib/modules/4.9.0-kali3-amd64/build: No such file or directory.  Stop.
Makefile:1264: recipe for target 'modules' failed
make: *** [modules] Error 2
root@kali:~/rtl8812au# 

```