astsam/rtl8812au #10: Raspberry Pi

capturepi, 2017-02-12:
Is there a walk through for installing on PwnPi 4.1.19-v7+ or the latest Raspbian 4.4.38.
I've installed the kernel headers.
Compiling appears to work however 8812au.ko states that it is in the wrong format when I run insmod.
The driver doesn't appear to load.

My Alfa adapters works on the latest Kali VM so it's not that.

astsam, 2017-02-12:
Run `file 8812au.ko` and attach output.

Gymnae, 2017-03-04:
What would be the make command or makefile adjustment to compile for arm v7? I can't find the architecture toggle for building on an arm system

astsam, 2017-03-06:
`$ make KSRC=<path to kernel sources> CROSS_COMPILE=<appropriate cross compilation prefix> ARCH=arm`

ledjohnny, 2017-03-06:
Here's what I did to get it to compile on my Pi (some steps taken from [here](https://www.max2play.com/en/forums/topic/howto-raspberry-pi-3-realtek-802-11ac-rtl8812au/)):

1. Download rpi-source [1]
`sudo wget https://raw.githubusercontent.com/notro/rpi-source/master/rpi-source -O /usr/bin/rpi-source && sudo chmod +x /usr/bin/rpi-source && /usr/bin/rpi-source -q --tag-update`

2. Install bc (required by rpi-source)
`sudo apt-get install bc`

3. Run rpi-source to get the required kernel headers
`rpi-source`

4. Download the driver
`git clone https://github.com/astsam/rtl8812au.git`

5. Make the driver
`make CROSS_COMPILE=arm-linux-gnueabihf- ARCH=arm`

6. Install the driver
`sudo make install
sudo modprobe 8812au`

astsam, 2017-03-08:
> 4. Download the driver

Better use the branch v4.3.21
`git clone -b v4.3.21 https://github.com/astsam/rtl8812au`

closed 2017-03-18 by astsam

tangibilizing, 2017-08-29:
I am on a raspberry pi zero w ,

Make the driver
make CROSS_COMPILE=arm-linux-gnueabihf- ARCH=arm 

should I change the "arm-linux-gnueabihf- ARCH=arm" to "arm-linux-gnueabiel- ARCH=arm"

cause I am using armel  not the armhf