astsam/rtl8812au #11: Mode monitor and injection not working on added interface

kcdtv, 2017-02-12:
```
Hi!
I was trying to understand why **airmon-ng** form aircrack-ng cannot enable mode monitor properly. 
I use kali linux (debian sid derivate) 
Before the latest releases **airmon-ng** was not able to detect correctly the driver:
Since i tried with your branch for txpower control (thanks again ;) ) airodump-ng can detect properly the driver :)

```
 sudo airmon-ng
PHY	Interface	Driver		Chipset
phy0	wlan0		ath9k		Qualcomm Atheros QCA9565 / AR9565 Wireless Network Adapter (rev 01)
phy1	wlan1		8812au		Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac WLAN Adapter
```
Nevertheless i always get a fail because it is unable to create the new wlanXmon interface
```
sudo airmon-ng check kill
[sudo] password for kcdtv: 

Killing these processes:

  PID Name
  637 wpa_supplicant
  647 dhclient

kcdtv@kalimuX0:~$ sudo airmon-ng start wlan1

PHY	Interface	Driver		Chipset

phy0	wlan0		ath9k		Qualcomm Atheros QCA9565 / AR9565 Wireless Network Adapter (rev 01)
phy1	wlan1		8812au		Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac WLAN Adapter
cat: /sys/class/ieee80211/phy1/device/net/wlan1mon/type: No existe el fichero o el directorio

Newly created monitor mode interface wlan1mon is *NOT* in monitor mode.
Removing non-monitor wlan1mon interface...

WARNING: unable to start monitor mode, please run "airmon-ng check kill"
```
There is indeed no "wlan1mon" interface in the directory ` /sys/class/ieee80211/phy1/device/net/`
So I tried to do it manually (create a new interface wlan1mon in monitor mode) using a syntax that should work with iwconfig
I didn't get any mistake message but the interface does... nothing. 
kcdtv@kalimuX0:/sys/class/ieee80211/phy1/device/net$ sudo iw phy phy1 interface add wlan1mon type monitor
kcdtv@kalimuX0:/sys/class/ieee80211/phy1/device/net$ sudo iwconfig
wlan1     IEEE 802.11  ESSID:off/any  
          Mode:Managed  Access Point: Not-Associated   Tx-Power=18 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          
wlan0     IEEE 802.11  ESSID:off/any  
          Mode:Managed  Access Point: Not-Associated   Tx-Power=16 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          
lo        no wireless extensions.

eth0      no wireless extensions.

wlan1mon  IEEE 802.11  Mode:Monitor  Tx-Power=18 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          
kcdtv@kalimuX0:/sys/class/ieee80211/phy1/device/net$ sudo aireplay-ng -9 wlan1mon
22:43:00  Trying broadcast probe requests...
22:43:02  No Answer...
22:43:02  Found 0 APs

```
And if i launch airodump-ng no PA shows up...
I looked to see if the new interface was created in the folder where airmon-ng looks for it.... 
And there is no folder for the new interface,,,
```
kcdtv@kalimuX0:/sys/class/ieee80211/phy1/device/net$ ls
wlan1
```
  I assume that they might be a little something going wrong with the compatibility with the add interface feature from iwtools.
  I can upload any kind of information that you require if you have time/can/want to check this
 Thanks again for your attention. 
Please ask for any extra information that might be helpful to you, i would be delighted to give it to you :)

ZeroChaos-, 2017-02-13:
In this case there are two problems which are independent.

1.) Monitor mode is enabled using WEXT and not nl80211, which is unexpected in 2017 (or 2010 for that matter)
2.) mac80211 drivers are expected to populate /sys/class/ieee80211/phy1/device/net/wlan1mon/type with mode information.  Not only is it not accurate in this case, it's missing.

Please don't interpret this as an attack, I realize this is vendor code and you are doing your best.  I'm simply pointing out some of the work which was not done yet.

astsam, 2017-02-13:
Do not try to add/delete interfaces. Use iwconfig utility to place interface in the monitor mode as noted in the README.md. The airmon-ng script is incompatible with the driver. This driver supports nl80211 interface for switching channels and proper frame injection. Therefore the driver is compatible with utilities like reaver, airodump etc. Just run
`$ sudo ip link set wlan1 down`
`$ sudo iwconfig wlan1 mode monitor`
`$ sudo ip link set wlan1 up`
`$ sudo aireplay-ng -9 wlan1`

>  Monitor mode is enabled using WEXT and not nl80211, which is unexpected in 2017 (or 2010 for that matter)

You can make the world better.

kcdtv, 2017-02-13:
Just to let it clear: I was experimenting and the readme and instruction given to use mode monitor  are concise and precise.
 

> You can make the world better.

For me you already did as i was waiting for this feature (packet injection) for nearly one year :smiley_cat:

astsam, 2017-02-16:
> Just to let it clear

Thank you for your words, testing and pushing me to implement the TX power control.

closed 2017-02-16 by astsam

ZeroChaos-, 2017-02-25:
You really intend to leave monitor mode control in WEXT only and not add support for nl80211?  And to ignore populating type?  I'm sad, I was hoping with a proper bug report this would at the very least stay open for tracking the issue, not closed with a wonderful comment like "You can make the world better".