astsam/rtl8812au #13: Device crash after some seconds of aireplay/airodump

Tetdoss, 2017-02-16:
Hello, first, thank you for your driver, it's been a while I'm looking for it.
So you're driver works fine but, only during some seconds, after, i have to plug out/plug in my device.
For example, I want to do an aireplay-ng -9 wlan2 :

```
root@debian:/# aireplay-ng -9 wlan2
17:33:47  Trying broadcast probe requests...
17:33:47  Injection is working!
17:33:48  Found 5 APs

17:33:48  Trying directed probe requests...
17:33:48  XX:XX:XX:DE:EB:1D - channel: 6 - 'Livebox-4c54'
17:33:49  Ping (min/avg/max): 3.374ms/8.020ms/16.214ms Power: -71.00
17:33:49  30/30: 100%

17:33:49  XX:XX:XX:DE:EB:1D - channel: 6 - 'orange'
17:33:49  Ping (min/avg/max): 3.060ms/6.744ms/15.140ms Power: -71.93
17:33:49  30/30: 100%

17:33:49  XX:XX:D7:E7:12:2D - channel: 6 - 'SFR WiFi FON'
17:33:52  Ping (min/avg/max): 3.712ms/10.320ms/54.999ms Power: -89.12
17:33:52  17/30:  56%

17:33:54  XX:XX:D7:E7:12:2F - channel: 6 - 'SFR WiFi Mobile'
17:33:57  Ping (min/avg/max): 3.108ms/7.849ms/17.405ms Power: -90.14
17:33:57  14/30:  46%

root@debian:/# aireplay-ng -9 wlan2
17:34:03  Trying broadcast probe requests...
17:34:05  No Answer...
17:34:05  Found 0 APs
```

As you can see, i'm losing more and more packets and this command doesn't work anymore at the end, I have to plug out/in my device.

I really don't understand why :/ Thank you for advance !

astsam, 2017-02-16:
1. APs E7:12:2D and E7:12:2F have very low signal strength. 50% missing packets is ok for such low signal level.

2.Will you recompile driver with DEBUG=1
`$ make DEBUG=1`
and reproduce the issue. I need more verbose  output from the driver.

Tetdoss, 2017-02-16:
Thank you for your quick answer, so here the more verbose output after a airodump-ng wlan2 : 

```
[44311.806554] RTL871X: cfg80211_rtw_set_monitor_channel : 12 0 0
[44312.572652] RTL871X: cfg80211_rtw_set_monitor_channel : 1 0 0
[44313.291919] RTL871X: cfg80211_rtw_set_monitor_channel : 7 0 0
[44314.001344] RTL871X: cfg80211_rtw_set_monitor_channel : 13 0 0
[44314.713894] RTL871X: cfg80211_rtw_set_monitor_channel : 2 0 0
[44315.433746] RTL871X: cfg80211_rtw_set_monitor_channel : 8 0 0
[44316.336246] RTL871X: cfg80211_rtw_set_monitor_channel : 3 0 0
[44317.056004] RTL871X: cfg80211_rtw_set_monitor_channel : 14 0 0
[44317.779373] RTL871X: cfg80211_rtw_set_monitor_channel : 9 0 0
[44318.577754] RTL871X: cfg80211_rtw_set_monitor_channel : 4 0 0
[44322.933059] RTL871X: cfg80211_rtw_set_monitor_channel : 6 0 0
[44322.933120] RTL871X: cfg80211_rtw_set_monitor_channel : 6 0 0
[44323.111481] RTL871X: cfg80211_rtw_set_monitor_channel : 6 0 0
[44323.111518] RTL871X: cfg80211_rtw_set_monitor_channel : 6 0 0
[44323.985683] RTL871X: ###=> usb_read_port_complete => urb status(-32)
[44323.988019] RTL871X: usb_read_port_complete() RX Warning! bDriverStopped(True) OR bSurpriseRemoved(False)
[44323.989126] RTL871X: usb_read_port_complete() RX Warning! bDriverStopped(True) OR bSurpriseRemoved(False)
[44323.990254] RTL871X: usb_read_port_complete() RX Warning! bDriverStopped(True) OR bSurpriseRemoved(False)
[44323.991439] RTL871X: usb_read_port_complete() RX Warning! bDriverStopped(True) OR bSurpriseRemoved(False)
[44323.992523] RTL871X: usb_read_port_complete() RX Warning! bDriverStopped(True) OR bSurpriseRemoved(False)
[44323.993678] RTL871X: usb_read_port_complete() RX Warning! bDriverStopped(True) OR bSurpriseRemoved(False)
[44323.994960] RTL871X: usb_read_port_complete() RX Warning! bDriverStopped(True) OR bSurpriseRemoved(False)
[44324.122139] RTL871X: rtw_cmd_thread: DriverStopped(True) SurpriseRemoved(False) break at line 597
```

I noticed that I only see my home router in debug mode, i can't see the other routers anymore.

Thank you again for your help !

Tetdoss, 2017-02-16:
After playing with airelplay-ng : 

`[48597.445776] RTL871X: _alloc_mgtxmitframe(wlan2) alloc xmitbuf fail
[48597.446786] RTL871X: _alloc_mgtxmitframe(wlan2) alloc xmitbuf fail
[48597.447085] RTL871X: _alloc_mgtxmitframe(wlan2) alloc xmitbuf fail
[48597.447533] RTL871X: _alloc_mgtxmitframe(wlan2) alloc xmitbuf fail
[48597.448204] RTL871X: _alloc_mgtxmitframe(wlan2) alloc xmitbuf fail
[48597.449213] RTL871X: _alloc_mgtxmitframe(wlan2) alloc xmitbuf fail
[48597.449514] RTL871X: _alloc_mgtxmitframe(wlan2) alloc xmitbuf fail
[48597.449962] RTL871X: _alloc_mgtxmitframe(wlan2) alloc xmitbuf fail
[48597.450633] RTL871X: _alloc_mgtxmitframe(wlan2) alloc xmitbuf fail
[48597.451643] RTL871X: _alloc_mgtxmitframe(wlan2) alloc xmitbuf fail
[48597.451944] RTL871X: _alloc_mgtxmitframe(wlan2) alloc xmitbuf fail
[48597.452392] RTL871X: _alloc_mgtxmitframe(wlan2) alloc xmitbuf fail
[48597.453064] RTL871X: _alloc_mgtxmitframe(wlan2) alloc xmitbuf fail
[48597.454070] RTL871X: _alloc_mgtxmitframe(wlan2) alloc xmitbuf fail
[48597.454370] RTL871X: _alloc_mgtxmitframe(wlan2) alloc xmitbuf fail
[48597.454819] RTL871X: _alloc_mgtxmitframe(wlan2) alloc xmitbuf fail
[48597.455491] RTL871X: _alloc_mgtxmitframe(wlan2) alloc xmitbuf fail
[48597.456501] RTL871X: _alloc_mgtxmitframe(wlan2) alloc xmitbuf fail
[48597.456801] RTL871X: _alloc_mgtxmitframe(wlan2) alloc xmitbuf fail
[48597.457249] RTL871X: _alloc_mgtxmitframe(wlan2) alloc xmitbuf fail
[48597.457940] RTL871X: _alloc_mgtxmitframe(wlan2) alloc xmitbuf fail
[48597.458949] RTL871X: _alloc_mgtxmitframe(wlan2) alloc xmitbuf fail
[48597.459249] RTL871X: _alloc_mgtxmitframe(wlan2) alloc xmitbuf fail
[48597.459697] RTL871X: _alloc_mgtxmitframe(wlan2) alloc xmitbuf fail
[48597.460370] RTL871X: _alloc_mgtxmitframe(wlan2) alloc xmitbuf fail
[48597.461377] RTL871X: _alloc_mgtxmitframe(wlan2) alloc xmitbuf fail
[48597.461676] RTL871X: _alloc_mgtxmitframe(wlan2) alloc xmitbuf fail
[48597.462124] RTL871X: _alloc_mgtxmitframe(wlan2) alloc xmitbuf fail
[48597.462795] RTL871X: _alloc_mgtxmitframe(wlan2) alloc xmitbuf fail
[48597.463301] RTL871X: usb_read_port_complete() RX Warning! bDriverStopped(True) OR bSurpriseRemoved(False)`

Tetdoss, 2017-02-17:
Found it ! I had to install USB 3 extension for VirtualBox !

Took me hours...

closed 2017-02-17 by Tetdoss