astsam/rtl8812au #14: does not work in band 2 as AP

atyg02, 2017-02-19:
I have compiled rtl8812au v4.3.21 for EW-7822UAC USB wifi in openwrt, 
It works on Band 1 as AP, however for Band 2 it complains with following message:
```
wlan0: IEEE 802.[ 2200.050060] device wlan0 left promiscuous mode
11 Configured ch[ 2200.055402] br-lan: port 2(wlan0) entered disabled state
annel (36) not found from the channel list of current mode (2) IEEE 802.11a
wlan0: IEEE 802.11 Hardware does not support configured channel
Could not select hw_mode and channel. (-3)
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
wlan0: Unable to setup interface.
wlan0: interface state DISABLED->DISABLED
wlan0: AP-DISABLED 
```

iw list output:
```
Wiphy phy1
        max # scan SSIDs: 9
        max scan IEs length: 2304 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Available Antennas: TX 0 RX 0
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * monitor
                 * P2P-client
                 * P2P-GO
        Band 1:
                Capabilities: 0x19e3
                        RX LDPC
                        HT20/HT40
                        Static SM Power Save
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 7935 bytes
                        DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 16 usec (0x07)
                HT Max RX data rate: 300 Mbps
                HT TX/RX MCS rate indexes supported: 0-15
                Frequencies:
                        * 2412 MHz [1] (20.0 dBm)
                        * 2417 MHz [2] (20.0 dBm)
                        * 2422 MHz [3] (20.0 dBm)
                        * 2427 MHz [4] (20.0 dBm)
                        * 2432 MHz [5] (20.0 dBm)
                        * 2437 MHz [6] (20.0 dBm)
                        * 2442 MHz [7] (20.0 dBm)
                        * 2447 MHz [8] (20.0 dBm)
                        * 2452 MHz [9] (20.0 dBm)
                        * 2457 MHz [10] (20.0 dBm)
                        * 2462 MHz [11] (20.0 dBm)
                        * 2467 MHz [12] (20.0 dBm) (no IR)
                        * 2472 MHz [13] (20.0 dBm) (no IR)
                        * 2484 MHz [14] (20.0 dBm) (no IR)
        Band 2:
                Capabilities: 0x19e3
                        RX LDPC
                        HT20/HT40
                        Static SM Power Save
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 7935 bytes
                        DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 16 usec (0x07)
                HT Max RX data rate: 300 Mbps
                HT TX/RX MCS rate indexes supported: 0-15
                VHT Capabilities (0x00000010):
                        Max MPDU length: 3895
                        Supported Channel Width: neither 160 nor 80+80
                        RX LDPC
                VHT RX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: MCS 0-9
                        4 streams: MCS 0-9
                        5 streams: MCS 0-9
                        6 streams: MCS 0-9
                        7 streams: MCS 0-9
                        8 streams: MCS 0-9
                VHT RX highest supported: 0 Mbps
                VHT TX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: MCS 0-9
                        4 streams: MCS 0-9
                        5 streams: MCS 0-9
                        6 streams: MCS 0-9
                        7 streams: MCS 0-9
                        8 streams: MCS 0-9
                VHT TX highest supported: 0 Mbps
                Frequencies:
                        * 5170 MHz [34] (30.0 dBm) (no IR)
                        * 5180 MHz [36] (30.0 dBm) (no IR)
                        * 5190 MHz [38] (30.0 dBm) (no IR)
                        * 5200 MHz [40] (30.0 dBm) (no IR)
                        * 5210 MHz [42] (30.0 dBm) (no IR)
                        * 5220 MHz [44] (30.0 dBm) (no IR)
                        * 5230 MHz [46] (30.0 dBm) (no IR)
                        * 5240 MHz [48] (30.0 dBm) (no IR)
                        * 5260 MHz [52] (30.0 dBm) (no IR, radar detection)
                        * 5280 MHz [56] (30.0 dBm) (no IR, radar detection)
                        * 5300 MHz [60] (30.0 dBm) (no IR, radar detection)
                        * 5320 MHz [64] (30.0 dBm) (no IR, radar detection)
                        * 5500 MHz [100] (30.0 dBm) (no IR, radar detection)
                        * 5520 MHz [104] (30.0 dBm) (no IR, radar detection)
                        * 5540 MHz [108] (30.0 dBm) (no IR, radar detection)
                        * 5560 MHz [112] (30.0 dBm) (no IR, radar detection)
                        * 5580 MHz [116] (30.0 dBm) (no IR, radar detection)
                        * 5600 MHz [120] (30.0 dBm) (no IR, radar detection)
                        * 5620 MHz [124] (30.0 dBm) (no IR, radar detection)
                        * 5640 MHz [128] (30.0 dBm) (no IR, radar detection)
                        * 5660 MHz [132] (30.0 dBm) (no IR, radar detection)
                        * 5680 MHz [136] (30.0 dBm) (no IR, radar detection)
                        * 5700 MHz [140] (30.0 dBm) (no IR, radar detection)
                        * 5745 MHz [149] (30.0 dBm) (no IR)
                        * 5765 MHz [153] (30.0 dBm) (no IR)
                        * 5785 MHz [157] (30.0 dBm) (no IR)
                        * 5805 MHz [161] (30.0 dBm) (no IR)
                        * 5825 MHz [165] (30.0 dBm) (no IR)
                        * 5920 MHz [184] (30.0 dBm)
                        * 5940 MHz [188] (30.0 dBm)
                        * 5960 MHz [192] (30.0 dBm)
                        * 5980 MHz [196] (30.0 dBm)
                        * 6000 MHz [200] (30.0 dBm)
                        * 6020 MHz [204] (30.0 dBm)
                        * 6040 MHz [208] (30.0 dBm)
                        * 6060 MHz [212] (30.0 dBm)
                        * 6080 MHz [216] (30.0 dBm)
        interface combinations are not supported
```

kcdtv, 2017-02-20:
I guess you are using **hostapd** to create the network.
Are you sure it does not work? 
I have tried with **airebase-ng** in _monitor mode_
Yes, there is some kind of warning about channel "-1" 
But it works perfectly: I got a client (i don't know who) connected 20 seconds after creating the PA... :smiley_cat:
I checked with airodump-ng on another interface in channel 2 and i perfectly see my PA with the client associated
[image]

atyg02, 2017-02-20:
@kcdtv yes I am using hostapd, but I said in band 2 which is in channel 36 - 196.[5GHz]
I think you have configured in channel 2 which is 2.4Ghz (band 1) that one is ok.
could you please try on band 2 and enable ieee80211ac.

kcdtv, 2017-02-20:
I got confused, it is the morning, i should have got my second cofee first... :smiley_cat: 
I have to get a second "a" band chipset to check it in airodump-ng...
when i create the interface with airebase-ng in channel 36 i have the same stdout (warning but it seems to work)
 I'll be back in 20 minutes with another device to confirm (or not) that the PA is created in chanel 36. 
I'll post the results asap.

kcdtv, 2017-02-20:
I managed to put my hand on a rt3572 device that is a band a device. 
With airebase-ng the PA is created in channel 36 (5ghz band) and i even got a unknown client connected: 
[image]
hostapd does enable automatically master mode to create the PA. 
What is kind of strange is to see that you are in "promiscuous mode", i don't use a lot hostapd so that might the usual stdout.  
Did you tried with other drivers?

atyg02, 2017-02-20:
yes I have tired with other drivers, They are good, I just want to try this driver to compare the performance and stability of  rtl8812au, but could not manage to start the hostapd in band 2 channels.

did you use V4.3.21 branch, because I have compiled that one ?

kcdtv, 2017-02-20:
>  I just want to try this driver to compare the performance and stability of rtl8812au

This drivers are more specifically for packet injection and monitor mode... 

> did you use V4.3.21 branch

Yes I did.

astsam, 2017-02-25:
This patch c8319348de74 may help. If the issue persists attach your hostapd.conf without any comments.

atyg02, 2017-02-27:
@astsam Thank you.
it works now.
I got around 220 mbps with an iperf 
```
[  4] local 192.168.10.192 port 5001 connected with 192.168.10.1 port 52741
[ ID] Interval       Transfer     Bandwidth
[  4]  0.0- 1.0 sec  24.3 MBytes   203 Mbits/sec
[  4]  1.0- 2.0 sec  29.0 MBytes   243 Mbits/sec
[  4]  2.0- 3.0 sec  28.4 MBytes   238 Mbits/sec
[  4]  3.0- 4.0 sec  27.8 MBytes   234 Mbits/sec
[  4]  4.0- 5.0 sec  28.1 MBytes   236 Mbits/sec
[  4]  5.0- 6.0 sec  28.3 MBytes   237 Mbits/sec
[  4]  6.0- 7.0 sec  27.7 MBytes   232 Mbits/sec
[  4]  7.0- 8.0 sec  26.8 MBytes   225 Mbits/sec
[  4]  8.0- 9.0 sec  24.6 MBytes   206 Mbits/sec
[  4]  9.0-10.0 sec  26.5 MBytes   223 Mbits/sec
[  4] 10.0-11.0 sec  23.6 MBytes   198 Mbits/sec
[  4] 11.0-12.0 sec  25.4 MBytes   213 Mbits/sec
[  4] 12.0-13.0 sec  25.5 MBytes   214 Mbits/sec
[  4] 13.0-14.0 sec  27.7 MBytes   232 Mbits/sec
[  4] 14.0-15.0 sec  28.0 MBytes   235 Mbits/sec
[  4] 15.0-16.0 sec  27.8 MBytes   233 Mbits/sec
[  4] 16.0-17.0 sec  28.0 MBytes   235 Mbits/sec
[  4] 17.0-18.0 sec  27.9 MBytes   234 Mbits/sec
[  4] 18.0-19.0 sec  27.2 MBytes   229 Mbits/sec
[  4] 19.0-20.0 sec  27.6 MBytes   231 Mbits/sec
[  4] 20.0-21.0 sec  27.1 MBytes   227 Mbits/sec
[  4] 21.0-22.0 sec  26.0 MBytes   218 Mbits/sec
[  4] 22.0-23.0 sec  27.3 MBytes   229 Mbits/sec
[  4] 23.0-24.0 sec  25.4 MBytes   213 Mbits/sec
[  4] 24.0-25.0 sec  25.2 MBytes   212 Mbits/sec
[  4] 25.0-26.0 sec  26.5 MBytes   222 Mbits/sec
[  4] 26.0-27.0 sec  28.8 MBytes   242 Mbits/sec
[  4] 27.0-28.0 sec  28.5 MBytes   239 Mbits/sec
[  4] 28.0-29.0 sec  27.1 MBytes   227 Mbits/sec
[  4] 29.0-30.0 sec  27.7 MBytes   232 Mbits/sec
[  4] 30.0-31.0 sec  28.5 MBytes   239 Mbits/sec
[  4] 31.0-32.0 sec  25.8 MBytes   216 Mbits/sec
[  4] 32.0-33.0 sec  27.6 MBytes   231 Mbits/sec
[  4] 33.0-34.0 sec  27.4 MBytes   230 Mbits/sec
[  4] 34.0-35.0 sec  28.6 MBytes   240 Mbits/sec
```

closed 2017-02-27 by atyg02

kcdtv, 2017-02-28:
Not bad :smiley_cat: 
Did you use iperf or iperf3?