astsam/rtl8812au #15: How to increase TX power?

vverloop, 2017-02-28:
I have installed the driver for my AWUS036ACH, but TX power is not high. It only 12 dBm.
I have tried everything to increase the TX power.

What do you suggest? What is the right way?

astsam, 2017-03-01:
1. Install v4.3.21 branch.
2. Attach adapter to USB 3.0 or externally powered USB 2.0 hub.
3. Use `$sudo iwconfig wlan0 txpower 30`

vverloop, 2017-03-01:
Thanks! I will buy a external powered USB 3.0 hub or/and an USB 3.0 Micro Y-cable. And try again.

kcdtv, 2017-03-02:
I also have the AWUS036ACH 
My laptop is weak and ti works perfectly on the USB 3.0 port with the original cable.
  Anyway: If you installed correctly the branch you should see 18dBm instead of 12dBm. 
   So before buying anything make sure that you installed correctly the branch for txpower.... 
cheers

vverloop, 2017-03-02:
Thanks for thinking with us.
I thought also that is should not be needed to use external power...i will try it tomorrow and let you know.

I use the AWUS036ACH on Kali Linux & Vmware at my Mac Book Pro model late 2013 (with USB 3.0). I also discovered that i must also set in Vmware the usb setting from usb 2.0 to usb 3.0.  It is in the advanced settings section of USB in Vmware.

vverloop, 2017-03-04:
Not working..Can anybody make a Youtube movie about this?

I should be simple, but i don't get it to work.

vverloop, 2017-03-04:
https://bugs.kali.org/view.php?id=3260

rujol, 2017-03-17:
Using this driver with 4.10.2-1-ARCH Linux and it is working, except increasing TxPower. Can this be done?

astsam, 2017-03-17:
TX power control is implemented in the v4.3.21 branch.

vverloop, 2017-03-17:
Not working @ kali?

vverloop, 2017-03-17:
Also make the driver Master, so nobody make that mistake.

closed 2017-03-18 by astsam

adrianslabu, 2017-11-23:
I fixed my problem!
With version 5.1.5 i had only 12 dBm, txpower didn't work, low signal.
I installed 4.3.21 and its perfect !

1. apt-get remove realtek-rtl88xxau-dkms
2. reboot
3. git clone -b v4.3.21 https://github.com/aircrack-ng/rtl8812au.git
4. cd rtl8812au
5. make
6. make install

Enjoy!

wanna-be-mr-robot, 2017-11-23:
all of this looks pretty recent-- i hope this works

wanna-be-mr-robot, 2017-11-23:
Package 'realtek-rtl88xxau-dkms' is not installed, so not removed
The following packages were automatically installed and are no longer required:
  casefile dconf-editor dconf-tools dissy gir1.2-nm-1.0 libbind9-140
  libblas-common libcdio-cdda1 libcdio-paranoia1 libcdio13 libdns162 libemu2
  libfwupd1 libgom-1.0-common libgtkspell3-3-0 libisc160 libisccfg140
  libllvm3.9 liblouis12 liblwgeom-2.3-0 libmozjs-24-0 libopencv-calib3d2.4v5
  libopencv-core2.4v5 libopencv-features2d2.4v5 libopencv-flann2.4v5
  libopencv-highgui2.4-deb0 libopencv-imgproc2.4v5 libopencv-objdetect2.4v5
  libopencv-video2.4v5 libpython3.5 libpython3.5-minimal libpython3.5-stdlib
  libqgis-core2.14.18 libqgis-gui2.14.18 libqgis-networkanalysis2.14.18
  libqgispython2.14.18 libradare2-1.6 libtracker-control-1.0-0
  libtracker-miner-1.0-0 libtracker-sparql-1.0-0 libva-drm1 libva-x11-1 libva1
  maltegoce peepdf python-brotlipy python-pylibemu python-rsvg python-unicorn
  python3.5 python3.5-minimal tcpd
Use 'apt autoremove' to remove them.
what the heck does this mean?

wanna-be-mr-robot, 2017-11-23:
i will still try the next step which is rebooting