astsam/rtl8812au #17: compile error on el7 (3.10.0-aaa.b.c)

ElCoyote27, 2017-03-01:
Hi everyone,
4.3.8 used to compile fine on el7 but didn't work too well. I'm trying to use your branch and getting these compile errors:
  CC [M]  /usr/src/rtl8812au/core/rtw_iol.o
  CC [M]  /usr/src/rtl8812au/core/rtw_sreset.o
  CC [M]  /usr/src/rtl8812au/core/rtw_btcoex.o
  CC [M]  /usr/src/rtl8812au/core/rtw_beamforming.o
  CC [M]  /usr/src/rtl8812au/core/rtw_odm.o
  CC [M]  /usr/src/rtl8812au/core/efuse/rtw_efuse.o
  CC [M]  /usr/src/rtl8812au/os_dep/osdep_service.o
  CC [M]  /usr/src/rtl8812au/os_dep/linux/os_intfs.o
/usr/src/rtl8812au/os_dep/linux/os_intfs.c:959:2: warning: initialization from incompatible pointer type [enabled by default]
  .ndo_select_queue = rtw_select_queue,
  ^
/usr/src/rtl8812au/os_dep/linux/os_intfs.c:959:2: warning: (near initialization for ‘rtw_netdev_ops.<anonymous>.ndo_select_queue’) [enabled by default]
  CC [M]  /usr/src/rtl8812au/os_dep/linux/usb_intf.o
  CC [M]  /usr/src/rtl8812au/os_dep/linux/usb_ops_linux.o
  CC [M]  /usr/src/rtl8812au/os_dep/linux/ioctl_linux.o
  CC [M]  /usr/src/rtl8812au/os_dep/linux/xmit_linux.o
  CC [M]  /usr/src/rtl8812au/os_dep/linux/mlme_linux.o
  CC [M]  /usr/src/rtl8812au/os_dep/linux/recv_linux.o
  CC [M]  /usr/src/rtl8812au/os_dep/linux/ioctl_cfg80211.o
/usr/src/rtl8812au/os_dep/linux/ioctl_cfg80211.c:93:12: error: ‘IEEE80211_BAND_2GHZ’ undeclared here (not in a function)
  .band   = IEEE80211_BAND_2GHZ,  \
            ^
/usr/src/rtl8812au/os_dep/linux/ioctl_cfg80211.c:151:2: note: in expansion of macro ‘CHAN2G’
  CHAN2G(1, 2412, 0),
  ^
/usr/src/rtl8812au/os_dep/linux/ioctl_cfg80211.c:102:12: error: ‘IEEE80211_BAND_5GHZ’ undeclared here (not in a function)
  .band   = IEEE80211_BAND_5GHZ,  \
            ^
/usr/src/rtl8812au/os_dep/linux/ioctl_cfg80211.c:168:2: note: in expansion of macro ‘CHAN5G’
  CHAN5G(34, 0),  CHAN5G(36, 0),
[....]

Any ideas?

astsam, 2017-03-02:
1. Checkout v4.3.21 branch.
2. Provide more information about kernel version and make options.

ElCoyote27, 2017-03-02:
Hi Atsam,
Thanks for the quick reply.
I checked out v4.3.21.
The OS is RHEL7 and the kernel is 3.10.0-514.6.2.el7.x86_64
Here's a log of my initial attempts:
bash-4.2$  git clone https://github.com/astsam/rtl8812au.git
Cloning into 'rtl8812au'...
remote: Counting objects: 1262, done.
remote: Compressing objects: 100% (16/16), done.
remote: Total 1262 (delta 5), reused 0 (delta 0), pack-reused 1246
Receiving objects: 100% (1262/1262), 3.92 MiB | 5.36 MiB/s, done.
Resolving deltas: 100% (774/774), done.
bash-4.2$ cd rtl8812au/
bash-4.2$ git checkout v4.3.21
Branch v4.3.21 set up to track remote branch v4.3.21 from origin.
Switched to a new branch 'v4.3.21'
bash-4.2$ make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/3.10.0-514.6.2.el7.x86_64/build M=/var/tmp/rtl8812au  modules
make[1]: Entering directory `/usr/src/kernels/3.10.0-514.6.2.el7.x86_64'
  CC [M]  /var/tmp/rtl8812au/core/rtw_cmd.o
cc1: error: -Werror=date-time: no option -Wdate-time
make[2]: *** [/var/tmp/rtl8812au/core/rtw_cmd.o] Error 1
make[1]: *** [_module_/var/tmp/rtl8812au] Error 2
make[1]: Leaving directory `/usr/src/kernels/3.10.0-514.6.2.el7.x86_64'
make: *** [modules] Error 2

this one is because gcc on EL7 is gcc 4.8.5

ElCoyote27, 2017-03-02:
After commenting out #EXTRA_CFLAGS += -Wno-error=date-time in the Makefile, I get:
bash-4.2$ gmake -j4
gmake ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/3.10.0-514.6.2.el7.x86_64/build M=/var/tmp/rtl8812au  modules
gmake[1]: Entering directory `/usr/src/kernels/3.10.0-514.6.2.el7.x86_64'
  CC [M]  /var/tmp/rtl8812au/core/rtw_cmd.o
  CC [M]  /var/tmp/rtl8812au/core/rtw_security.o
  CC [M]  /var/tmp/rtl8812au/core/rtw_debug.o
  CC [M]  /var/tmp/rtl8812au/core/rtw_io.o
  CC [M]  /var/tmp/rtl8812au/core/rtw_ioctl_query.o
  CC [M]  /var/tmp/rtl8812au/core/rtw_ioctl_set.o
  CC [M]  /var/tmp/rtl8812au/core/rtw_ieee80211.o
  CC [M]  /var/tmp/rtl8812au/core/rtw_mlme.o
  CC [M]  /var/tmp/rtl8812au/core/rtw_mlme_ext.o
  CC [M]  /var/tmp/rtl8812au/core/rtw_wlan_util.o
/var/tmp/rtl8812au/core/rtw_ieee80211.c: In function ‘dump_ht_cap_ie’:
/var/tmp/rtl8812au/core/rtw_ieee80211.c:1571:24: warning: ‘ht_cap_ielen’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  dump_ht_cap_ie_content(sel, ht_cap_ie+2, ht_cap_ielen);
                        ^
  CC [M]  /var/tmp/rtl8812au/core/rtw_vht.o
  CC [M]  /var/tmp/rtl8812au/core/rtw_pwrctrl.o
  CC [M]  /var/tmp/rtl8812au/core/rtw_rf.o
  CC [M]  /var/tmp/rtl8812au/core/rtw_recv.o
  CC [M]  /var/tmp/rtl8812au/core/rtw_sta_mgt.o
  CC [M]  /var/tmp/rtl8812au/core/rtw_ap.o
  CC [M]  /var/tmp/rtl8812au/core/rtw_xmit.o
  CC [M]  /var/tmp/rtl8812au/core/rtw_p2p.o
  CC [M]  /var/tmp/rtl8812au/core/rtw_tdls.o
  CC [M]  /var/tmp/rtl8812au/core/rtw_br_ext.o
  CC [M]  /var/tmp/rtl8812au/core/rtw_iol.o
  CC [M]  /var/tmp/rtl8812au/core/rtw_sreset.o
  CC [M]  /var/tmp/rtl8812au/core/rtw_btcoex.o
  CC [M]  /var/tmp/rtl8812au/core/rtw_beamforming.o
  CC [M]  /var/tmp/rtl8812au/core/rtw_odm.o
  CC [M]  /var/tmp/rtl8812au/core/efuse/rtw_efuse.o
  CC [M]  /var/tmp/rtl8812au/os_dep/osdep_service.o
  CC [M]  /var/tmp/rtl8812au/os_dep/linux/os_intfs.o
  CC [M]  /var/tmp/rtl8812au/os_dep/linux/usb_intf.o
/var/tmp/rtl8812au/os_dep/linux/os_intfs.c:914:2: warning: initialization from incompatible pointer type [enabled by default]
  .ndo_select_queue = rtw_select_queue,
  ^
/var/tmp/rtl8812au/os_dep/linux/os_intfs.c:914:2: warning: (near initialization for ‘rtw_netdev_ops.<anonymous>.ndo_select_queue’) [enabled by default]
  CC [M]  /var/tmp/rtl8812au/os_dep/linux/usb_ops_linux.o
  CC [M]  /var/tmp/rtl8812au/os_dep/linux/ioctl_linux.o
  CC [M]  /var/tmp/rtl8812au/os_dep/linux/xmit_linux.o
  CC [M]  /var/tmp/rtl8812au/os_dep/linux/mlme_linux.o
  CC [M]  /var/tmp/rtl8812au/os_dep/linux/recv_linux.o
  CC [M]  /var/tmp/rtl8812au/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /var/tmp/rtl8812au/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /var/tmp/rtl8812au/os_dep/linux/wifi_regd.o
/var/tmp/rtl8812au/os_dep/linux/ioctl_cfg80211.c:93:12: error: ‘IEEE80211_BAND_2GHZ’ undeclared here (not in a function)
  .band   = IEEE80211_BAND_2GHZ,  \
            ^
/var/tmp/rtl8812au/os_dep/linux/ioctl_cfg80211.c:151:2: note: in expansion of macro ‘CHAN2G’
  CHAN2G(1, 2412, 0),
  ^
/var/tmp/rtl8812au/os_dep/linux/ioctl_cfg80211.c:102:12: error: ‘IEEE80211_BAND_5GHZ’ undeclared here (not in a function)
  .band   = IEEE80211_BAND_5GHZ,  \
            ^
/var/tmp/rtl8812au/os_dep/linux/ioctl_cfg80211.c:168:2: note: in expansion of macro ‘CHAN5G’
  CHAN5G(34, 0),  CHAN5G(36, 0),
  ^
/var/tmp/rtl8812au/os_dep/linux/ioctl_cfg80211.c:168:2: error: initializer element is not constant
/var/tmp/rtl8812au/os_dep/linux/ioctl_cfg80211.c:168:2: error: (near initialization for ‘rtw_5ghz_a_channels[0].band’)
/var/tmp/rtl8812au/os_dep/linux/ioctl_cfg80211.c:168:2: error: initializer element is not constant
/var/tmp/rtl8812au/os_dep/linux/ioctl_cfg80211.c:168:2: error: (near initialization for ‘rtw_5ghz_a_channels[1].band’)
/var/tmp/rtl8812au/os_dep/linux/ioctl_cfg80211.c:169:2: error: initializer element is not constant
  CHAN5G(38, 0),  CHAN5G(40, 0),
  ^
/var/tmp/rtl8812au/os_dep/linux/ioctl_cfg80211.c:169:2: error: (near initialization for ‘rtw_5ghz_a_channels[2].band’)
/var/tmp/rtl8812au/os_dep/linux/ioctl_cfg80211.c:169:2: error: initializer element is not constant
/var/tmp/rtl8812au/os_dep/linux/ioctl_cfg80211.c:169:2: error: (near initialization for ‘rtw_5ghz_a_channels[3].band’)
/var/tmp/rtl8812au/os_dep/linux/ioctl_cfg80211.c:170:2: error: initializer element is not constant
  CHAN5G(42, 0),  CHAN5G(44, 0),
  ^
[...........]

ElCoyote27, 2017-03-02:
[out.txt](https://github.com/astsam/rtl8812au/files/814964/out.txt)
Attaching full log.

astsam, 2017-03-02:
1. Try `make KSRC=/usr/src/kernels/3.10.0-514.6.2.el7.x86_64`.
2. Attach file /usr/src/kernels/3.10.0-514.6.2.el7.x86_64/include/net/cfg80211.h.

ElCoyote27, 2017-03-02:
with 'make KSRC=....' , I got the same error so I guess it's looking in the right place.
That may be the fact that this enterprise kernel (el7) includes some old headers.
driver v4.3.8 used to compile fine, though...
Please note that RHEL7 is very similar to Centos7 (free).
Attached is the file you requested.

[cfg80211.h.txt](https://github.com/astsam/rtl8812au/files/815102/cfg80211.h.txt)

astsam, 2017-03-02:
Thanks. Will you attach another file /usr/src/kernels/3.10.0-514.6.2.el7.x86_64/Makefile

ElCoyote27, 2017-03-02:
Thanks to you! :) Here's the Makefile..
I'm also attaching the kernel config file for el7 (
/usr/src/kernels/3.10.0-514.6.2.el7.x86_64/.config)
[Makefile.txt](https://github.com/astsam/rtl8812au/files/815337/Makefile.txt)
[config.txt](https://github.com/astsam/rtl8812au/files/815338/config.txt)

astsam, 2017-03-03:
Bad news. Kernel version (accordingly Makefile) is 3.10.0, but the cfg80211 header is taken from 4.7 or higher kernel version. I see no way to adapt the driver for such  deeply patched kernel. Consider to use vanilla kernel or switch to Debian, Ubuntu, Kali, Mint etc.

closed 2017-03-03 by astsam