astsam/rtl8812au #19: 8814 build not recognizing wireless card.

PrometheanSet, 2017-03-13:
Arch linux, kernel 4.9.11.

Nothing makes this driver recognize the device. Not even building with "make RTL8814=1".

Is this functionality supposed to be up yet, as described in the README.md?

astsam, 2017-03-14:
Will you attach make output?

pablocyber, 2017-03-17:
After compiling I had to run make install RTL8814=1, then modprobe 8814au for it to load

PrometheanSet, 2017-03-18:
make RTL8814=1
[code]make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.9.0-kali3-amd64/build M=/root/rtl8812au  modules
make[1]: Entering directory '/usr/src/linux-headers-4.9.0-kali3-amd64'
  CC [M]  /root/rtl8812au/core/rtw_cmd.o
  CC [M]  /root/rtl8812au/core/rtw_security.o
/root/rtl8812au/core/rtw_security.c: In function ‘aes_cipher’:
/root/rtl8812au/core/rtw_security.c:1598:5: warning: this ‘for’ clause does not guard... [-Wmisleading-indentation]
     for (j = 0; j < 8; j++)
     ^~~
/root/rtl8812au/core/rtw_security.c:1601:2: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘for’
  payload_index = hdrlen + 8;
  ^~~~~~~~~~~~~
/root/rtl8812au/core/rtw_security.c: In function ‘aes_decipher’:
/root/rtl8812au/core/rtw_security.c:1984:5: warning: this ‘for’ clause does not guard... [-Wmisleading-indentation]
     for (j = 0; j < 8; j++)
     ^~~
/root/rtl8812au/core/rtw_security.c:1987:2: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘for’
  payload_index = hdrlen + 8;
  ^~~~~~~~~~~~~
  CC [M]  /root/rtl8812au/core/rtw_debug.o
  CC [M]  /root/rtl8812au/core/rtw_io.o
  CC [M]  /root/rtl8812au/core/rtw_ioctl_query.o
  CC [M]  /root/rtl8812au/core/rtw_ioctl_set.o
  CC [M]  /root/rtl8812au/core/rtw_ieee80211.o
  CC [M]  /root/rtl8812au/core/rtw_mlme.o
  CC [M]  /root/rtl8812au/core/rtw_mlme_ext.o
/root/rtl8812au/core/rtw_mlme_ext.c: In function ‘rtw_delba_check’:
/root/rtl8812au/core/rtw_mlme_ext.c:12242:7: warning: this ‘else’ clause does not guard... [-Wmisleading-indentation]
       else
       ^~~~
/root/rtl8812au/core/rtw_mlme_ext.c:12244:8: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘else’
        psta->recvreorder_ctrl[i].enable = _FALSE;
        ^~~~
/root/rtl8812au/core/rtw_mlme_ext.c:12245:7: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
       if (ret != _FAIL)
       ^~
/root/rtl8812au/core/rtw_mlme_ext.c:12247:8: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘if’
        rtw_reset_continual_no_rx_packet(psta, i);
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /root/rtl8812au/core/rtw_wlan_util.o
  CC [M]  /root/rtl8812au/core/rtw_vht.o
  CC [M]  /root/rtl8812au/core/rtw_pwrctrl.o
  CC [M]  /root/rtl8812au/core/rtw_rf.o
  CC [M]  /root/rtl8812au/core/rtw_recv.o
  CC [M]  /root/rtl8812au/core/rtw_sta_mgt.o
  CC [M]  /root/rtl8812au/core/rtw_ap.o
  CC [M]  /root/rtl8812au/core/rtw_xmit.o
  CC [M]  /root/rtl8812au/core/rtw_p2p.o
  CC [M]  /root/rtl8812au/core/rtw_tdls.o
  CC [M]  /root/rtl8812au/core/rtw_br_ext.o
  CC [M]  /root/rtl8812au/core/rtw_iol.o
  CC [M]  /root/rtl8812au/core/rtw_sreset.o
  CC [M]  /root/rtl8812au/core/rtw_btcoex.o
  CC [M]  /root/rtl8812au/core/rtw_beamforming.o
  CC [M]  /root/rtl8812au/core/rtw_odm.o
  CC [M]  /root/rtl8812au/core/efuse/rtw_efuse.o
  CC [M]  /root/rtl8812au/os_dep/osdep_service.o
  CC [M]  /root/rtl8812au/os_dep/linux/os_intfs.o
  CC [M]  /root/rtl8812au/os_dep/linux/usb_intf.o
  CC [M]  /root/rtl8812au/os_dep/linux/usb_ops_linux.o
  CC [M]  /root/rtl8812au/os_dep/linux/ioctl_linux.o
  CC [M]  /root/rtl8812au/os_dep/linux/xmit_linux.o
  CC [M]  /root/rtl8812au/os_dep/linux/mlme_linux.o
  CC [M]  /root/rtl8812au/os_dep/linux/recv_linux.o
  CC [M]  /root/rtl8812au/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /root/rtl8812au/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /root/rtl8812au/os_dep/linux/wifi_regd.o
  CC [M]  /root/rtl8812au/os_dep/linux/rtw_android.o
  CC [M]  /root/rtl8812au/os_dep/linux/rtw_radiotap.o
  CC [M]  /root/rtl8812au/os_dep/linux/rtw_proc.o
[... 63 lines cut ...]
  CC [M]  /root/rtl8812au/hal/phydm/rtl8812a/halhwimg8812a_rf.o
  CC [M]  /root/rtl8812au/hal/phydm/rtl8812a/halphyrf_8812a_ce.o
  CC [M]  /root/rtl8812au/hal/phydm/rtl8812a/phydm_regconfig8812a.o
  CC [M]  /root/rtl8812au/hal/phydm/rtl8812a/phydm_rtl8812a.o
  CC [M]  /root/rtl8812au/hal/phydm/txbf/haltxbfjaguar.o
  CC [M]  /root/rtl8812au/hal/phydm/rtl8821a/halhwimg8821a_fw.o
  CC [M]  /root/rtl8812au/hal/phydm/rtl8821a/halhwimg8821a_mac.o
  CC [M]  /root/rtl8812au/hal/phydm/rtl8821a/halhwimg8821a_bb.o
  CC [M]  /root/rtl8812au/hal/phydm/rtl8821a/halhwimg8821a_rf.o
  CC [M]  /root/rtl8812au/hal/phydm/rtl8821a/halphyrf_8821a_ce.o
  CC [M]  /root/rtl8812au/hal/phydm/rtl8821a/phydm_regconfig8821a.o
  CC [M]  /root/rtl8812au/hal/phydm/rtl8821a/phydm_rtl8821a.o
  CC [M]  /root/rtl8812au/hal/phydm/rtl8821a/phydm_iqk_8821a_ce.o
  CC [M]  /root/rtl8812au/platform/platform_ops.o
  CC [M]  /root/rtl8812au/core/rtw_mp.o
  CC [M]  /root/rtl8812au/core/rtw_mp_ioctl.o
  CC [M]  /root/rtl8812au/core/rtw_bt_mp.o
  LD [M]  /root/rtl8812au/8812au.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /root/rtl8812au/8812au.mod.o
  LD [M]  /root/rtl8812au/8812au.ko
make[1]: Leaving directory '/usr/src/linux-headers-4.9.0-kali3-amd64'[/code]

[8812au.ko.zip](https://github.com/astsam/rtl8812au/files/852591/8812au.ko.zip)

PrometheanSet, 2017-03-18:
'make install && modprobe 8814au'
or 
'make install --RTL8814=1 && modprobe 8814au'
-------output-------
install -p -m 644 8812au.ko  /lib/modules/4.9.0-kali3-amd64/kernel/drivers/net/wireless/
/sbin/depmod -a 4.9.0-kali3-amd64
modprobe: FATAL: Module 8814au not found in directory /lib/modules/4.9.0-kali3-amd64
------------------
Perhaps you are accidentally using the diederickdehaas variant, which supports neither frame injection nor monitor mode? I've been using it for a while myself. My host OS is using it now. But sadly, Ima need them features. That's kinda what I bought the card to do.

PrometheanSet, 2017-03-18:
From the Makefile, I think you meant:

make CONFIG_RTL8814A=y

----output-----

make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.9.0-kali3-amd64/build M=/root/rtl8812au  modules
make[1]: Entering directory '/usr/src/linux-headers-4.9.0-kali3-amd64'
  CC [M]  /root/rtl8812au/core/rtw_cmd.o
In file included from /root/rtl8812au/include/../hal/phydm/phydm_precomp.h:79:0,
                 from /root/rtl8812au/include/hal_data.h:25,
                 from /root/rtl8812au/core/rtw_cmd.c:23:
/root/rtl8812au/include/../hal/phydm/phydm.h:43:38: fatal error: rtl8814a/phydm_iqk_8814a.h: No such file or directory
 #include "rtl8814a/phydm_iqk_8814a.h"
                                      ^
compilation terminated.
/usr/src/linux-headers-4.9.0-kali3-common/scripts/Makefile.build:298: recipe for target '/root/rtl8812au/core/rtw_cmd.o' failed
make[4]: *** [/root/rtl8812au/core/rtw_cmd.o] Error 1
/usr/src/linux-headers-4.9.0-kali3-common/Makefile:1507: recipe for target '_module_/root/rtl8812au' failed
make[3]: *** [_module_/root/rtl8812au] Error 2
Makefile:150: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.9.0-kali3-amd64'
Makefile:1264: recipe for target 'modules' failed
make: *** [modules] Error 2
----------------------

Which probably means that upstream hasn't properly merged RTL8814au drivers yet.

astsam, 2017-03-18:
Try `$ sudo insmod 8814au.ko` after make.

astsam, 2017-03-18:
> From the Makefile, I think you meant:

No.

PrometheanSet, 2017-03-18:
Yeah, I was on the master branch. The v4.3.21 branch just compiled.

Thank you so much! I can't wait to go test it out 👍

closed 2017-03-19 by PrometheanSet