astsam/rtl8812au #20: 50% frames missing on kali

pablocyber, 2017-03-14:
On my kali (permanent install) / 4.9.0-kali3-amd64 / rtl8812au master branch
I ran wireshark simultaneously with an Alfa awus051NH card (works out of the box with kali), and Alfa awus036ac card.  My expectations are that I would see duplicate frames.  However, the rtl8812au only represents 25% of the frames monitored during a 90 second test. (I was expecting 50%.)  I noticed airodump-ng would be inconsistent in how quickly it identified access points, and verified in tshark that beacon frames are missing frequently (expecting one every 10ms.)

pablocyber, 2017-03-14:
Update: similar behavior with linux-mint, though I noticed that when the LED on the card would blink fast (twice per second) then a flood of beacon frames would be seen in tshark.  Otherwise when the LED was blinking slow - most of the time (once every two seconds), beacon frames were not there.

mveilleux, 2017-03-18:
I'm seeing similar behavior on the latest build of Raspian.

TommyWhite, 2017-04-05:
I have similar issues with AWUS036AC, efficiency is less than classic AWUS036H.

kcdtv, 2017-04-05:
> I have similar issues with AWUS036AC, efficiency is less than classic AWUS036H.

Why would you expect more?
The old AWUS036H is still the best option for 2.4 Ghz hacking. 
And you are using a "low cost" version of the AWUS036ACH (the hgih power version with rtl8812au) which would be more comparable with AWUS036H.

pablocyber, 2017-04-05:
No, the "H" designation refers to the high power output amplifier in these
models. There should be very little difference in the antenna receive
strength between the H and non-H models.  Besides, even if this were a
reception issue, at the very least, beacon frames should not be missing,
which should be the easiest frames to decode due to the low bitrate
modulation method used.

kcdtv, 2017-04-06:
No? :smiley_cat: 
Well.... antennas have absolutely nothing to do with this, they have the same **gain** (5dBi) because they are the sames . 
Why you say "no"? 
Believe it or not you do not get the same with the 30€ device than the 60€ one..  Yes, life is a bitch I know , I know.... Believe it or not the hardware are very different if you open them and really check what is inside... 
 Your device performance are comparable to TP-Link Archer T4UH  which is more or less at the same price (the alfa antennas are removable and better)
 Anyway that is offtopic  
  What is certain at this point is that there is not at all 50% of packet loss with the driver (in Kali or wherever)  :smiley_cat:

pablocyber, 2017-04-06:
Okay - you are correct in that no hardware is exactly the same, but if you open it up I'm pretty sure the big metal square houses the power amplifier.  Anyway - even if the RSSI is way different between the two, I ran an experiment with both adapters about 2 feet from the AP, so we should be at least seeing beacon frames consistently, about 10 per second, from each adapter, when capturing from both interfaces on wireshark.  However, in most cases the 036ac does not even see the beacon frames.  Airodump captures are also off...see below comparison...note the ESSID Mifi* is the AP that is two feet away:

**ALFA AWUS051NH - Channel 1, 10 second capture**
CH  1 ][ Elapsed: 6 s ][ 2017-04-06 08:06                                         
                                                                                               
 BSSID              PWR RXQ  Beacons    #Data, #/s  CH  MB   ENC  CIPHER AUTH ESSID
                                                                                               
 00:7F:28:69:*:F3   -9   8      103       40    4   1  54e. WPA2 CCMP   PSK  MiFi*            
 60:A4:4C:21:*:04  -36   0        1        1    0 157  54e  WPA2 CCMP   PSK  welcome-*
 9C:34:26:95:*:18  -70   1        0        0    0   1  54e. WPA2 CCMP   PSK  bik*             
 10:86:8C:DC:*:08  -72   3        0        0    0   1  54e. WPA2 CCMP   PSK  Domin*           
 AE:34:26:95:*:18  -72   0        1        0    0   1  54e. OPN              xfinitywif*      
 88:DC:96:3D:*:88  -73  61       99        0    0   1  54e. WPA2 CCMP   PSK  CheeseL*         
                                                                                               
 BSSID              STATION            PWR   Rate    Lost    Frames  Probe                     
                                                                                               
 (not associated)   B8:27:EB:D6:*:1A  -28    0 - 1      9        3                             
 00:7F:28:69:*:F3  B8:27:EB:3A:*:E2   -1    0e- 0      0        1                             
 00:7F:28:69:*:F3  B8:27:EB:61:*:EE  -38    0e- 0e     0        2                             
 00:7F:28:69:*:F3  02:22:91:00:*:04  -24    0 - 0e     0        6                             
 00:7F:28:69:*:F3  4C:66:41:E7:*:7B  -18    0 -24      0        4                             
 00:7F:28:69:*:F3  B8:27:EB:28:*:83  -22    0e- 0e     0        2                             

**ALFA AWUS036AC - Channel 1, 10 second capture**
CH  1 ][ Elapsed: 12 s ][ 2017-04-06 08:13 ][ fixed channel wlan2: -1         
                                                                               
 BSSID              PWR RXQ  Beacons    #Data, #/s  CH  MB   ENC  CIPHER AUTH E
                                                                               
 00:7F:28:69:*:F3  -18   0        1        0    0   1  54e. WPA2 CCMP   PSK  M
                                                                               
 BSSID              STATION            PWR   Rate    Lost    Frames  Probe     
                                                                               
 (not associated)   E8:4E:06:3E:*:2E  -71    0 - 1     18        8             
 00:7F:28:69:*:F3  02:22:91:00:*:04  -32    0 - 6      0        2             

**ALFA AWUS036AC - Channel 1,** captured about 20 seconds _**to include when blue light blinks fast - and most of the stuff appears when the blue light blinks fast:**_
Note that the channels are not just limited to channel 1, even though airodump is set with --channel 1

CH  1 ][ Elapsed: 18 s ][ 2017-04-06 08:14 ][ fixed channel wlan2: -1                                         
                                                                                                                       
 BSSID              PWR RXQ  Beacons    #Data, #/s  CH  MB   ENC  CIPHER AUTH ESSID
                                                                                                                       
 60:A4:4C:21:*:00  -35   0        8        1    0  11  54e  WPA2 CCMP   PSK  welcom*                                  
 22:86:8C:DC:*:08  -79   0        3        0    0   1  54e. OPN              xfinitywif*                              
 5C:96:9D:65:*:BF  -30  33        8        5    0   6  54e. WPA2 CCMP   PSK  welcome 2*                               
 00:7F:28:69:*:F3  -25   0       15        9    0   1  54e. WPA2 CCMP   PSK  MiFi*                                    
 30:B5:C2:B9:*:EA  -51  18        6        0    0   6  54e. WPA2 CCMP   PSK  Yodel_2.*                                 
 F8:E4:FB:4A:*:B0  -61  10        5        0    0   6  54e. WPA2 CCMP   PSK  8JYV*                                     
 00:26:62:A6:*:F2  -73   0        2        0    0   6  54e. WEP  WEP         5YUH*                                     
 00:18:01:EE:*:C8  -72   7        4        0    0   6  54e. WEP  WEP         FSGH*                                     
 88:DC:96:3D:*:88  -73  17       13        0    0   1  54e. WPA2 CCMP   PSK  CheeseL*                                  
 AE:34:26:95:*:18  -80  16       10        0    0   1  54e. OPN              xfinitywifi*                               
 10:86:8C:DC:*:08  -76   6        2        0    0   1  54e. WPA2 CCMP   PSK  Domin*                                   
 DE:34:26:95:*:18  -81  17        9        0    0   1  54e. WPA2 CCMP   MGT  <length:  0>                             
[... cut ...]
 12:86:8C:DC:*:08  -81  15        8        0    0   1  54e. WPA2 CCMP   PSK  <length:  0>                              
                                                                                                                        
 BSSID              STATION            PWR   Rate    Lost    Frames  Probe                                              
                                                                                                                        
 (not associated)   78:DD:08:D9:07:59  -33    0 - 1      0        1  welcom*

astsam, 2017-04-19:
> pablocyber

Will you check v5.1.5?

pablocyber, 2017-04-25:
I just compiled and loaded the v5.1.5 driver from uminokoe/rtl8812AU/tree/driver-5.1.5 on rpi 4.1.12-v7+. I was able to put into monitor mode but unable to change channels (it was stuck on channel 1).  Injection using aireplay --test was not working for me.
Next I tried mk-fg/rtl8812au and I was able to change channels, inject, and monitor most data rates.  The similar issue to 4.3.2.1 with datarates.  Because my phone (I suspect most phones) run 2x2 802.11 AC, the highest supported rate (on a 20mhz channel) by this driver appears to be 156 Mbps (I can see those occasionally on wireshark).  Because it is not decoding the 173 Mbps frames (the phone tries to connect at the highest rate) - I miss most of the data frame traffic in wireshark, and instead only see "VHT NDP Announcement" and "Ack" control frames as evidence that invisible traffic is still flowing.
I will check v5.1.5 variants on Kali next.

pablocyber, 2017-04-25:
update: I couldn't get uminokoe version to compile on Kali 4.9.0-kali3-amd64.  Maybe there is something I should change in the makefile? mk-fg version would compile but crash the kernel when I tried iwconfig wlan1 mode monitor.

astsam, 2017-04-26:
> I just compiled and loaded the v5.1.5 driver from uminokoe/rtl8812AU/tree/driver-5.1.5 on rpi 4.1.12-v7+.

Clone and complie the v5.1.5 driver from this repo.

pablocyber, 2017-04-26:
> I just compiled and loaded the v5.1.5 driver from uminokoe/rtl8812AU/tree/driver-5.1.5 on rpi 4.1.12-v7+.

> Clone and complie the v5.1.5 driver from this repo.

I tried to add the flag CONFIG_PLATFORM_ARM_RPI = y to the makefile but it says recipe for target 'modules' failed.  I tried the mk-fg Makefile and the 4.3.2.1 Makefile also.  Unfortunately I don't know how to edit Makefiles /recipies very well to get it to compile for the rpi.

astsam, 2017-04-26:
Restore the original Makefile and follow the instructions #10.

pablocyber, 2017-04-26:
Thanks I had forgotten about that.  I just compiled 5.1.5 from this repo for rpi3.  Same issue with missing certain VHT datarates exists.  
I also just tried on kali3-amd64 and same issue with missing a lot of frames except when the light blinks fast.

tangibilizing, 2017-09-12:
how can  cross compile the kernel , I am using the raspy zero w?

RavSS, 2017-09-26:
> how can I cross compile the kernel , I am using the raspy zero w, thx so much

If you're just trying to get the adapter to work, then the instructions above didn't work for me, even with the headers of my kernel version it failed to make, but I managed to get it working on an RPI Zero W (Stretch) by changing things around in the Makefile (it thinks the architecture is armv6l instead of just arm or armhf if I can remember correctly).

I then had to mess with the modules not loading which can be fixed by placing them (8812au.ko) in /lib/modules/<kernel> and running sudo depmod. For some reason it doesn't like the directory I choose for the driver (pointing to a more sensible location), but oh well.

[Here's a 8812au driver repo with a much better recipe than the one I made, goes from lines 890 to 899.](https://github.com/diederikdehaas/rtl8812AU/blob/driver-4.3.14/Makefile) It should avoid the above. Just remember to add "CONFIG_PLATFORM_ARM_RPI = y" as well and I think disable the i386 one. 

Hope this helped. Still no idea why I had to do this for the RPI Zero W but not my RPI 3B. Different arch?