astsam/rtl8812au #22: will you add v5.1.5 driver?

gsra99, 2017-03-25:
see [https://github.com/diederikdehaas/rtl8812AU/issues/63](url)

ojnickel, 2017-04-05:
would you also add support for rp?

kcdtv, 2017-04-05:
What do you mean by adding the driver? Which parts of the code, which feature are you lacking? You cannot just merge two branch between them and make things works like this.  
Does this version from Tenda  brings anything news in monitor mode or packet injection? 
Since when tenda does injection and monitor mode? I must have missed on step :smiley_cat: 
.

astsam, 2017-04-19:
Version 5.1.5 was added with additional support for RTL8814AU. RTL8821AU is not supported in this branch.

astsam, 2017-04-19:
> would you also add support for rpi?

 See #10

pablocyber, 2017-04-29:
I'm having trouble compiling for RTL8814AU on the pi - any suggestions?
I get this output:

sudo make CROSS_COMPILE=arm-linux-gnueabihf- ARCH=arm RTL8814=1
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- -C /lib/modules/4.1.19-v7+/build M=/home/pi/rtl8812au-5.1.5  modules
make[1]: Entering directory '/home/pi/linux'
  CC [M]  /home/pi/rtl8812au-5.1.5/core/rtw_cmd.o
arm-linux-gnueabihf-gcc: error: -mfloat-abi=soft and -mfloat-abi=hard may not be used together
scripts/Makefile.build:258: recipe for target '/home/pi/rtl8812au-5.1.5/core/rtw_cmd.o' failed
make[2]: *** [/home/pi/rtl8812au-5.1.5/core/rtw_cmd.o] Error 1
Makefile:1384: recipe for target '_module_/home/pi/rtl8812au-5.1.5' failed
make[1]: *** [_module_/home/pi/rtl8812au-5.1.5] Error 2
make[1]: Leaving directory '/home/pi/linux'
Makefile:1855: recipe for target 'modules' failed
make: *** [modules] Error 2

astsam, 2017-04-29:
Try to apply the attached patch.
[aarch64-compilation-fix.txt](https://github.com/astsam/rtl8812au/files/965997/aarch64-compilation-fix.txt)