astsam/rtl8812au #27: WPS Scanning not work?

QuasarX, 2017-04-04:
I was tried in Kali 2016.2 but WPS scan not work.
```
root@kali:~# wash -i wlan0

Wash v1.5.2 WiFi Protected Setup Scan Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
mod by t6_x <t6_x@hotmail.com> & DataHead & Soxrok2212

BSSID                  Channel       RSSI       WPS Version       WPS Locked        ESSID
---------------------------------------------------------------------------------------------------------------
[!] Found packet with bad FCS, skipping...
[!] Found packet with bad FCS, skipping...
[!] Found packet with bad FCS, skipping...
[!] Found packet with bad FCS, skipping...
[!] Found packet with bad FCS, skipping...
....
```

How we can scan WPS with reaver?

kcdtv, 2017-04-05:
It works, use the correct option, this is well documented and explained in wash argument description, search a little.....

QuasarX, 2017-04-09:
I was fixed with this codes;

```
sudo airmon-ng check kill

sudo ip link set wlan0 down
sudo iwconfig wlan0 mode monitor
sudo ip link set wlan0 up
```

But when I try with reaver

`reaver -i wlan0 -f -c 2 -b MyBSSID -vv`

I'm getting this error

`[!] WPS transaction failed (code: 0x04), re-trying last pin`

It is not try WPS key :(

QuasarX, 2017-04-09:
OK I fixed 💯 

I use old libpcap;

**Download [for x86 Kali];**
```
ftp://ftp.filearena.net/.pub0/linuxmint-debian/pools/B/pool/main/libp/libpcap/libpcap0.8_1.4.0-2_i386.deb
ftp://ftp.filearena.net/.pub0/linuxmint-debian/pools/B/pool/main/libp/libpcap/libpcap0.8-dev_1.4.0-2_i386.deb
```

**Download [for x64 Kali];**
```
ftp://ftp.filearena.net/.pub0/linuxmint-debian/pools/B/pool/main/libp/libpcap/libpcap0.8_1.4.0-2_amd64.deb
ftp://ftp.is.freebsd.org/pub/mint/latest/pools/B/pool/main/libp/libpcap/libpcap0.8-dev_1.4.0-2_amd64.deb
```

Open folder and use Terminal with this codes

**[for x86 Kali]**
`sudo dpkg -i libpcap0.8_1.4.0-2_i386.deb libpcap0.8-dev_1.4.0-2_i386.deb`

**[for x64 Kali]**
`sudo dpkg -i libpcap0.8_1.4.0-2_amd64.deb libpcap0.8-dev_1.4.0-2_amd64.deb`

closed 2017-04-09 by QuasarX