astsam/rtl8812au #29: V4.3.21 not effect

simepy, 2017-04-25:
Hello,

In ArchLinux 4.10.11
V4.3.21 don't work, I do:

`make`
`make install`
plug my Alfa
and
`sudo ip link set wlp0s20u3u3 down`
`sudo iw wlp0s20u3u3 set txpower 30`

And my Tx is always Tx-Power=12 dBm
Help please nothing change ?

Thanks

astsam, 2017-04-26:
Clone and compile the v4.3.21 branch from this repo. You compiled some driver without tx power control.

simepy, 2017-04-26:
Yes, I do
`git clone https://github.com/astsam/rtl8812au.git`
and `git branch v4.3.21`
`make`
`make install`
plug my Alfa
and
`sudo ip link set wlp0s20u3u3 down`
`sudo iw wlp0s20u3u3 set txpower 30`

astsam, 2017-04-26:
Use `sudo iw wlp0s20u3u3 set txpower fixed 3000` or `sudo iwconfig wlp0s20u3u3 txpower 30`.

simepy, 2017-05-03:
OK it's work thanks

closed 2017-05-03 by simepy