astsam/rtl8812au #30: buffered or paused / bursty data frame injection for v5.1.5

pablocyber, 2017-04-28:
I decided to test data frame injection since aireplay-ng --test is only mgt frames.  The use case is direct scanning / nmap, etc of stations associated to a wireless network where client isolation (no direct communicating to other stations) is imposed by the AP.

Injection testing v5.1.5 driver on on rpi3, 802.11n/ channel 161 using airtun-ng -a <apmacaddress> wlan1 to ping another station directly via the at0 interface airtun creates, I get 5-10 so pings that work naturally, then a pause for 5 or so seconds, then a burst of ping packets set out to the other station, which replies back rapidly to the queue of 5 pings. Looks like this:

pi@153:~ $ sudo ping 192.168.8.13
PING 192.168.8.13 (192.168.8.13) 56(84) bytes of data.
64 bytes from 192.168.8.13: icmp_seq=1 ttl=128 time=2.21 ms
64 bytes from 192.168.8.13: icmp_seq=2 ttl=128 time=1.17 ms
64 bytes from 192.168.8.13: icmp_seq=3 ttl=128 time=1.25 ms
64 bytes from 192.168.8.13: icmp_seq=4 ttl=128 time=1.18 ms
64 bytes from 192.168.8.13: icmp_seq=5 ttl=128 time=1.11 ms
64 bytes from 192.168.8.13: icmp_seq=6 ttl=128 time=1.16 ms
64 bytes from 192.168.8.13: icmp_seq=7 ttl=128 time=1.15 ms
64 bytes from 192.168.8.13: icmp_seq=8 ttl=128 time=1.18 ms
64 bytes from 192.168.8.13: icmp_seq=9 ttl=128 time=1.14 ms
64 bytes from 192.168.8.13: icmp_seq=10 ttl=128 time=1.12 ms
64 bytes from 192.168.8.13: icmp_seq=11 ttl=128 time=1.19 ms
64 bytes from 192.168.8.13: icmp_seq=12 ttl=128 time=1.20 ms
**64 bytes from 192.168.8.13: icmp_seq=13 ttl=128 time=5003 ms
64 bytes from 192.168.8.13: icmp_seq=14 ttl=128 time=3995 ms
64 bytes from 192.168.8.13: icmp_seq=15 ttl=128 time=2996 ms
64 bytes from 192.168.8.13: icmp_seq=16 ttl=128 time=1996 ms**
64 bytes from 192.168.8.13: icmp_seq=17 ttl=128 time=996 ms
64 bytes from 192.168.8.13: icmp_seq=18 ttl=128 time=1.20 ms
64 bytes from 192.168.8.13: icmp_seq=19 ttl=128 time=1.13 ms
64 bytes from 192.168.8.13: icmp_seq=20 ttl=128 time=1.20 ms
64 bytes from 192.168.8.13: icmp_seq=21 ttl=128 time=1.15 ms
64 bytes from 192.168.8.13: icmp_seq=22 ttl=128 time=1.84 ms
64 bytes from 192.168.8.13: icmp_seq=23 ttl=128 time=1.12 ms
**64** bytes from 192.168.8.13: icmp_seq=24 ttl=128 time=4996 ms
64 bytes from 192.168.8.13: icmp_seq=25 ttl=128 time=3994 ms
64 bytes from 192.168.8.13: icmp_seq=26 ttl=128 time=2995 ms
64 bytes from 192.168.8.13: icmp_seq=27 ttl=128 time=1995 **ms**

kcdtv, 2017-05-17:
You are right 
The issue is not directly linked to monitor mode, that's happens with all driver i could find around  in managed mode too... 
I did not realize it with speedtest because the downloading speed seems the expected one and this sudden ping increase does not really shows up, or you might think this is an interference, am issue with server...
The fact is that every 50 pings something like this happens:

> 64 bytes from 192.168.1.1: icmp_seq=10 ttl=64 time=2.20 ms
64 bytes from 192.168.1.1: icmp_seq=11 ttl=64 time=2.54 ms
64 bytes from 192.168.1.1: icmp_seq=12 ttl=64 time=2.54 ms
64 bytes from 192.168.1.1: icmp_seq=13 ttl=64 time=3.13 ms
64 bytes from 192.168.1.1: icmp_seq=14 ttl=64 time=2.15 ms
64 bytes from 192.168.1.1: icmp_seq=15 ttl=64 time=2.12 ms
64 bytes from 192.168.1.1: icmp_seq=16 ttl=64 time=2.51 ms
64 bytes from 192.168.1.1: icmp_seq=17 ttl=64 time=14.8 ms
64 bytes from 192.168.1.1: icmp_seq=18 ttl=64 time=11.6 ms
64 bytes from 192.168.1.1: icmp_seq=19 ttl=64 time=6.07 ms
64 bytes from 192.168.1.1: icmp_seq=20 ttl=64 time=2.53 ms
64 bytes from 192.168.1.1: icmp_seq=21 ttl=64 time=3.40 ms
64 bytes from 192.168.1.1: icmp_seq=22 ttl=64 time=4.49 ms
64 bytes from 192.168.1.1: icmp_seq=23 ttl=64 time=2.17 ms
64 bytes from 192.168.1.1: icmp_seq=24 ttl=64 time=2.60 ms
64 bytes from 192.168.1.1: icmp_seq=25 ttl=64 time=6.97 ms
64 bytes from 192.168.1.1: icmp_seq=26 ttl=64 time=4.34 ms
64 bytes from 192.168.1.1: icmp_seq=27 ttl=64 time=2.84 ms
64 bytes from 192.168.1.1: icmp_seq=28 ttl=64 time=2.62 ms
64 bytes from 192.168.1.1: icmp_seq=29 ttl=64 time=2.82 ms
64 bytes from 192.168.1.1: icmp_seq=30 ttl=64 time=2.49 ms
64 bytes from 192.168.1.1: icmp_seq=31 ttl=64 time=2.82 ms
64 bytes from 192.168.1.1: icmp_seq=32 ttl=64 time=2.61 ms
64 bytes from 192.168.1.1: icmp_seq=33 ttl=64 time=2.63 ms
64 bytes from 192.168.1.1: icmp_seq=34 ttl=64 time=2.71 ms
64 bytes from 192.168.1.1: icmp_seq=35 ttl=64 time=2.19 ms
64 bytes from 192.168.1.1: icmp_seq=36 ttl=64 time=2.64 ms
64 bytes from 192.168.1.1: icmp_seq=37 ttl=64 time=6.71 ms
64 bytes from 192.168.1.1: icmp_seq=38 ttl=64 time=2.18 ms
64 bytes from 192.168.1.1: icmp_seq=39 ttl=64 time=2.69 ms
64 bytes from 192.168.1.1: icmp_seq=40 ttl=64 time=3.81 ms
64 bytes from 192.168.1.1: icmp_seq=41 ttl=64 time=6.90 ms
64 bytes from 192.168.1.1: icmp_seq=42 ttl=64 time=5.01 ms
64 bytes from 192.168.1.1: icmp_seq=43 ttl=64 time=2.33 ms
64 bytes from 192.168.1.1: icmp_seq=44 ttl=64 time=2.23 ms
64 bytes from 192.168.1.1: icmp_seq=45 ttl=64 time=8.63 ms
64 bytes from 192.168.1.1: icmp_seq=46 ttl=64 time=3.53 ms
**64 bytes from 192.168.1.1: icmp_seq=47 ttl=64 time=3954 ms**
**64 bytes from 192.168.1.1: icmp_seq=48 ttl=64 time=2934 ms**
**64 bytes from 192.168.1.1: icmp_seq=49 ttl=64 time=1910 ms**
64 bytes from 192.168.1.1: icmp_seq=50 ttl=64 time=887 ms
64 bytes from 192.168.1.1: icmp_seq=51 ttl=64 time=2.29 ms
64 bytes from 192.168.1.1: icmp_seq=52 ttl=64 time=2.35 ms
64 bytes from 192.168.1.1: icmp_seq=53 ttl=64 time=3.60 ms

I tried all this drivers (in kali 2017.1) in managed mode:
**astsam  diederikdehaas  Grawp  paspro  rmbq  ulli-kroll  vsurrel  xxNull-lsk**
Some of them did not compile, some of them did not work with network manager, it would not associate.
 So i conected with wpa_supplicant and got the same ping problem: Network manager doesn't seems to be the problem in this case. 
  There are still some warning about strangely declared function when compiling ... May this be the origin of the problem? 
  There is definitely some instability that doesn't allow to use the device at 100% of its capacities.

kimocoder, 2017-05-17:
kcdtv: I have also worked with the ones you mention.. this drivers still got some issues that we all know. have been looking at several patches the last days, but still few of interest.

* buffered data frame injection
* airmon-ng support (adding virtual interface)

would also refer to [https://bugs.kali.org/view.php?id=3989](https://bugs.kali.org/view.php?id=3989) where issues are mentioned

kcdtv, 2017-05-18:
HI! 
Mine? That was just a fork I made at a certain moment from this branch 4.3.21. to try stuffs and I didn't get anywhere. You can forget about it. :smiley_cat:
I got the AWUS1900 and observed the same phenomenon
It appears less: + or - every 100 ping paquets instead of + or - 50 with rtl8812au 
But the "colapse" is longer and the average speed is strongly affected
you can see it in the picture:
[image]
Same AP, same distance, ping suddenly goes realy high and average ping speed is very affected,
If you just focus on download speed you would not see it. 
That's in managed mode but if it is so in managed mode, same happens in monitor mode. and we know that it is not due to any mod for monitor mode/packet injection.