astsam/rtl8812au #31: 4.3.21 branch suddenly won't compile rtl8814?

PrometheanSet, 2017-05-05:
with a fresh clone:
make RTL8814=1
yields the same kind of output as if I simply typed 'make'. Output below. 

Is there something silly that I'm forgetting or missing?

make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.10.13-1-ARCH/build M=/build/rtl8812au  modules
make[1]: Entering directory '/usr/lib/modules/4.10.13-1-ARCH/build'
  CC [M]  /build/rtl8812au/core/rtw_cmd.o
  CC [M]  /zazz/build/rtl8812au/core/rtw_security.o
/build/rtl8812au/core/rtw_security.c: In function ‘aes_cipher’:
/build/rtl8812au/core/rtw_security.c:1598:5: warning: this ‘for’ clause does not guard... [-Wmisleading-indentation]
     for (j = 0; j < 8; j++)
     ^~~
/build/rtl8812au/core/rtw_security.c:1601:2: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘for’
  payload_index = hdrlen + 8;
  ^~~~~~~~~~~~~
/build/rtl8812au/core/rtw_security.c: In function ‘aes_decipher’:
/build/rtl8812au/core/rtw_security.c:1984:5: warning: this ‘for’ clause does not guard... [-Wmisleading-indentation]
     for (j = 0; j < 8; j++)
     ^~~
/home/zazz/build/rtl8812au/core/rtw_security.c:1987:2: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘for’
  payload_index = hdrlen + 8;
  ^~~~~~~~~~~~~
  CC [M]  /build/rtl8812au/core/rtw_debug.o
  CC [M]  /build/rtl8812au/core/rtw_io.o
  CC [M]  /build/rtl8812au/core/rtw_ioctl_query.o
  CC [M]  /build/rtl8812au/core/rtw_ioctl_set.o
  CC [M]  /build/rtl8812au/core/rtw_ieee80211.o
  CC [M]  /build/rtl8812au/core/rtw_mlme.o
  CC [M]  /build/rtl8812au/core/rtw_mlme_ext.o
/build/rtl8812au/core/rtw_mlme_ext.c: In function ‘rtw_delba_check’:
/build/rtl8812au/core/rtw_mlme_ext.c:12242:7: warning: this ‘else’ clause does not guard... [-Wmisleading-indentation]
       else
       ^~~~
/build/rtl8812au/core/rtw_mlme_ext.c:12244:8: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘else’
        psta->recvreorder_ctrl[i].enable = _FALSE;
        ^~~~
/build/rtl8812au/core/rtw_mlme_ext.c:12245:7: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
       if (ret != _FAIL)
       ^~
/build/rtl8812au/core/rtw_mlme_ext.c:12247:8: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘if’
        rtw_reset_continual_no_rx_packet(psta, i);
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /build/rtl8812au/core/rtw_wlan_util.o
  CC [M]  /build/rtl8812au/core/rtw_vht.o
  CC [M]  /build/rtl8812au/core/rtw_pwrctrl.o
  CC [M]  /build/rtl8812au/core/rtw_rf.o
  CC [M]  /build/rtl8812au/core/rtw_recv.o
  CC [M]  /build/rtl8812au/core/rtw_sta_mgt.o
  CC [M]  /build/rtl8812au/core/rtw_ap.o
  CC [M]  /build/rtl8812au/core/rtw_xmit.o
  CC [M]  /build/rtl8812au/core/rtw_p2p.o
  CC [M]  /build/rtl8812au/core/rtw_tdls.o
  CC [M]  /build/rtl8812au/core/rtw_br_ext.o
  CC [M]  /build/rtl8812au/core/rtw_iol.o
  CC [M]  /build/rtl8812au/core/rtw_sreset.o
  CC [M]  /build/rtl8812au/core/rtw_btcoex.o
  CC [M]  /build/rtl8812au/core/rtw_beamforming.o
  CC [M]  /build/rtl8812au/core/rtw_odm.o
  CC [M]  /build/rtl8812au/core/efuse/rtw_efuse.o
  CC [M]  /build/rtl8812au/os_dep/osdep_service.o
  CC [M]  /build/rtl8812au/os_dep/linux/os_intfs.o
  CC [M]  /build/rtl8812au/os_dep/linux/usb_intf.o
  CC [M]  /build/rtl8812au/os_dep/linux/usb_ops_linux.o
  CC [M]  /build/rtl8812au/os_dep/linux/ioctl_linux.o
  CC [M]  /build/rtl8812au/os_dep/linux/xmit_linux.o
  CC [M]  /build/rtl8812au/os_dep/linux/mlme_linux.o
  CC [M]  /build/rtl8812au/os_dep/linux/recv_linux.o
  CC [M]  /build/rtl8812au/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /build/rtl8812au/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /build/rtl8812au/os_dep/linux/wifi_regd.o
  CC [M]  /build/rtl8812au/os_dep/linux/rtw_android.o
  CC [M]  /build/rtl8812au/os_dep/linux/rtw_radiotap.o
  CC [M]  /build/rtl8812au/os_dep/linux/rtw_proc.o
  CC [M]  /build/rtl8812au/os_dep/linux/ioctl_mp.o
  CC [M]  /build/rtl8812au/hal/hal_intf.o
  CC [M]  /build/rtl8812au/hal/hal_com.o
  CC [M]  /build/rtl8812au/hal/hal_com_phycfg.o
  CC [M]  /build/rtl8812au/hal/hal_phy.o
  CC [M]  /build/rtl8812au/hal/hal_dm.o
  CC [M]  /build/rtl8812au/hal/hal_btcoex.o
  CC [M]  /build/rtl8812au/hal/hal_mp.o
  CC [M]  /build/rtl8812au/hal/hal_hci/hal_usb.o
  CC [M]  /build/rtl8812au/hal/led/hal_usb_led.o
  CC [M]  /build/rtl8812au/hal/HalPwrSeqCmd.o
  CC [M]  /build/rtl8812au/hal/rtl8812a/Hal8812PwrSeq.o
  CC [M]  /build/rtl8812au/hal/rtl8812a/Hal8821APwrSeq.o
  CC [M]  /build/rtl8812au/hal/rtl8812a/rtl8812a_xmit.o
  CC [M]  /build/rtl8812au/hal/rtl8812a/rtl8812a_sreset.o
  CC [M]  /build/rtl8812au/hal/rtl8812a/rtl8812a_hal_init.o
  CC [M]  /build/rtl8812au/hal/rtl8812a/rtl8812a_phycfg.o
  CC [M]  /build/rtl8812au/hal/rtl8812a/rtl8812a_rf6052.o
  CC [M]  /build/rtl8812au/hal/rtl8812a/rtl8812a_dm.o
  CC [M]  /build/rtl8812au/hal/rtl8812a/rtl8812a_rxdesc.o
  CC [M]  /build/rtl8812au/hal/rtl8812a/rtl8812a_cmd.o
  CC [M]  /build/rtl8812au/hal/rtl8812a/usb/usb_halinit.o
  CC [M]  /build/rtl8812au/hal/rtl8812a/usb/rtl8812au_led.o
  CC [M]  /build/rtl8812au/hal/rtl8812a/usb/rtl8812au_xmit.o
/build/rtl8812au/hal/rtl8812a/usb/rtl8812au_xmit.c: In function ‘rtl8812au_xmitframe_complete’:
/build/rtl8812au/hal/rtl8812a/usb/rtl8812au_xmit.c:618:2: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
  if (pbuf < bulkPtr)
  ^~
/build/rtl8812au/hal/rtl8812a/usb/rtl8812au_xmit.c:620:3: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘if’
   if (descCount == pHalData->UsbTxAggDescNum)
   ^~
  CC [M]  /build/rtl8812au/hal/rtl8812a/usb/rtl8812au_recv.o
  CC [M]  /build/rtl8812au/hal/rtl8812a/usb/usb_ops_linux.o
  CC [M]  /build/rtl8812au/hal/efuse/rtl8812a/HalEfuseMask8812A_USB.o
  CC [M]  /build/rtl8812au/hal/efuse/rtl8812a/HalEfuseMask8821A_USB.o
  CC [M]  /build/rtl8812au/hal/phydm/phydm_debug.o
  CC [M]  /build/rtl8812au/hal/phydm/phydm_antdiv.o
  CC [M]  /build/rtl8812au/hal/phydm/phydm_antdect.o
  CC [M]  /build/rtl8812au/hal/phydm/phydm_interface.o
  CC [M]  /build/rtl8812au/hal/phydm/phydm_hwconfig.o
/build/rtl8812au/hal/phydm/phydm_hwconfig.c: In function ‘odm_RxPhyStatusJaguarSeries_Parsing’:
/build/rtl8812au/hal/phydm/phydm_hwconfig.c:1543:40: warning: ‘EVM’ may be used uninitialized in this function [-Wmaybe-uninitialized]
       pPhyInfo->RxMIMOSignalQuality[i] = EVM;
       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
  CC [M]  /build/rtl8812au/hal/phydm/phydm.o
  CC [M]  /build/rtl8812au/hal/phydm/halphyrf_ce.o
  CC [M]  /build/rtl8812au/hal/phydm/phydm_edcaturbocheck.o
  CC [M]  /build/rtl8812au/hal/phydm/phydm_dig.o
  CC [M]  /build/rtl8812au/hal/phydm/phydm_pathdiv.o
  CC [M]  /build/rtl8812au/hal/phydm/phydm_rainfo.o
  CC [M]  /build/rtl8812au/hal/phydm/phydm_dynamicbbpowersaving.o
  CC [M]  /build/rtl8812au/hal/phydm/phydm_powertracking_ce.o
  CC [M]  /build/rtl8812au/hal/phydm/phydm_dynamictxpower.o
  CC [M]  /build/rtl8812au/hal/phydm/phydm_adaptivity.o
  CC [M]  /build/rtl8812au/hal/phydm/phydm_cfotracking.o
  CC [M]  /build/rtl8812au/hal/phydm/phydm_noisemonitor.o
  CC [M]  /build/rtl8812au/hal/phydm/phydm_acs.o
  CC [M]  /build/rtl8812au/hal/phydm/phydm_beamforming.o
  CC [M]  /build/rtl8812au/hal/phydm/txbf/halcomtxbf.o
  CC [M]  /build/rtl8812au/hal/phydm/txbf/haltxbfinterface.o
  CC [M]  /build/rtl8812au/hal/phydm/rtl8812a/halhwimg8812a_fw.o
  CC [M]  /build/rtl8812au/hal/phydm/rtl8812a/halhwimg8812a_mac.o
  CC [M]  /build/rtl8812au/hal/phydm/rtl8812a/halhwimg8812a_bb.o
  CC [M]  /build/rtl8812au/hal/phydm/rtl8812a/halhwimg8812a_rf.o
  CC [M]  /build/rtl8812au/hal/phydm/rtl8812a/halphyrf_8812a_ce.o
  CC [M]  /build/rtl8812au/hal/phydm/rtl8812a/phydm_regconfig8812a.o
  CC [M]  /build/rtl8812au/hal/phydm/rtl8812a/phydm_rtl8812a.o
  CC [M]  /build/rtl8812au/hal/phydm/txbf/haltxbfjaguar.o
  CC [M]  /build/rtl8812au/hal/phydm/rtl8821a/halhwimg8821a_fw.o
  CC [M]  /build/rtl8812au/hal/phydm/rtl8821a/halhwimg8821a_mac.o
  CC [M]  /build/rtl8812au/hal/phydm/rtl8821a/halhwimg8821a_bb.o
  CC [M]  /build/rtl8812au/hal/phydm/rtl8821a/halhwimg8821a_rf.o
  CC [M]  /build/rtl8812au/hal/phydm/rtl8821a/halphyrf_8821a_ce.o
  CC [M]  /build/rtl8812au/hal/phydm/rtl8821a/phydm_regconfig8821a.o
  CC [M]  /build/rtl8812au/hal/phydm/rtl8821a/phydm_rtl8821a.o
  CC [M]  /build/rtl8812au/hal/phydm/rtl8821a/phydm_iqk_8821a_ce.o
  CC [M]  /build/rtl8812au/platform/platform_ops.o
  CC [M]  /build/rtl8812au/core/rtw_mp.o
  CC [M]  /build/rtl8812au/core/rtw_mp_ioctl.o
  CC [M]  /build/rtl8812au/core/rtw_bt_mp.o
  LD [M]  /build/rtl8812au/8812au.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /build/rtl8812au/8812au.mod.o
  LD [M]  /build/rtl8812au/8812au.ko
make[1]: Leaving directory '/usr/lib/modules/4.10.13-1-ARCH/build'

PrometheanSet, 2017-05-05:
Tried all branches. Same behavior persists.

Reverting to previous 4.3.21 iteration from March, which is unaffected:
[rtl8812au.old.zip](https://github.com/astsam/rtl8812au/files/980537/rtl8812au.old.zip)

astsam, 2017-05-06:
I see no errors in the output. Just several warnings. It's ok. The driver was built successfully. BTW, you miss RTL8814=1 and  built the driver for RTL8812AU. Try the v5.1.5 branch it should have better sensitivity in the monitor mode.

PrometheanSet, 2017-05-11:
That's my problem. 

'make RTL8814=1' simply builds the 8812 driver. This problem persists across all branches, including the v5.1.5 branch.

The only version I see that works properly? The old version of the 4.3.21 branch that I zipped, and posted in a reply to my own thread.

astsam, 2017-05-13:
Both v4.3.21 and v5.1.5 branches may be built for RTL8812AU/21AU or for RTL8814AU. The master branch has no support for RTL8814AU. There is no problem you noticed in the issue.

kcdtv, 2017-05-18:
I just got an AWUS1900 and compilation worked
```
make RTL8814=1 DEBUG=1
(...)
 Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/kcdtv/RTL88XXAU/rtl8812au/astsam/5.1.5/rtl8812au/8814au.mod.o
  LD [M]  /home/kcdtv/RTL88XXAU/rtl8812au/astsam/5.1.5/rtl8812au/8814au.ko
make[1]: se sale del directorio '/usr/src/linux-headers-4.9.0-kali4-amd64'
```
Are you sure that you are in the correct branch?
What did not worked properly is the **make install** order that is not supposed to be used but that i always use. 
```
sudo make install
install -p -m 644 8812au.ko  /lib/modules/4.9.0-kali4-amd64/kernel/drivers/net/wireless/
install: no se puede efectuar `stat' sobre '8812au.ko': No existe el fichero o el directorio
Makefile:1861: fallo en las instrucciones para el objetivo 'install'
make: *** [install] Error 1
```
It looks for 8812au.ko and should search for 8814au.ko
Maybe you tried it and thought that compilation did not work? 
I am pretty sure that or you forgot to be in correct branch, or you forgot to use the RTL814AU=1 at compilation time or you tried to "make install it" and have been confused by the error message.

kimocoder, 2017-05-18:
I also got the AWUS1900 and confirm compilation working for 8814AU.

alisidani, 2017-08-24:
guys i need a working driver rtl8814au for kali 4.11 for awus1900

kimocoder, 2017-08-24:
@alisidani Kali just updated the realtek-rtl88xxau-dkms package today, but it can be manually build from https://github.com/aircrack-ng/rtl8812au since the repo is more up2date than "astsam"'s unfortunately

alisidani, 2017-08-24:
I installed the realtek-rtl88xxau-dkms many times..kali can see the awus adapter and i can scan networks  but it fails to connect to any one..the status keep connecting and re-asking the password noting that im sure that my password is correct

kimocoder, 2017-08-24:
There is an update for the package available now. Use apt-get to update it

alisidani, 2017-08-24:
I updated and still asking for password and not connecting

alisidani, 2017-08-24:
Anyone got the awus1900 working on kali?or other linux distribution? Even on ubuntu it doesn't connect to wireless networks

kimocoder, 2017-08-24:
It works for me (",)

alisidani, 2017-08-24:
What driver did you install? What version of linux do you use? Are you booting from vmware?

alisidani, 2017-08-25:
When i try to make from https://github.com/aircrack-ng/rtl8812au i get an error: 
/bin/sh: 1: syntax error: "(" unexpected
Makefile:1264: recipe for target 'modules' failed

kimocoder, 2017-08-25:
Do you have the "linux-headers" installed? And which branch are you compiling? I've tested the all, so everyone should work besides the v5.1.5 duo to missing kernel v4.12 support. Anyways, installing the newly updated "realtek-rtl88xx-dkms" with apt-get also works fine & should/may now be used again because it contains both 4.11 and 4.12 support

kimocoder, 2017-08-25:
the connection problem you got is another problem, which personally i can't reproduce or take a look at at the time because of holiday, but Im home in 8 days and will absolutely take a look at the problem as i use this adapter myself :)

alisidani, 2017-08-25:
Okay 
I installed the linux 4.11 amd64 headers

Ill keep trying to figure out the problem and ill contact you

kimocoder, 2017-08-25:
great.. smells like a networkmanager problem, but.who knows for sure :/
anytime sir, would be happy to get more information for this problem to ensure it'll be fixed

alisidani, 2017-08-25:
I don't think its a network managrt problem..i installed WICD and still the same problem..i can see networks but cannot connect
I tried this on kali-parrot-ubuntu  and black arch

kimocoder, 2017-08-25:
Ok. I appreciate your search for infornation/solution for this problem, if you could file an issue report with given information in the link some posts above (the aircrack-ng repo) it would be great, then more people could be aware of the problem & others may also take interest in getting the problem solved with their findings/discussions around it. Again, thanks for your help