astsam/rtl8812au #36: v5.1.5 for 8812au and 8814au cannot authorise on Ubuntu 17.04

paspro, 2017-05-19:
I have a problem with the v5.1.5 driver using either an 8812au or an 8814au device on Ubuntu 17.04. The authorisation process when WPA2 is enabled fails and the system keeps asking for the password. If I disable the authorisation on the router then there is no such problem. The driver worked on Ubuntu 16.10 before I upgraded.

kcdtv, 2017-05-19:
I observed this kind of issue with some drivers and releases. 
I could connect using wpa_supplicant directly so this is more a network manager thing.
Can you connect in command line to your wpa network?

paspro, 2017-05-19:
@kcdtv I don't know how to do that. Could you please give me detailed instructions?

kcdtv, 2017-05-20:
:smiley_cat:  It is not something that is not documented...
  1) turn off network manager
`sudo systemctl stop network-mamnager`
  2) kill any potentially conflictive process 
`sudo kill wpa_supplicant`
`sudo kill dhclient`
  3) generate the configuration file to be used with wpa_supplicant
 `wpapassphrase <eSSID> <Wireless_key> >> /tmp/test.conf`
(eSSID is the name of your network and you have to put the wifi key you have configured.)
It will generate something like that

> network={
	ssid="WiFi_XXXX"
	#psk="WiFilibr3!"
	psk=3af35a94a429490ea2ae3e5fa6fd970516c45566255e0bd5c2094b93f3cfac0a
}

4) Launch wpa_supplicant with the configuration file
`sudo wpa_supplicant -i<interface> -c /tmp/test.conf `
Instead of interface put the identifier of your wifi device   
5) Open another shell (do not close the one you've just used right now; the process with wpa_suplicant is running) and get an IP
`sudo dhclient <interface>`
6) You should be connected. Check it (ping google for example) 
Once you checked you can close the first shell (or press ctrl+c to kill the process) and to restart network-manager simply do
`sudo systemctl restart network-manager`

paspro, 2017-05-22:
Thank you for the instructions. I can confirm that this process does indeed work so there must be some problem with the Network Manager in Ubuntu 17.04 which causes the authentication process to fail. I also noticed that I need to keep the terminal with the wpa_supplicant command up and running for the network to work. If I terminate this process the network stops working.

kcdtv, 2017-06-18:
Sorry for the delay, but i bearly used ubuntu. 
I had exactly the same issue than you.
Newtrok manager see perfectly my PA, the connexion process start.... and fail. 
I can see that it is not able to negociate the handshake (the PA dot is not green, as if the key was incorrect and dmesg confrims it )
The thing is that it ocurres with... all my USB device (tried with old ralink and atheros too) 
It seems that ubuntu network manager has a big instability problems (I got many crashes when i manipulate interfaces and drivers) and does not handle properly the USB wifi interfaz (the internal card: no problems) 
It is definitely something that has to do with network manager and Ubuntu.

astsam, 2017-06-23:
NM tries to change MAC on every connect but something goes wrong.
Add the following lines
`[device]`
`wifi.scan-rand-mac-address=no`
at the end of the /etc/NetworkManager/NetworkManager.conf
and restart NetworkManager
`sudo service NetworkManager restart`

kimocoder, 2017-06-24:
* NetworkManager just got updated in Kali,  btw

kcdtv, 2017-06-25:
The automatic mac change done by network manager (for scanning and connecting) goes well in Kali. since quite a while:
[image]
I added _random_ here too
```
[connection]
wifi.cloned-mac-address=random
```
Real mac is never used.
If that would be the reason of the fail in Ubuntu I guess we could not connect with the internal interface neither... But yeah, weird things can happens too, there is nothig to loose for trying.

kcdtv, 2017-07-07:
I tried in ubuntu tongiht with the trick
`[device]
wifi.scan-rand-mac-address=no`
 and it worked! I can connect. 
I won't try to understand why it affects the USB devices only.... :smiley_cat:
I guess we could close this issue as the problem with ubuntu is clearly this. 
Cheers

astsam, 2017-07-18:
There is a race condition between wpa_supplicant and the NetworkManager for hotplugged devices. It has no relations to the driver.

tdaitx, 2017-07-18:
@astsam do you have more information on this issue? I would appreciate any pointers on other bugs/discussions about this race condition - when I searched google this bug report was the only reference that made any sense at all, I couldn't find any other reports about the race condition.

astsam, 2017-07-30:
After inserting an USB WIFI:
1. Driver is loaded, interface is created.
2. The NetworkManager set a random MAC for the interface to scan APs.
3. The wpa_supplicant detect the new interface and got its MAC. At this point the interface already has the random MAC assigned by the NetworkManager. For the following operations the  wpa_supplicant will use this (randomly assigned) MAC.  
4. The NwtworkManager complete scan an restore the orignal MAC.
5. During the WPA handshake the wpa_suplicant use the random MAC instead of the original one.

For an internal WIFI adapter p.2 and 3 are swapped, the wpa_supplicant get the original MAC and all works smoothly.