astsam/rtl8812au #40: 8814au [as Access point]  Kernel warning

atyg02, 2017-06-06:
Hello!

I have a problem 8814au as Access point  in openwrt.
If I change the ssid or  the encryption key and then restarting the network manager.
it supposed to kill the hostapd process and start it again. However it gives kernel Warning 
and after that all the clients can not connect. But if I reboot the router, no Kernel warning and the clients can connect perfectly.

do you have any idea why this Kernel warning ?
Thanky you !

```
------------[ cut here ]------------
[  190.792066] WARNING: CPU: 1 PID: 2174 at /home/client/Downloads/LEDE-APU2-0.1-Beta/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/linux-x86_64/rtl8814au-2017-02-11/core/rtw_mlme_ext.c:13078 createbss_)
[  190.812928] Modules linked in: tunprox pppoe ppp_async option iptable_nat ath9k usb_wwan rt2800usb rt2800lib pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE i
[  190.979291] CPU: 1 PID: 2174 Comm: RTW_CMD_THREAD Not tainted 3.18.45 #22
[  190.986280]  0000000000000286 0000000000000000 ffffffff8141169f ffff88011ec8c718
[  190.994052]  0000000000000000 0000000000000009 ffffffff8103d2ed ffffc900010f6000
[  191.001757]  ffffc900010f6000 00000000ffffd54b ffffc900010f7860 ffff880119828000
[  191.009629] Call Trace:
[  191.012208]  [<ffffffff8141169f>] ? dump_stack+0x5e/0x7a
[  191.017769]  [<ffffffff8103d2ed>] ? warn_slowpath_common+0x6d/0x90
[  191.024343]  [<ffffffffa02fcde6>] ? createbss_hdl+0x136/0x150 [8814au]
[  191.031113]  [<ffffffffa02d9a4e>] ? rtw_cmd_thread+0x348/0x502 [8814au]
[  191.038052]  [<ffffffffa02d9706>] ? rtw_stop_cmd_thread+0x49/0x49 [8814au]
[  191.045227]  [<ffffffff81053305>] ? kthread+0xd5/0xe0
[  191.050538]  [<ffffffff81053230>] ? kthread_create_on_node+0x170/0x170
[  191.057290]  [<ffffffff81415bd8>] ? ret_from_fork+0x58/0x90
[  191.063085]  [<ffffffff81053230>] ? kthread_create_on_node+0x170/0x170
[  191.069826] ---[ end trace 323a4b05f4e4bd31 ]---
[  191.074816] RTL871X: rtw_cfg80211_indicate_connect(wlan2)
[  191.080357] RTL871X: assoc success
[  191.083976] IPv6: ADDRCONF(NETDEV_CHANGE): wlan2: link becomes ready
[  191.090686] br-lan: port 2(wlan2) entered forwarding state
[  191.096425] br-lan: port 2(wlan2) entered forwarding state
[  191.102448] RTL871X: cfg80211_rtw_change_bss(wlan2)
[  193.092811] br-lan: port 2(wlan2) entered forwarding state
```

NishaYume, 2017-08-31:
Hello im interested in use a 8814 as access point. 3 month later how works now? Are the warning fixed ?

kimocoder, 2017-08-31:
This repo hangs abit behind the fork found [here](https://github.com/aircrack-ng/rtl8812au/tree/v5.1.5) which got kernel v4.12 support, tlp issues are solved and 0 warnings when buildt. again, credits still goes to astsam for the main work + monitor/injection support