astsam/rtl8812au #41: Problem with injection

scutta2000, 2017-06-08:
Hi i just installed this drivers and i finally managed to get monitor mode to work THANKS. I still have problem with injection though. I am using a USB drive from alfa network: the AWUS036AC on a new install of kali linux weekly build 23. Thanks in advance

astsam, 2017-06-09:
Will you clarify injection issue.

kcdtv, 2017-06-09:
Maybe it is useful to point out that there is an issue with _aireplay-ng_ in 5Ghz band. 
It does not inject in some access point; see: [https://trac.aircrack-ng.org/ticket/1722](url)
If your issue is that, inexplicably, you cannot inject in some PA in 5Ghz frequencies despite very good injection.... That is likely to be the reason why. 
The issue affects ralink rt3572 too, and it is always exactly the same PA that is not "injectable", with rtl814au, rtl6612au and rt3572. 
 It took me time to understand that and it happens with my PA  :smile_cat:

kimocoder, 2017-06-09:
Confirmed here too, on 8814au. I dont use v5.1.5 due too better performance on the earlier driver

kcdtv, 2017-06-12:
Yep...
for rtl8814au : -b 5.1.5
for rtl8814au : -b 4.3.21
The best combo :smile_cat:

kimocoder, 2017-06-13:
Yeah, Kali is considering adding the new one, which we all are experiencing several problems with, 
some to mention for me on the 8814au:

* the frame buffering issue
* injection got hurt somewhere in the code, but on the older it seems slightly better.
* sometimes both firnwares struggles with scanning in monitor mode
* virtual device "support" for airmon-ng would be nice of course
* on v5.1.5 i get max 18 dBm, but the other offers 30 as you know (",)

the performance + injection simply are better on the old driver still & the adapter is AWUS1900 
running the 8814au as mentioned and what you @kcdtv mentioned you got

Mister_X,: would be perfect for some input (commits ;) over here
since you anyways seem to take interest in the driver ..(?) (^^,)

kimocoder, 2017-06-22:
@kcdtv take a look at the 2 commits by jpmv27 @ https://github.com/jpmv27/aircrack-ng/commits/jpmv27_master -- is that the 5Ghz fix?

jpmv27, 2017-06-22:
I am using this driver with an ALFA AWUS036AC dongle.
Before those commits, I was not seeing all APs, especially in 5 GHz band.
With those commits, I am now seeing all the APs and I was able to successfully
run --deauth against an AP that wasn't visible previously.
May not fix all the problems, but fixed some for me.

kcdtv, 2017-06-22:
thank you very much kimocoder!  
I will see if it solves this very rare issue (one PA out of 12) 😺

kimocoder, 2017-06-22:
creds all goes to @jpmv27 boys :)

kcdtv, 2017-06-23:
Excellent!   
It solves the issue and now i can inject in every 5Ghz PA in range. 100% instead of 0% on my ac AP :smiley_cat:

zpsmack, 2017-07-06:
Please help on how to apply this commit?

jc765, 2017-10-01:
@jpmv27 Does the injection work on any DFS channels?