astsam/rtl8812au #42: usb adapter doesn't work

Tikani, 2017-06-22:
Hello!
There is a device: [TP-LINK Archer T9UH](https://wikidevi.com/wiki/TP-LINK_Archer_T9UH)
I've successfully compiled 8814au from v5.1.5 branch and installed 8814au.ko and I can load it with modprobe. But whenever I plug the device nothing happens - no flashing or glowing LED or availabe interface in "ip link show". I've tried to plug it to both USB 3.0/USB 2.0 ports - with the same effect.
Have I done anything wrong?
P.S. The device is definitely not broken - it works under Windows.

kcdtv, 2017-06-22:
Which distribution do you use? 
Could you paste here the stdout for `lsusb` ?
It doesn't look like you have done something wrong as you can launch the module...At first sight.

astsam, 2017-06-22:
Will you attach dmesg output.

Tikani, 2017-06-22:
My distribution is Debian 9.0 Stretch Stable.

**lsusb** stdout when the device **is** plugged:

Bus 002 Device 002: ID 8087:8001 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8009 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 046d:c328 Logitech, Inc. 
Bus 003 Device 009: ID 2357:0106  
Bus 003 Device 006: ID 248a:8367  
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

**dmesg** output when the device is plugged:

[28318.771337] usb 3-2: new high-speed USB device number 9 using xhci_hcd
[28318.911667] usb 3-2: New USB device found, idVendor=2357, idProduct=0106
[28318.911668] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[28318.911669] usb 3-2: Product: 802.11ac NIC
[28318.911670] usb 3-2: Manufacturer: Realtek
[28318.911670] usb 3-2: SerialNumber: 123456

**sudo lsmod | grep 8814au** output:

8814au               1351680  0
cfg80211              589824  1 8814au
usbcore               249856  8 usbhid,usb_storage,ehci_hcd,8814au,xhci_pci,uas,xhci_hcd,ehci_pci

Tikani, 2017-06-24:
Heeelp please :'-)

astsam, 2017-06-26:
Try commit dd20cc42a6f217.

Tikani, 2017-06-27:
Huge thanks! All are working like a charm!

closed 2017-06-27 by Tikani