astsam/rtl8812au #44: Problem with installation

Tikani, 2017-06-27:
The driver for RTL8814AU from v5.1.5 branch can be successfully compiled (`make RTL8814 = 1`), but `make install` doesn't work properly - `make` uses `install` target with RTL8812 in mind. 
Thus I have to do:
`sudo install -p -m 644 8814au.ko /lib/modules/4.9.0-3-amd64/kernel/drivers/net/wireless/
sudo depmod -a $(uname -r)`
manually in a terminal.
OK, it may be the case that I don't know something (for example, it's needed to set a specific environment variable or whatever), but even so, I've not found any mention about it in the Readme.

astsam, 2017-06-30:
To install RTL8814AU driver try `sudo make RTL8814=1 install`.

Tikani, 2017-07-03:
Thank a lot. I'm closing the issue.

closed 2017-07-03 by Tikani