astsam/rtl8812au #45: Compiling kernel error htc one m7cdwg

sviswas7, 2017-06-28:
Error with v4l2 driver video for linux during compiliation help pls

Tikani, 2017-06-28:
Show logs and output please

kcdtv, 2017-06-28:
And explain yourself a bit better too...
 

> Error with v4l2 driver video

What is the relation with WiFi drivers? This is very confusing.

sviswas7, 2017-06-29:
i dont know whats the relation with wifi drivers but as the tutorial states i have enabled dvd frontends in the last step there i have seen that after compiling iam getting an error reguarding v4l2 i will post the log soon

sviswas7, 2017-06-29:
thank you for replying iam recompiling the kernel as i found the fix my kernel doesnt support multimedia controller api (which iam enabling from the tutorial) disabling it crossed the error for v4l2....still compiling the kernel wii update for any other errors

sviswas7, 2017-06-29:
i have successfully compiled the kernel :)

Tikani, 2017-06-29:
This repository contains WiFi-drivery for the specific chipset only. Topic of your issue doesn't related with it at all. Please, close thread (anyway, your problem has been successfully resolved).

closed 2017-06-29 by sviswas7