astsam/rtl8812au #46: V5.15 for rtl8814au only got 300M at 5GHZ

zyworkshop, 2017-07-05:
IEEE 802.11  ESSID:"LEDE_5G"  
          Mode:Managed  Frequency:5.26 GHz  Access Point: 20:76:93:38:F9:EA   
          Bit Rate=300 Mb/s   Tx-Power=12 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on
          Link Quality=57/70  Signal level=-53 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

Tikani, 2017-07-05:
What speed v4.3.21 offers in comparison with v5.1.5? What distance is between client station and access point? Are there any other WLANs nearby working on 5GHz frequency?

zyworkshop, 2017-07-13:
V4.3.21 offers me 520Mb/s. Only 4 metres between client station and access point and there aren't not any Wlans working on 5GHZ except mine.

lhernanz, 2017-09-26:
It seems that AC mode does not work for 5.1.5. At least, it does not work for me with the Alfa AC1900. If your card only has 2 streams, the 300 would be the maximum speed at N rates. Therefore, you would be suffering the same problem than me.

kimocoder, 2017-09-26:
A fix is included on https://github.com/aircrack-ng/rtl8812au but it needs testing as I dont got 5ghz networks close for some weeks.

lhernanz, 2017-09-26:
Hi,
@kimocoder. Unfortunately that branch suffers from the same problem. I only connect using "N" speeds. AC is shown in the device capabilities (i.e. when running iw phy) but it is not used in practice. I have tried with two different AP systems where I can connect with AC using other cards, so the problem is clearly in the driver.

Can you point me to the specific code that you have changed? Maybe I can try to debug it a little bit.

Thanks

Regards

kimocoder, 2017-09-26:
It's [here](https://github.com/aircrack-ng/rtl8812au/commit/4282ab418914d508179a199a5bde0e0a0139fed2), remember the aircrack-ng/rtl8812au is a fork of this (astsam) repo so bugs would be the same for both as for the AC  issue.

would be great to pinpoint the issue, but I haven't been able to locate it yet.
im using the AWUS1900 on 8814au for both usage & testing

lhernanz, 2017-09-26:
Yes, I know about the fork, I have tried the two branches without too much success. The patch you are pointing at is the one that I contributed to switch the card from USB2 to USB3. That piece is working, but the card still stays at N speeds, so the problem is unrelated.

kimocoder, 2017-09-26:
Oh, my mistake.. It's been a long day :/ well, thanks for the heads up, will continue looking for a solution now that Im aware.

dernyn, 2018-10-15:
I have confirmed that 8812au does AC at 867 Mb/s

Yet the 8814au is hard set for some reason at 450 Mb/s and won’t change to anything even on signal degradation.

Why ?