astsam/rtl8812au #48: v5.1.5 doesn't work with TP-Link Archer T4U

francoism90, 2017-07-07:
See details: https://wikidevi.com/wiki/TP-LINK_Archer_T4U

Thanks. :)

Tikani, 2017-07-07:
I had the same problem with TP-LINK Archer T9UH. Please show `lsusb` amd `dmesg` output. In my case, maintainer of this repo just added VID/PID of the device to `os_dep/linux/usb_intf.c`. I recompiled kernel module after that and everything became OK.

francoism90, 2017-07-07:
$ lsusb: `ID 2357:010d`
$ dmesg: `usb 3-9.2: new high-speed USB device number 15 using xhci_hcd`

The `dmesg` output isn't that helpful, unfortunately this is everything I see when the USB has been attached.

Tikani, 2017-07-08:
I've tried to add support of your device to v5.1.5 branch and suggested a pull request, but I don't know will it be approved or not. Anyway, you can try out my fork directly if you need it urgently. Please give a response in case of success.
P.S. Sorry for my English, I know it quite bad.

jo80, 2017-07-30:
Hi, Please add to a code my device of USB-AC56 too.
Unfortunately the monitor mode does not work on my device. Thank you very much.

lsusb:    Bus 001 Device 002: ID 0b05:17d2 ASUSTek Computer, Inc. 
dmesg:
[    1.509173] usb 1-1: new high-speed USB device number 2 using xhci_hcd
[    1.637963] usb 1-1: New USB device found, idVendor=0b05, idProduct=17d2
[    1.637969] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    1.637973] usb 1-1: Product: 802.11n NIC
[    1.637976] usb 1-1: Manufacturer: Realtek
[    1.637978] usb 1-1: SerialNumber: 123456

rhertzog, 2017-08-19:
Looks like that "ID 2357:010d" is missing, yes. I got a similar bug report in Kali: https://bugs.kali.org/view.php?id=4157

And the ID is present in https://github.com/abperiasamy/rtl8812AU_8821AU_linux ... how many concurrent drivers do we have here? Couldn't we get some cooperation and get those merged upstream?

kimocoder, 2017-08-19:
gnab got one, abperiasamy got one, but the only one supporting monitor/injection is astsam's.

Anyways, it's easy to add device id's to the driver, but if they get merged or not & or how long time that may take is another story :/

rhertzog, 2017-08-21:
@kimocoder Adding support for monitor/injection should still be seen as an add-on. So you should be able to regularly merge from the reference implementation of the driver... but what is the reference implementation? It would be the one that gets merged into mainline Linux. Is anyone working on that?

kimocoder, 2017-08-21:
No, not that I've seen. 

But duo to several problems with both mon/inj. on this driver stack, I put my trust to that Mister_X would take a look at those issues (regarding missing virtual interface + ping buffer + several others) we have here. The "abperiasamy" repo also uses another realtek stack with lots of other drivers included (take a look at the code), while this mainly supports rtl8812au & rtl8814au. Which one would be "smartest" to work with regarding to Kali support? Really dont know where to go with this.. drivers may almost be seen as 'experimental' at this point.

rhertzog, 2017-08-22:
@kimocoder This is now off-topic for this thread, let's continue this sub-discussion here: https://bugs.kali.org/view.php?id=3989#c7067

closed 2018-09-26 by francoism90