astsam/rtl8812au #51: 5.2.9 driver for 8812AU only

gordboy, 2017-07-12:
Not exactly an issue with your work, more a heads-up about my first proper github repo, featuring the new 5.2.9 driver for 8812AU only, direct from Realtek FTP site.

https://github.com/gordboy/rtl8812au

I'm fairly new to all this github stuff, so please excuse my rudeness, and thanks for all your work on the 5.1.5 driver.

kcdtv, 2017-07-13:
Thanks for sharing! 
Compilation is 100% clean which is a very good sign.
Where did you find the source code from realtek?
I cannot put my hand on it.

kimocoder, 2017-07-13:
Thanks from me too!
Interesting readings, will take a look around for an update for the rtl8814 but feel free too share your findings (sources) for the 8812 which @kcdtv mentions :)

again, thanks for the heads up sir!

kimocoder, 2017-07-13:
kcdtv, im on vecation and left my laptop at home but [here](http://www.realtek.com.tw/DOWNLOADS/ExpressFTPSite.aspx?DownTypeID=3&DownID=436&PFid=1&Conn=5) is a link for the realtek official ftp site/sources

kcdtv, 2017-07-13:
Thanks! 
I will be able to try them tonight but i have a very good feeling seeing the changelog from realtek. 
Enjoy your vacations!

paspro, 2017-07-13:
Thank you for the information but I cannot access the FTP site with the given information. The site 218.210.127.133 does not exist. There is however ftp4.realtek.com which does not accept the password provided in the link you gave us.

kimocoder, 2017-07-13:
Seems offline that noe, but ftp://ftp3.realtek.com.tw is online. Anyone, a Google search would reveal more information too. I just cannot check this without my computer at the time. Im sure kcdtv will figure it out though :) and thanks for both your reply & your work in the wifi scene in general, mr kcdtv :)

gordboy, 2017-07-13:
I got the source from Realtek FTP - ftp3.realtek.com.tw, user: RTL8812A pass: J2e8Fg and
I found that when reading http://www.zapo.com.cn/product_page.asp?id=68 ... which I found when reading
https://aur.archlinux.org/packages/rtl8812au-v5-dkms-git/ ... which I found after searching Archlinux AUR for 8812AU https://aur.archlinux.org/packages/?O=0&SeB=nd&K=rtl8812au&SB=n&SO=a&PP=50&do_Search=Go. I would have commented to zebulon2 on his github page, but he has no Issues tab, which is why I came here, as zebulon2, paspro and astsam are all forkmates.

paspro, 2017-07-13:
Great. This provides access to the RTL8812AU driver. I am looking for the RTL8814AU driver but I guess that for that one needs another combination of user name / password. Why all this trouble to get their Linux driver?

gordboy, 2017-07-13:
@paspro My guess is that the RTL8814AU driver FTP username is "8814A", by analogy with the other drivers on the Zapo website. I realise that getting the password might be tricky ... unless it is on another OEM website :)

gordboy, 2017-07-13:
And I just tried user "8814A" with Filezilla and the same password as 8812A, with no luck, but it did say that user 8814A exists. This is a good start.

Connecting to 60.251.198.230:21...
Status:	Connection established, waiting for welcome message...
Status:	Insecure server, it does not support FTP over TLS.
Command:	USER RTL8814A
Response:	331 User RTL8814A OK. Password required
Command:	PASS ******
Response:	530 Login authentication failed

kcdtv, 2017-07-13:
> Why all this trouble to get their Linux driver?

That's insane isn't it?
Ok, I have just tried briefly and the first impression is good. I have no time now to go in deep testing. l  will do it this week-end and i'll share the results with you
Despite the fact that it is fucked up to get their source, it is a good new to see that realtek seems to be a bit more attentive to GNU-Linux support than in the past years.
  Because after rtl8817l, the support just went worst and worst....
See you :smiley_cat:

kcdtv, 2017-07-14:
First little feedback: 
I still observe the "ping burst" with version `5.29.` as i observe it with `4.3.21` or `5.1.5` 
[image]
You can see that when I connect with the rtl8812au chipset from the router AC1200R (client mode) to my box and in this case  i do not have any ping increase. 
 I was hoping that the new drivers would help solve this issues but it doesn't look like that it is the case, i would say that it remains the same.

gordboy, 2017-07-14:
Yes, I can confirm that there are occasional ping anomalies here too. But the overall connection is very good, and the memory leaks from 5.1.5 seem to have been cured. I was getting all sorts of warnings and kernel stack traces with 5.1.5.

I'm not as bothered about the occasional non-fatal glitch as I am about kernel stack traces messing up my pristine logs ... :)

I have added commits to take us up to kernel 4.12. And I can confirm that it compiles 100% clean on 4.10, 4.11 and 4.12. I more-or-less used other peoples' commits from 5.1.5 for the changes to 5.2.9, after carefully reviewing the code.

I changed the @zebulon2 4.11.9 commit, to make the driver follow the new ioctl regime, whereby the driver does not free the netdev itself in the destructor. See discussion with @MilhouseVH at https://github.com/zebulon2/rtl8812au/commit/55280fa97c1383ee53308fd7fc6400b2c6aaaa47#commitcomment-23088480

And I added a couple of changes for 4.12 to account for the new ioctl private function prototypes, to the commit for 4.12 by @kimocoder.

I think I need a nice cup of tea after all this testing :)

kimocoder, 2017-07-14:
Good job guys, will jump on it too when Im available. Portions may/seem to be available for porting to 8814au too. The driver i need for my "Alfa Networks AWUS1900" which currently got some issues similar to 8812au at current version. Again, great start over there!

MilhouseVH, 2017-07-14:
Just an aside: It would be really great if you guys could somehow consolidate your efforts (perhaps group ownership of a repo for RTL8812AU?) rather than repeated forking of the same drivers, which eventually die once the owner loses interest. There's obviously a great deal of interest in keeping these drivers up to date, however it's really frustrating when the work stops in one repo only to starts again in another!

kimocoder, 2017-07-14:
Know that feeling from other projects, but @astsam is the maintainer and developer which added the monitor and injection support, so he should also be credited for his work with it. future will tell if the project will be abandoned, but I doubt that at the moment, because most brand new adapters seem to run both 8812 and 8814 drivers and if so happens, it will always be alive since the work he did would not disappear. Dont worry too much about that one & lets collaborate here :)

gordboy, 2017-07-17:
## Status report on new 5.2.9 driver, which only supports 8812au

Added patches for new kernels, as discussed above.

Added all known USB IDs from the popular 8812au projects, and sorted them into ID order, making new additions less haphazard.

Added dkms deb package for convenience. Provided robust dkms.conf, that does not need editing for every kernel version.

### Testers Welcome :)

Tested with 100% clean compile and no kernel errors or warnings thus far on 4.10, 4.11, 4.12 and 4.13-rc1.

MilhouseVH, 2017-07-17:
Great stuff.

We (LibreELEC) have been adding additional devices (see [patch](https://github.com/LibreELEC/LibreELEC.tv/blob/master/packages/linux-drivers/RTL8812AU/patches/RTL8812AU-01-add_new_cards.patch)) - two are already in your 5.2.9, but the Elecom device is not. Any thoughts if it can (or should) be added - we can drop the Elecom ID if it's not needed or compatible.

I'm going to try a build of LibreELEC with your 5.2.9 driver as  I might have [one user](https://forum.kodi.tv/showthread.php?tid=298461&pid=2616217#pid2616217) on the Kodi forum able to test it with latest 4.12.2 kernel. Based on the v5.1.5 driver from the paspro repo he's got a stable connection so long as AC is disabled and N over 5Ghz is forced ([details](https://forum.kodi.tv/showthread.php?tid=298461&pid=2616143#pid2616143). Fingers crossed 5.2.9 is an improvement and he can get a stable AC connection.

Hopefully the lack of rtl8814au support isn't going to be too much of an issue for us - unfortunately we've no visibility on how many LibreELEC users are using that hardware (hopefully: none).

MilhouseVH, 2017-07-17:
No issues building, which is a rare thing indeed! :)

I've [posted a build](https://forum.kodi.tv/showthread.php?tid=298461&pid=2618318#pid2618318) with your new driver so hopefully there will be some feedback soon. If there are any issues that need opening they'll be done on your repo, which is where this conversation should probably continue... not sure if you're able to keep an eye on the Kodi thread but it might be worth checking it for the next few days just in case! Even better if you're able to test LibreELEC (I have x86_64 builds too - let me know if one would be useful to you!) :)

gordboy, 2017-07-17:
@MilhouseVH I've never heard of the Elecom device anywhere, until now. And it is an 8821A not an 8812A by the looks of it. Someone has been playing around with the gnab repo here:

[http://l52secondary.blog.fc2.com/blog-entry-23.html](url)

I get good speeds with my TP-Link T4U with 5.2.9:

`[01:50 gordboy@Tower 1 ~]
$ ssh Laptop-5g`
`Last login: Mon Jul 17 01:04:17 2017 from 192.168.0.3`

`[02:39 gordboy@Laptop 1 ~]
$ iwconfig`
`wlxf4f26d1a0b21  IEEE 802.11AC  ESSID:"GordNet5"  Nickname:"<WIFI@REALTEK>"`
          `Mode:Managed  Frequency:5.5 GHz  Access Point: E4:F4:C6:9B:46:40`
          `Bit Rate:867 Mb/s   Sensitivity:0/0` 
         ` Retry:off   RTS thr:off   Fragment thr:off`
          `Power Management:off`
          `Link Quality=89/100  Signal level=47/100  Noise level=0/100`
          `Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0`
          `Tx excessive retries:0  Invalid misc:0   Missed beacon:0`

`[02:39 gordboy@Laptop 1 ~]
$ ifconfig wlxf4f26d1a0b21`
`wlxf4f26d1a0b21: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500`
        `inet 192.168.0.5  netmask 255.255.255.0  broadcast 192.168.0.255`
        `inet6 fe80::f6f2:6dff:fe1a:b21  prefixlen 64  scopeid 0x20<link>`
        `ether f4:f2:6d:1a:0b:21  txqueuelen 1000  (Ethernet)`
        `RX packets 884089  bytes 1327170647 (1.3 GB)`
        `RX errors 0  dropped 11260  overruns 0  frame 0`
        `TX packets 113473  bytes 25595851 (25.5 MB)`
        `TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0`

A few dropped packets, but not drastically bad.

I'll keep an eye on your thread. And yes, let's move this to my repo. Open an issue over there, and we can continue this further.

kimocoder, 2017-07-17:
Creds to @gordboy & @MilhouseVH (+ @kcdtv & few others) you sure did a good job here guys. Thanks a lot!

yimmy86, 2017-07-19:
I recently installed the driver on  my computer that's running kali linux, and it allows me to connect to connect to my WiFi, but I'm having an issue when attempting to set the device to monitor mode. When I enter the command, iwconfig wlan1 mode monitor, my system freezes, and I have to  hard reset. I've tried removing, and reinstalling, but I still get the same results.

gordboy, 2017-07-22:
@yimmy86 5.2.9 driver does not support monitor mode. And I have no plans to add support for monitor mode, nor to add any other chipset than 8812AU.

closed 2017-07-22 by gordboy

yangxuan8282, 2017-09-02:
I found some other ftp user/passwd info of ftp://ftp3.realtek.com.tw on a [website](http://freeaccount.biz/accounts/ftp3.realtek.com.tw)
Have not found rtl8814au yet, but found a account related to another chipset [RTL8188EUS](https://wikidevi.com/wiki/Special:Ask?title=Special%3AAsk&q=%5B%5BChip1+model::RTL8188EUS%5D%5D&po=%3FInterface%0D%0A%3FForm+factor=FF%0D%0A%3FInterface+connector+type=USB+conn.%0D%0A%3FFCC+ID%0D%0A%3FManuf%0D%0A%3FManuf+product+model=Manuf.+mdl%0D%0A%3FVendor+ID%0D%0A%3FDevice+ID%0D%0A%3FChip1+model%0D%0A%3FSupported+802dot11+protocols=PHY+modes%0D%0A%3FMIMO+config%0D%0A%3FOUI%0D%0A%3FEstimated+year+of+release=Est.+year&eq=yes&p%5Bformat%5D=broadtable&order%5B0%5D=ASC&sort_num=&order_num=ASC&p%5Blimit%5D=500&p%5Boffset%5D=&p%5Blink%5D=all&p%5Bsort%5D=&p%5Bheaders%5D=show&p%5Bmainlabel%5D=&p%5Bintro%5D=&p%5Boutro%5D=&p%5Bsearchlabel%5D=%E2%80%A6+further+results&p%5Bdefault%5D=&p%5Bclass%5D=sortable+wikitable+smwtable), some adapters like TP-LINK TL-WN725N v2 should suitable for it

login: RTL8188E
passwd: M6b3Er
it is valid, I can login and the latest driver is: RTL8188EUS_linux_v5.2.2.2_22644.20170615.zip, which is newer than the repo on github (lwfinger/rtl8188eu)

fariouche, 2017-12-05:
Just for information, I've started to port the 8821a chipset.
You can check my fork on github (yes another one)
This seems not functionnal yet (compilation is ok, driver loads ok and hostapd is able to communicate with the usb dongle.)
I'm still missing vht mode.
I don't know when I will have it ready but I will keep you posted

fariouche, 2017-12-10:
Hi
My fork is now fully supporting 8821a chipset.
I'm using it as an AP, so maybe it will not work in station mode.
It should automatically detect the chipset (the auto detection was removed I do not know why.)

jershell, 2018-01-08:
> I still observe the "ping burst" with version 5.29. as i observe it with 4.3.21 or 5.1.5
This bug how fixed?

traud, 2018-08-24:
Thank you guys for this discussion! Last week, Edimax posted version 5.3.4 (2018 June) for their RTL8811AU based USB adapters [Edimax EW-7811UTC](https://www.edimax.com/edimax/download/download/data/edimax/au/download/product/wireless_adapters/wireless_adapters_ac600_dual-band/ew-7811utc/) and [Edimax EW-7811UAC](https://www.edimax.com/edimax/download/download/data/edimax/au/download/product/wireless_adapters/wireless_adapters_ac600_dual-band/ew-7811uac/). Worked right away for me in Ubuntu 18.04 LTS with a device using the default USB IDs from Realtek.
`make`
`sudo make install`
restart, and turning off Secure Boot in my UEFI. By the way, the [Edimax EW-7611UCB](https://www.edimax.com/edimax/download/download/data/edimax/global/download/product/wireless_adapters/wireless_adapters_ac600_dual-band/ew-7611ucb/) was not updated yet and still gets version 5.2.6.3 (2018 March). Go figure!

Those are RTL8821AU-only. But after two changes in the Makefile
`CONFIG_RTL8812A = y`
`CONFIG_BT_COEXIST = n`
it compiled as well. However, I do not have any RTL8812AU hardware to double-check. Edimax had a RTL8812AU based adapter in the past: [Edimax EW-7822UAC](https://www.edimax.com/edimax/download/download/data/edimax/global/download/product/wireless_adapters/wireless_adapters_ac1200_dual-band/ew-7822uac/). For that, Edimax still offers just version 5.2.9.3 (2017 November). As we know, the FTP server offers 5.2.20.2 (2018 June) already.

Anyway, after looking at the differences between those two drivers, I wonder whether
a) those version numbers are fake or
b) the drivers for RTL8812AU and RTL8821AU diverged long ago.
@fariouche how did you combine RTL8812AU and RTL8821AU again?

kimocoder, 2018-08-24:
Interesting.. do you have a link for the driver v5.3.4 ?

traud, 2018-08-24:
When you click on one of the models in my post, you are forwarded to its support/downloads page, for example the [Edimax EW-7811UTC…](https://www.edimax.com/edimax/download/download/data/edimax/au/download/product/wireless_adapters/wireless_adapters_ac600_dual-band/ew-7811utc/)

fariouche, 2018-08-24:
As far as I understand the logic of logitech, they have a kind of generic template driver internally, that they update regularily, and most likely a kind of script to fork it at some point with only the selected chipsets...
So we end up with the current mess: older chipsets with older forked code base + patches
I believe the 8811 is a pretty new chipset.
Anyway, the versions of the different forks in realteck seems to not match the actual version of the trunk.
For example: the latest driver for 8188 is 5.2.2.4.... and its core is old, comparable to a 4.1.x!

For the merge, I've done it using 3 way diff from:
1 : The target driver, 5.2.20 from 8812au (was the newest I've seen so far)
2: the 5.1.9 from 8821au, the latest driver supporting both 8812 and 8821

Diff between 1 and 2 for 8821, and 2 for 8812
This way I can see if one modification is a realtek modification when they moved from 5.1.9 to 5.2.20 for the 8812, and then merge equivalent modifications for the 8821... pfew! long task

fariouche, 2018-08-24:
looking at the 5.3.4, this is indeed a newer code!
Even more exciting, the 8821 code is very close to the one we have done here, + some of the missing functions, especially the one that we were missing to disable the one antenna when scanning!
This is worth the creation of a new branch. I will then merge the 8814 again, but this time it will be simpler

kimocoder, 2018-08-24:
@fariouche how do you want med to do it? Should i fork the "clean" one from gordboys repo, without all our patches or simply copy the existing and take it from there?

fariouche, 2018-08-24:
The idea is to take the 5.3.4 as is for now, with no modifications, and then add our mods back

fariouche, 2018-08-24:
I believe that the 5.3.4 will work as is for 8812 and 8821.
Adding VHT and injection should not be too hard

kimocoder, 2018-08-24:
That's true. Getting right to it. Leaving a note here when done.

kimocoder, 2018-08-24:
@fariouche here you go. https://github.com/kimocoder/rtl8812au/tree/v5.3.4
you probably want to just delete your fork and re-fork it to get it.

kimocoder, 2018-08-24:
I messed up something, fixed both repos now. They are clean ones.

traud, 2018-08-25:
Puh, I hoped someone has found the pattern how those drivers are created. I am looking at that code just for 24 hours. I did a diff between the latest 8812AU, 8821AU, 8812AE, and the new Wave 2 enabled chipsets. From my point of view, the differences are huge, in the core itself already. On the first glance, its looks like having one template six year ago when 802.11ac started and then branch each chipset from that. By now, Realtek patched around each branch individually when the ‘customer’ was big enough (like an embedded device maker with an order of 100-thousands units) to get its complain through to the engineer coding this stuff. But this is just a wild guess.

One could merge the chipsets again, create one driver with one core. However, the question is, if my assumption is right, wouldn’t it make more sense to create individual drivers, one from each driver the community finds?

> will then merge the RTL8814AU again

I hope you have seen the [latest version for that…](https://github.com/diederikdehaas/rtl8814AU/issues/22)
What about the drivers for the latest Wave 2 enabled chipsets like the RTL8812BU
- [Edimax EW-7822UTC](https://www.edimax.com/edimax/download/download/data/edimax/global/download/product/wireless_adapters/wireless_adapters_ac1200_dual-band/ew-7822utc/) 5.2.4.3 November 2017
- [Edimax EW-7822ULC](https://www.edimax.com/edimax/download/download/data/edimax/global/download/product/wireless_adapters/wireless_adapters_ac1200_dual-band/ew-7822ulc/) 5.3.1 April 2018

and [RTL8811CU…](https://github.com/whitebatman2/rtl8821CU) (5.2.5.3 October 2017)