astsam/rtl8812au #53: Installing 5.1.5 with 4.12

werdjockey, 2017-07-18:
Hello, I am not a programmer and really appreciate your work to allow for injection.  I have updated my kernel to 4.12, 
uname -r 
4.12.0-041200-generic

results from make command:
root@kali:~/rtl8812au-5.1.5# make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.12.0-041200-generic/build M=/root/rtl8812au-5.1.5  modules
make[1]: Entering directory '/usr/src/linux-headers-4.12.0-041200-generic'
  CC [M]  /root/rtl8812au-5.1.5/os_dep/linux/ioctl_cfg80211.o
/root/rtl8812au-5.1.5/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_indicate_connect’:
/root/rtl8812au-5.1.5/os_dep/linux/ioctl_cfg80211.c:758:6: error: passing argument 2 of ‘cfg80211_roamed’ from incompatible pointer type [-Werror=incompatible-pointer-types]
    , notify_channel
      ^~~~~~~~~~~~~~
In file included from /root/rtl8812au-5.1.5/include/osdep_service_linux.h:91:0,
                 from /root/rtl8812au-5.1.5/include/osdep_service.h:41,
                 from /root/rtl8812au-5.1.5/include/drv_types.h:32,
                 from /root/rtl8812au-5.1.5/os_dep/linux/ioctl_cfg80211.c:22:
./include/net/cfg80211.h:5435:6: note: expected ‘struct cfg80211_roam_info *’ but argument is of type ‘struct ieee80211_channel *’
 void cfg80211_roamed(struct net_device *dev, struct cfg80211_roam_info *info,
      ^~~~~~~~~~~~~~~
/root/rtl8812au-5.1.5/os_dep/linux/ioctl_cfg80211.c:760:6: warning: passing argument 3 of ‘cfg80211_roamed’ makes integer from pointer without a cast [-Wint-conversion]
    , cur_network->network.MacAddress
      ^~~~~~~~~~~
In file included from /root/rtl8812au-5.1.5/include/osdep_service_linux.h:91:0,
                 from /root/rtl8812au-5.1.5/include/osdep_service.h:41,
                 from /root/rtl8812au-5.1.5/include/drv_types.h:32,
                 from /root/rtl8812au-5.1.5/os_dep/linux/ioctl_cfg80211.c:22:
./include/net/cfg80211.h:5435:6: note: expected ‘gfp_t {aka unsigned int}’ but argument is of type ‘unsigned char *’
 void cfg80211_roamed(struct net_device *dev, struct cfg80211_roam_info *info,
      ^~~~~~~~~~~~~~~
/root/rtl8812au-5.1.5/os_dep/linux/ioctl_cfg80211.c:756:3: error: too many arguments to function ‘cfg80211_roamed’
   cfg80211_roamed(padapter->pnetdev
   ^~~~~~~~~~~~~~~
In file included from /root/rtl8812au-5.1.5/include/osdep_service_linux.h:91:0,
                 from /root/rtl8812au-5.1.5/include/osdep_service.h:41,
                 from /root/rtl8812au-5.1.5/include/drv_types.h:32,
                 from /root/rtl8812au-5.1.5/os_dep/linux/ioctl_cfg80211.c:22:
./include/net/cfg80211.h:5435:6: note: declared here
 void cfg80211_roamed(struct net_device *dev, struct cfg80211_roam_info *info,
      ^~~~~~~~~~~~~~~
/root/rtl8812au-5.1.5/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_add_monitor_if’:
/root/rtl8812au-5.1.5/os_dep/linux/ioctl_cfg80211.c:3639:10: error: ‘struct net_device’ has no member named ‘destructor’; did you mean ‘priv_destructor’?
  mon_ndev->destructor = rtw_ndev_destructor;
          ^~
/root/rtl8812au-5.1.5/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_preinit_wiphy’:
/root/rtl8812au-5.1.5/os_dep/linux/ioctl_cfg80211.c:6333:18: error: ‘WIPHY_FLAG_SUPPORTS_SCHED_SCAN’ undeclared (first use in this function)
  wiphy->flags |= WIPHY_FLAG_SUPPORTS_SCHED_SCAN;
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/rtl8812au-5.1.5/os_dep/linux/ioctl_cfg80211.c:6333:18: note: each undeclared identifier is reported only once for each function it appears in
/root/rtl8812au-5.1.5/os_dep/linux/ioctl_cfg80211.c: At top level:
/root/rtl8812au-5.1.5/os_dep/linux/ioctl_cfg80211.c:6366:25: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .change_virtual_intf = cfg80211_rtw_change_iface,
                         ^~~~~~~~~~~~~~~~~~~~~~~~~
/root/rtl8812au-5.1.5/os_dep/linux/ioctl_cfg80211.c:6366:25: note: (near initialization for ‘rtw_cfg80211_ops.change_virtual_intf’)
/root/rtl8812au-5.1.5/os_dep/linux/ioctl_cfg80211.c:6389:22: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .add_virtual_intf = cfg80211_rtw_add_virtual_intf,
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/rtl8812au-5.1.5/os_dep/linux/ioctl_cfg80211.c:6389:22: note: (near initialization for ‘rtw_cfg80211_ops.add_virtual_intf’)
cc1: some warnings being treated as errors
scripts/Makefile.build:302: recipe for target '/root/rtl8812au-5.1.5/os_dep/linux/ioctl_cfg80211.o' failed
make[2]: *** [/root/rtl8812au-5.1.5/os_dep/linux/ioctl_cfg80211.o] Error 1
Makefile:1512: recipe for target '_module_/root/rtl8812au-5.1.5' failed
make[1]: *** [_module_/root/rtl8812au-5.1.5] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.12.0-041200-generic'
Makefile:1855: recipe for target 'modules' failed
make: *** [modules] Error 2

Once again thank you very much for your work!

kimocoder, 2017-07-18:
Hi there!
Kernel v4.12 is not aupported, but take a look [here](https://github.com/astsam/rtl8812au/issues/51#issuecomment-315654701)

werdjockey, 2017-07-18:
Thank you very much for your response. I have checked out gordboys repo and it compiles and installs fine, but it freezes my computer if I issue the 'iwconfig wlan0 mode monitor'. So I am currently stuck between being able to use 4.12 with no injection or 4.9 with injection. Any advice is welcomed.

kimocoder, 2017-07-18:
well, monitoring & injection support is not added yet, because @gordboy and @MillhouseVH did this work the past days. But just watch for changes the next days, or you may clone the repo for v5.1.5 and add the v4.12 patch found in v5.2.9 yourself (easy work ;) no problem sir

kimocoder, 2017-07-18:
Do you need help, just tell me and I'll do it for you btw

werdjockey, 2017-07-18:
haha thank you! Yes, I would love to learn how to patch it! I tried just
substituting out the ioctl_cfg80211.c file with the gordboy one, but that
was to no avail... I like to try to learn rather than have things done for
me, so even though I prefer that you do it so I know it's done right, I
would also like to know the steps so that I can help other people in the
future...Thank you again for being so cool about this!