astsam/rtl8812au #55: Edup 1621

NishaYume, 2017-07-22:
hello all! Recently buyed a **EDUP 1621** who comes with RTL8814AU. Im trying to compile the 5.1.5 drivers and fails. Im using Antergos Linux (Arch  Linux based)
  CC [M]  /home/x/Downloads/rtl8812au-5.1.5/os_dep/linux/ioctl_cfg80211.o
/home/x/Downloads/rtl8812au-5.1.5/os_dep/linux/ioctl_cfg80211.c: En la función ‘rtw_cfg80211_add_monitor_if’:
/home/x/Downloads/rtl8812au-5.1.5/os_dep/linux/ioctl_cfg80211.c:3639:12: error: ‘struct net_device’ has no member named ‘destructor’; did you mean ‘priv_destructor’?
  mon_ndev->destructor = rtw_ndev_destructor;
            ^~~~~~~~~~
            priv_destructor
make[2]: *** [scripts/Makefile.build:295: /home/x/Downloads/rtl8812au-5.1.5/os_dep/linux/ioctl_cfg80211.o] Error 1
make[1]: *** [Makefile:1492: _module_/home/x/Downloads/rtl8812au-5.1.5] Error 2
make[1]: se sale del directorio '/usr/lib/modules/4.11.9-1-ARCH/build'
make: *** [Makefile:1855: modules] Error 2

uname -r 4.11.9-1-ARCH

MilhouseVH, 2017-07-22:
You need to apply the fix in #50.

NishaYume, 2017-07-22:
going to Close the issue, the fix #50 works very nice

thanks for hard work !!!!

closed 2017-07-22 by NishaYume