astsam/rtl8812au #56: tx-power 12dBm and injection issue

shidevil, 2017-08-06:
Hi,

I am using a rtl8812au card, Running Kali 2017 06 22.  

> Linux 4.9.0-kali4-amd64 #1 SMP Debian 4.9.30-2kali1 (2017-06-22) x86_64 GNU/Linux

Bus 001 Device 007: ID 0bda:881a Realtek Semiconductor Corp. 

Managed to compiled the driver and it works perfectly except that tx-power and aireplay injection having issue. 

Tried to do a sudo iwconfig wlan0 txpower 30 does not work. Tried changing from 5ghz to 2.4ghz to do a aireplay is not working as well for injection. 

The Router and rtl8812au wireless card is just beside each other. 

Any idea what's the issue?

[image]
[image]

sale7x, 2017-08-10:
Same for me , stuck with 12dBm and cant change tx power to 30 dBm

kimocoder, 2017-08-10:
The v5.1.5 got the 12dBm issue, that i may confirm, but the older one seems to work fine :)

shidevil, 2017-08-10:
I tested the older one v4.3.21 and it seems to work fine for txpower. however injection is the same for 5.15 and 4.3.21. 

Anyone have similar experience?

kimocoder, 2017-08-10:
Confirmed, but injection works fine on my Alfa AWUS1900 on the 8814AU driver though, but both drivers got ping buffer issue.

shidevil, 2017-08-10:
Thanks for the input! Appreciate it. Will wait for someone with similar injection issue on 8812au like mine.

kimocoder, 2017-08-10:
This issue has been reported in the past, im on vecation without my computer but @kcdtv and several others have submitted/reported and discussed this issue allready :) see other issue reports, simply dont have the time to look it up at this time. No problem sir

shidevil, 2017-08-10:
@kimocoder yes i saw the reports in the past. However seems like @kcdtv managed to get full injection and some others too.

sale7x, 2017-08-10:
I saw other reports and tried to do the same . I tried every solutions here and other repo the result i managed to get few times with injections if i do one injection that is it i have to a lot of manuall reboot and remove driver and compile other to get another one time injection. 
However thanks to everyone works in those repo's and sorry for my english guys .

kimocoder, 2017-08-10:
This driver stacks (8812/8814) are fairly new, just give it some more time/work it will be sorted out. Your english is understandable, no problem 👍

sale7x, 2017-08-16:
So guys , after a lot of work and time i managed to get complete working adapter based on Branch V4.3.21 it is now stable and good . tested on both Kali 2017.1 and Wifislax64 1.1 .
i faced minor problems after i got adapter work and stable that some fixed and some not affecting hard so : 
1- AP's no show sometimes with Airodump-ng so i found a solution here that worked for me , https://github.com/astsam/rtl8812au/issues/41 by @jpmv27 thanks to him , so i compiled his fix in Wifislax with no problems and it worked for airodump and the whole aircrack suite ,  for kali i couldn't compile @jpmv27 because of  Python issue and in kali adapter works just fine except for Airodump sometimes not always for me i can live with that as long as my adapter works just fine .
2- Monitor Mode : i'm using method as it says in ReadMe file so it works fine in both Kali and Wifislax, but i'm annoyed because some tools putting Adapter in monitor mode automatically  when they do i got error mounting adapter to monitor mode , monitor mode manually works perfect for me but it is annoying that i have to put my adapter in monitor mode every time , for example i'm using  a script and sometimes i have to chose an option require managed  mode so the script couldn't i got an error so i have to open the terminal and do it manually and wait for the other option in the script that require monitor mode and do it again so i can have the script going on .

  i hope it is helpful for you guys what i wrote , and respect for every one spending time to share .
sorry guys about my language and my knowledge about linux i'm not pro so pardon me guys .

kimocoder, 2017-08-16:
The driver does not have "virtual interface" support, therefor in mosts scripts (like wifite) it must be set manually with

ifconfig <adapter> down
iwconfig <adapter> mode monitor
ifconfig <adapter> up

But thanks for sharing your thoughts & findings regarding the fix by @jpmv27 to the others (again).

wifiuk, 2017-09-27:
RTL8812AU , i am also getting the 12dBm issue

kimocoder, 2017-09-27:
yeah we all do I think, have been investigating the issue for some time, yesterday I went through lotas of commits/code to try patching with v4.3.21 branch patches that contained txpower related code, but with no luck, since patches for implementing txpower control was allready added.

The same patches applied in v4.3.21 branch is allready used in v5.1.5 branch, so it should work the same.. but there are few things I've noted

* wifite2 seems to be running on 20-23dBm when used (even if it says 12dBm in iwconfig)
   could be bug in wifite2 or iwconfig?

* after inspecting [this](https://github.com/astsam/rtl8812au/pull/16/commits/9c09c7f0d2444b1fecf0fb633748af685361fbca) there are some code that may be used/investigated further (it seems weird that some code for implementing txpower control is missing/could be added). Since all branches is a bit similar, besides added code on the driver side, pulling patches would be hard, since the lines usually don't match, but manually adding strings/code between branches would work.

* could the issue be on the driver part somewhere?

chrisvollo, 2018-04-22:
I am seeing this issue in 8192EU as well ... was there an resolution for this?

Jo900, 2018-10-10:
i have the latest version istalled with dpkg, tx power 18dBm.
Is possible to edit tx power now ?

nakkins, 2018-11-27:
Any news on changing txpower?