astsam/rtl8812au #57: Error 2 "target 'modules' failed"

armeclemes, 2017-08-08:
Hello, i'm on Kali linux "4.6.0-kali1-amd64"

I have an error when i do "make", i've this error : 
```
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.6.0-kali1-amd64/build M=/root/crack/pilote/4.3.21/rtl8812au  modules
make[1]: *** /lib/modules/4.6.0-kali1-amd64/build: No such file or directory.  Stop.
Makefile:1264: recipe for target 'modules' failed
make: *** [modules] Error 2
```
note : I've tested with "RTL8814=1" and with the 3 branch, it's the same result...

Merci d'avance :+1:

Tikani, 2017-08-10:
Do you have appropriate `linux-headers` installed?

kimocoder, 2017-08-10:
Missing kernel headers as @Tikani has pointed out

armeclemes, 2017-08-19:
Ok, I did not think about it thank's !

closed 2017-08-19 by armeclemes