astsam/rtl8812au #62: Error Trying To Install On 4.11

KKHAN1991, 2017-08-16:
Hi,

i had a few errors trying to "make" this module.

make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.11.0-kali1-amd64/build M=/root/Documents/rtl8812au  modules
make[1]: Entering directory '/usr/src/linux-headers-4.11.0-kali1-amd64'
  CC [M]  /root/Documents/rtl8812au/os_dep/osdep_service.o
/root/Documents/rtl8812au/os_dep/osdep_service.c: In function ‘rtw_change_ifname’:
/root/Documents/rtl8812au/os_dep/osdep_service.c:2174:6: error: implicit declaration of function ‘rtnl_is_locked’ [-Werror=implicit-function-declaration]
  if(!rtnl_is_locked())
      ^~~~~~~~~~~~~~
cc1: some warnings being treated as errors
/usr/src/linux-headers-4.11.0-kali1-common/scripts/Makefile.build:299: recipe for target '/root/Documents/rtl8812au/os_dep/osdep_service.o' failed
make[4]: *** [/root/Documents/rtl8812au/os_dep/osdep_service.o] Error 1
/usr/src/linux-headers-4.11.0-kali1-common/Makefile:1509: recipe for target '_module_/root/Documents/rtl8812au' failed
make[3]: *** [_module_/root/Documents/rtl8812au] Error 2
Makefile:152: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.11.0-kali1-amd64'
Makefile:1264: recipe for target 'modules' failed
make: *** [modules] Error 2

kimocoder, 2017-08-16:
Do you have the linux-headers package installed ?

KKHAN1991, 2017-08-16:
Yes i do.

KKHAN1991, 2017-08-17:
Maybe an issue with master branch, 

I tried sudo apt install realtek-rtl88xxau-dkms and it works this gets 4.3.21~ branch i believe

kimocoder, 2017-08-23:
Use https://github.com/aircrack-ng/rtl8812au instead. This repo is still missing kernel v4.11 & v4.12 support at the time.

KKHAN1991, 2017-08-23:
Your commit was just to fix the import? Which i solved by just doing sudo apt install realtek-rtl88xxau-dkms, or is there something else?

rawkahn, 2017-09-28:
@kimocoder - and how does one clone the 4.3.21 version of the repo you provided? It installs 5.1.5 on my kali 4.12, but I want txpower control. 12 dbm is no bueno. 

PS Sorry for the stray/noob question, new to git and linux & having a hard time finding proper instructions...

kimocoder, 2017-09-28:
git clone https://github.com/aircrack-ng/rtl8812au -b v4.3.21

rawkahn, 2017-09-28:
@kimocoder Thank you!

closed 2023-03-12 by kaykhan