astsam/rtl8812au #63: Stops Capturing after 2-3 mins. Kali 2017 Kernal 4.11

KKHAN1991, 2017-08-17:
Hi,

I install via sudo apt install realtek-rtl88xxau-dkms and it works on Kali 4.11 so my previous issue is solved, but i have a new one now.

When i start capturing. airodump-ng wlan0 ( after following the steps to put it into monitor mode). It captures for 2-3 minutes and then all data dissapears. With a message saying "interface wlan0 down         "

 CH 13 ][ Elapsed: 3 mins ][ 2017-08-17 09:31 ][ interface wlan0 down                                  
                                                                                                       
 BSSID              PWR  Beacons    #Data, #/s  CH  MB   ENC  CIPHER AUTH ESSID
                                                                                                       
                                                                                                       
 BSSID              STATION            PWR   Rate    Lost    Frames  Probe     

but i check iwconfig and 

eth0      no wireless extensions.

lo        no wireless extensions.

wlan0     IEEE 802.11  Mode:Monitor  Frequency:2.417 GHz  Tx-Power=18 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off

it is still up in monitor mode.

KKHAN1991, 2017-08-17:
And if i try to stop it and run airodump-ng again, nothing captures at all

shidevil, 2017-08-17:
Wrong place to talk about this. Should go to aircrack github and the driver you use isn't astsam driver.

KKHAN1991, 2017-08-17:
Are you sure its not astsam drivers? https://bugs.kali.org/view.php?id=3260. Look at the notes #~0006567 "we created a package realtek-rtl88xxau-dkms for these drivers."

kimocoder, 2017-08-17:
can confirm it is the astsam driver used in Kali.

shidevil, 2017-08-17:
@KKHAN1991 sorry! Apologize for not checking. Have not encountered any problem similar to yours though.

KKHAN1991, 2017-08-17:
I copied a vbox instance from my laptop (which works for more than 2-3 minutes) to my desktop computer. On my desktop im having the issue even with this copied vbox. So i don't think its this driver thats the issue.

Must be a problem with either my wifi adapter https://www.amazon.co.uk/EDUP-Wireless-AC600Mbps-External-Antennas/dp/B01CDUJHXK

OR the host machine, my laptop host was ubuntu and desktop is windows 10.

KKHAN1991, 2017-08-18:
Network card keeps going down even on my laptop, could it be my adapteR? Can anyone reccomend me an adapter that supports this driver?

kimocoder, 2017-08-18:
Alfa Network AWUS036ACH found [here](https://www.alfa.com.tw/products_show.php?pc=137&ps=225)

KKHAN1991, 2017-08-18:
Not able to find that one were i live :(

kimocoder, 2017-08-18:
Internett.. Ordering.. :p

KKHAN1991, 2017-08-19:
That one is quite expensive will have to pay shipping aswell and import from abroad.  Do you know if this TP-LINK Archer T4U AC1200 is compatible?

kimocoder, 2017-08-19:
Weird that you asked, pushed a patch for Archer T4U AC1200 support 5 hours ago, so the answer would be that is not officially supported now, but it can be built from source and be supported (",)

The patch is [here](https://github.com/astsam/rtl8812au/pull/64)
Howto build + more information on [here on Kali bug tracker](https://bugs.kali.org/view.php?id=4157) read the comment field

KKHAN1991, 2017-08-19:
That patch says AC1300, thats version 2 of T4U. Thanks tho

kimocoder, 2017-08-19:
Bwah sorry, my mistake reading AC1300 on your post, but anyway.. as long as the chipset is rtl8812au I may help you get it supported anyways, no problem

KKHAN1991, 2017-08-20:
May have found the solution to this problem, the Wifi Adapter is 2.0. On my VM settings i had it set to 1.0. After downloading VM extensions i was able to set it to 2.0 and now it captures longer than 2-3 mins.

But now i have another problem :). Its not capturing a handshake when i send deauth packets i get 0 or few AKS back

closed 2023-03-12 by kaykhan