astsam/rtl8812au #66: Awus1900 not working on kali or other linux distribution

alisidani, 2017-08-24:
Status keep connecting and ask for password  but doesn't connect

yesimxev, 2017-08-25:
Doesn't work for me either. Just updated yesterday's release using apt-get (dkms) - airodump-ng still not compatible at all (doesn't scan / capture / aireplay doesn't inject)

kimocoder, 2017-08-25:
airmon-ng is not supported because the missing "virtual interface" support. Monitor mode have to be set manually with:

airmon-ng check kill
ifconfig <adapter> down
iwconfig <adapter> mode monitor
ifconfig <adapter> up

besides that, Kali now builds from aircrack-ng's repo [here](https://github.com/aircrack-ng/rtl8812au) and not this anymore. anyway, please give feedback if the manual works or not :)

yesimxev, 2017-08-25:
Ah I see.
I can put into monitor mode as seen on the readme. But can't scan, capture and deauth etc.

alisidani, 2017-08-26:
I have the same problem man but i can only scan networks 
Cant connect-capture-deauth...
I contacted alfa networks and they told me: "that's not something we can support,goodluck"

kimocoder, 2017-08-27:
Problems seem to have been fixed. Please check the post in issue reports on
howto build the v5.1.5 branch.

Best regards

26. aug. 2017 9.45 a.m.

alisidani, 2017-08-27:
Man everytime i upgrade to 4.12 my VM crashes and stop respondingThe fix also works on kali 4.9?

kimocoder, 2017-08-27:
I've just updated to v5.1.5 branch to support kernel v4.12.

* Howto do fix this?
git clone https://github.com/aircrack-ng/rtl8812au -b v5.1.5
make
make RTL8814=1
make install
cp 8814au.ko /lib/modules/4.12.0-kali1-amd64/kernel/drivers/net/wireless/