astsam/rtl8812au #68: kali 2017.2 error driver v4.3.21

lapolis, 2017-09-22:
Hi everyone.

When I compile the driver I got this error:

`/root/Desktop/rtl8812au-4.3.21/os_dep/linux/ioctl_cfg80211.c:807:6: error: passing argument 2 of ‘cfg80211_roamed’ from incompatible pointer type [-Werror=incompatible-pointer-types]
    , notify_channel
      ^~~~~~~~~~~~~~
In file included from /root/Desktop/rtl8812au-4.3.21/include/osdep_service_linux.h:91:0,
                 from /root/Desktop/rtl8812au-4.3.21/include/osdep_service.h:41,
                 from /root/Desktop/rtl8812au-4.3.21/include/drv_types.h:32,
                 from /root/Desktop/rtl8812au-4.3.21/os_dep/linux/ioctl_cfg80211.c:22:
/usr/src/linux-headers-4.12.0-kali2-common/include/net/cfg80211.h:5435:6: note: expected ‘struct cfg80211_roam_info *’ but argument is of type ‘struct ieee80211_channel *’
 void cfg80211_roamed(struct net_device *dev, struct cfg80211_roam_info *info,
      ^~~~~~~~~~~~~~~
/root/Desktop/rtl8812au-4.3.21/os_dep/linux/ioctl_cfg80211.c:809:6: warning: passing argument 3 of ‘cfg80211_roamed’ makes integer from pointer without a cast [-Wint-conversion]
    , cur_network->network.MacAddress
      ^~~~~~~~~~~
In file included from /root/Desktop/rtl8812au-4.3.21/include/osdep_service_linux.h:91:0,
                 from /root/Desktop/rtl8812au-4.3.21/include/osdep_service.h:41,
                 from /root/Desktop/rtl8812au-4.3.21/include/drv_types.h:32,
                 from /root/Desktop/rtl8812au-4.3.21/os_dep/linux/ioctl_cfg80211.c:22:
/usr/src/linux-headers-4.12.0-kali2-common/include/net/cfg80211.h:5435:6: note: expected ‘gfp_t {aka unsigned int}’ but argument is of type ‘unsigned char *’
 void cfg80211_roamed(struct net_device *dev, struct cfg80211_roam_info *info,
      ^~~~~~~~~~~~~~~
/root/Desktop/rtl8812au-4.3.21/os_dep/linux/ioctl_cfg80211.c:805:3: error: too many arguments to function ‘cfg80211_roamed’
   cfg80211_roamed(padapter->pnetdev
   ^~~~~~~~~~~~~~~
In file included from /root/Desktop/rtl8812au-4.3.21/include/osdep_service_linux.h:91:0,
                 from /root/Desktop/rtl8812au-4.3.21/include/osdep_service.h:41,
                 from /root/Desktop/rtl8812au-4.3.21/include/drv_types.h:32,
                 from /root/Desktop/rtl8812au-4.3.21/os_dep/linux/ioctl_cfg80211.c:22:
/usr/src/linux-headers-4.12.0-kali2-common/include/net/cfg80211.h:5435:6: note: declared here
 void cfg80211_roamed(struct net_device *dev, struct cfg80211_roam_info *info,
      ^~~~~~~~~~~~~~~
/root/Desktop/rtl8812au-4.3.21/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_add_monitor_if’:
/root/Desktop/rtl8812au-4.3.21/os_dep/linux/ioctl_cfg80211.c:3962:10: error: ‘struct net_device’ has no member named ‘destructor’; did you mean ‘priv_destructor’?
  mon_ndev->destructor = rtw_ndev_destructor;
          ^~
/root/Desktop/rtl8812au-4.3.21/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_preinit_wiphy’:
/root/Desktop/rtl8812au-4.3.21/os_dep/linux/ioctl_cfg80211.c:6618:18: error: ‘WIPHY_FLAG_SUPPORTS_SCHED_SCAN’ undeclared (first use in this function)
  wiphy->flags |= WIPHY_FLAG_SUPPORTS_SCHED_SCAN;
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/Desktop/rtl8812au-4.3.21/os_dep/linux/ioctl_cfg80211.c:6618:18: note: each undeclared identifier is reported only once for each function it appears in
/root/Desktop/rtl8812au-4.3.21/os_dep/linux/ioctl_cfg80211.c: At top level:
/root/Desktop/rtl8812au-4.3.21/os_dep/linux/ioctl_cfg80211.c:6651:25: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .change_virtual_intf = cfg80211_rtw_change_iface,
                         ^~~~~~~~~~~~~~~~~~~~~~~~~
/root/Desktop/rtl8812au-4.3.21/os_dep/linux/ioctl_cfg80211.c:6651:25: note: (near initialization for ‘rtw_cfg80211_ops.change_virtual_intf’)
/root/Desktop/rtl8812au-4.3.21/os_dep/linux/ioctl_cfg80211.c:6676:22: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .add_virtual_intf = cfg80211_rtw_add_virtual_intf,
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/Desktop/rtl8812au-4.3.21/os_dep/linux/ioctl_cfg80211.c:6676:22: note: (near initialization for ‘rtw_cfg80211_ops.add_virtual_intf’)
cc1: some warnings being treated as errors
/usr/src/linux-headers-4.12.0-kali2-common/scripts/Makefile.build:307: recipe for target '/root/Desktop/rtl8812au-4.3.21/os_dep/linux/ioctl_cfg80211.o' failed
make[4]: *** [/root/Desktop/rtl8812au-4.3.21/os_dep/linux/ioctl_cfg80211.o] Error 1
/usr/src/linux-headers-4.12.0-kali2-common/Makefile:1532: recipe for target '_module_/root/Desktop/rtl8812au-4.3.21' failed
make[3]: *** [_module_/root/Desktop/rtl8812au-4.3.21] Error 2
Makefile:152: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.12.0-kali2-amd64'
Makefile:1717: recipe for target 'modules' failed
make: *** [modules] Error 2
`
Since the 2017.1 release we can just apt-get install the new driver, and they work well. I was just wondering to try the old driver in order to check if was possible to to modify txpower.

Is there any way to compile this driver on kali 2017.1 or 2017.2?

Thanks in advance!

kimocoder, 2017-09-22:
@lapolis kernel v4.12 support is missing, try https://github.com/aircrack-ng/rtl8812au (which is an astsam fork) that supports the new kernels

lapolis, 2017-09-22:
@kimocoder thak you! they are working properly!

closed 2017-09-22 by lapolis