astsam/rtl8812au #69: AWUS036ACH increasing TXPower doesn't increase speed or signal strength on ap

ShotTard, 2017-09-25:
[image]

OS: Kali

I'm running an AP with NetworkManager, and wanted to increase the signal strength.
But increasing TXPower from 15 to 30 then 32 has made no difference.

Am I missing something?

Thanks

rawkahn, 2017-09-28:
Same here with v4.3.21 and same AWUS036ACH model.

ghostindex, 2017-10-28:
Same here. v4.3.21, AWUS036ACH, no signal strength boost.

adrianslabu, 2017-11-22:
How you changed txpower?
For me it will change only the Frequency.
[image]