astsam/rtl8812au #70: AWUS036ACH no longer seen after recent update

Jinovas, 2017-10-01:
This driver was working perfectly before (4.3.21 branch) a few weeks ago when I last used it. However after a few system updates the past several days, something has changed to where the device is no longer seen by iwconfig/ifconfig/iw list. However I can see it plugged in via "lsusb". There seems to be a couple others with the same or similar issue in the kali forums which I also posted but still waiting to see if anyone knows anything more.
[https://forums.kali.org/showthread.php?37447-Problem-with-wireless-internet-and-ALFA-AWUS036ACH-in-Kali-Linux-4-9-0](url)

Hoping maybe anyone else has ran into this before and has some advice how to resolve? =/ 

running kali version: 4.12.0-kali2-amd64

Jinovas, 2017-10-01:
Looks like after trying to recompile the kernel I'm getting a bunch of errors. Strange. Trying to work them out =/

kimocoder, 2017-10-01:
Kali changed repo to https://github.com/aircrack-ng/rtl8812au because of missing kernel v411/4.12 support here.. Try building v4.3.21 from there

Jinovas, 2017-10-01:
Oh snap. Will do! Thanks I think you may have saved me a lit of time and headache lol I'll give it a shot!

Jinovas, 2017-10-03:
@kimocoder that worked great and easily. Ty!

closed 2017-10-03 by Jinovas