astsam/rtl8812au #72: Channel lock does not work on Ubuntu 16.04 using Alfa AWUS1900

aghasi, 2017-10-09:
I compiled this using `make RTL8814=1` and then loaded the module by `insmod 8812au.ko`.  Changing mode is successfully and i can receive frames, but channel lock does not work. I'm using `iw wlx00c0ca8fdf52 set channel 6`. No error is shown, but i'm reciecing Beacon frames from channel 44,36,165,13,11 , etc.

yesimxev, 2017-10-21:
Having the same issue on Kali too. Can't run any wireless pentesting as channel keeps changing

kimocoder, 2017-10-21:
Kali now builds from [here](https://github.com/aircrack-ng/rtl8812au) btw..

yesimxev, 2017-10-21:
Ah sorry just found that issue channel lock thing. Posted straightaway to
have a record of happening on kali too. Happy that I think now I know why
mdk3 stops after a few mins or even aircrack injection didn't want to start
(i guess). Because the channel keeps changing. But I have a noob question.
If I install with apt-get install realtek-rtl88xxau-dkms , is it the latest
(since the date 20170828 which it shows when installing)? So is there a
newer version since? Or github one has some of the issues repaired but
that's not in that dkms yet? Not sure how to ask

kimocoder, 2017-10-21:
We did lots of work the last 24 hours that has not been pushed to Kali yet.
Build it manually from [https://github.com/aircrack-ng/rtl8812/](https://github.com/aircrack-ng/rtl8812au)

Be sure to use the v5.1.5 branch using "git clone https://github.com/aircrack-ng/rtl8812au -b v5.1.5"

yesimxev, 2017-10-22:
Got it, can see the 12dbm issue is solved, but still hops channels around

kimocoder, 2017-10-23:
Please open an issue reports in correct repo.