astsam/rtl8812au #73: Problem with installation

Bluewild93, 2017-10-09:
I updated my Kali linux to 4.13.0. And i can't install the driver. Hot to fix the problem? Can you make a driver that we can install in kali 4.13.0?

kimocoder, 2017-10-09:
@Bluewild93 use [this](https://github.com/aircrack-ng/rtl8812au/tree/v4.3.21) which got the kernel support

Bluewild93, 2017-10-09:
Thank you. But
I installed your driver, it works perfect for the first time when i put it in the monitor mode then sent DeAuth to my target devices. My target devices was disconnect from my wifi. But i couldn't get any handshakes. The second time, i couldn't get anything from the airodump-ng 
[image]

alsetema, 2017-10-11:
Same here, cannot compile on Lububtu 17.10, running 4.13 (tried with the versions 4.3.21 of the driver and version 5 as well. The only one that seems to compile is the one from aircrack-ng, which doesnt seem to support TXpower, which i really need.

kimocoder, 2017-10-11:
Use the link provided above, it supports newer kernels including kernel v4.13

alsetema, 2017-10-11:
@kimocoder Correct, that is the one that works for me, yet TXpower is not working unfortunately

kimocoder, 2017-10-11:
be sure to use the v4.3.21 branch, the v5.1.5 seems a bit dirty

alsetema, 2017-10-11:
@kimocoder I am, i do think there is an issue open about it.

ajayv707, 2018-04-14:
same problem in monitor mode my device not showing