astsam/rtl8812au #76: Cant Compile on linux 11_11_2017

123maxmix123, 2017-11-11:
Cant compile main branch, and also other branches have similar errors.
My wifi device: Edup ac1621 (chip: rtl8814au)

`make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.14.0-1-MANJARO/build M=/home/max/Downloads/rtl8812au-master  modules
make[1]: Entering directory '/usr/lib/modules/4.14.0-1-MANJARO/build'
  CC [M]  /home/max/Downloads/rtl8812au-master/core/rtw_cmd.o
In file included from /home/max/Downloads/rtl8812au-master/include/drv_types.h:32:0,
                 from /home/max/Downloads/rtl8812au-master/core/rtw_cmd.c:22:
/home/max/Downloads/rtl8812au-master/include/osdep_service.h: In function ‘thread_enter’:
/home/max/Downloads/rtl8812au-master/include/osdep_service.h:343:2: error: implicit declaration of function ‘allow_signal’; did you mean ‘do_signal’? [-Werror=implicit-function-declaration]
  allow_signal(SIGTERM);
  ^~~~~~~~~~~~
  do_signal
/home/max/Downloads/rtl8812au-master/include/osdep_service.h: In function ‘flush_signals_thread’:
/home/max/Downloads/rtl8812au-master/include/osdep_service.h:353:6: error: implicit declaration of function ‘signal_pending’; did you mean ‘timer_pending’? [-Werror=implicit-function-declaration]
  if (signal_pending (current))
      ^~~~~~~~~~~~~~
      timer_pending
/home/max/Downloads/rtl8812au-master/include/osdep_service.h:355:3: error: implicit declaration of function ‘flush_signals’; did you mean ‘do_signal’? [-Werror=implicit-function-declaration]
   flush_signals(current);
   ^~~~~~~~~~~~~
   do_signal
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:314: /home/max/Downloads/rtl8812au-master/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1502: _module_/home/max/Downloads/rtl8812au-master] Error 2
make[1]: Leaving directory '/usr/lib/modules/4.14.0-1-MANJARO/build'
make: *** [Makefile:1264: modules] Error 2
`

any solution?
I tried other git drivers, and they compiles but dont work properly.
They detects network but cant connect.

juwilie, 2017-11-15:
Check out my pull request, where I fixed kernel dependencies. Works well with 4.13, and I think would work with your kernel.

kimocoder, 2017-11-15:
https://github.com/aircrack-ng/rtl8812au

123maxmix123, 2017-11-15:
Sorry, but its still seems to not work.
Before i had real system, but now i opted for windows just becouse of wifi
driver :(

I created an manjaro VM, updated, and installed kernel you said it would
work with: 4.13.12-1
But also with default kernel cant compile.

Here is my device, its relatively new on market i think.
https://wikidevi.com/wiki/EDUP_EP-AC1621

kimocoder, 2017-11-15:
Well, the "0bda:8813" have to be added into "os_dep/linux/usb__intf.c" og it's missing (the adapter is new)

kimocoder, 2017-11-15:
Made a patch for you to have it supported.

git clone https://github.com/aircrack-ng/rtl8812au -b v5.1.5
cd rtl8812au
wget https://github.com/evilphish/rtl8812au/pull/8/commits/3e70835ef0febe9005b38368a09253aeed1360ad.patch
patch -p1 < 3e70835ef0febe9005b38368a09253aeed1360ad.patch
make RTL8814=1 && make install RTL8814=1

123maxmix123, 2017-11-15:
Really thanks to taking into consideration my case.
Anyway, i tried all steps listed by you in previous mail, and it stucks
again on make RTL8814=1

I think i will need to wait some months or years to get work properly that
device, as i tried all github drivers.
Some of them i was able to compile, and maybe 2 or 1 of them even worked,
but not good enough as drivers on windows.

I also tried to use ndriwrapper, to install windows drivers, and tried to
install different versions, but no way.

For now, until first virus i'm comfortable on windows, at least if windows
is not a virus that i'talking about :)

kimocoder, 2017-11-15:
They work great here :)

123maxmix123, 2017-12-01:
Strange, but today i tried to redo all steps with latest cernel, and they worked.
Installation is successful, BUT driver itself dont work, it doesnt connects to wifi network.
Here i retailed some output of my wifi manager.
You can search for <warn>.

Personally i dont know what happens, but something goes wrong.