astsam/rtl8812au #78: AWUS036ACH MONITOR MODE.

Inyotef, 2017-11-22:
Its driver.
`apt install realtek-rtl88xxau-dkms`

**Please Help why dont have a wlan0mon..**
```
root@kali:~# iwconfig
wlan0     IEEE 802.11  ESSID:off/any  
          Mode:Managed  Access Point: Not-Associated   Tx-Power=12 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          
eth1      no wireless extensions.

lo        no wireless extensions.

eth0      no wireless extensions.

```

**Airmon-ng :** 

```
root@kali:~# airmon-ng check kill

Killing these processes:

   PID Name
   500 wpa_supplicant
   502 dhclient

root@kali:~# iwconfig wlan0 mode monitor
root@kali:~# airmon-ng start wlan0

PHY	Interface	Driver		Chipset

phy1	wlan0		8812au		Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac WLAN Adapter

		(mac80211 monitor mode already enabled for [phy1]wlan0 on [phy1]10)

```

**Airodump-ng wlan0mon**

```
root@kali:~# airodump-ng wlan0mon
Interface wlan0mon: 
ioctl(SIOCGIFINDEX) failed: No such device

```

kimocoder, 2017-11-22:
@Inyotef first of, Kali build from [here](https://github.com/aircrack-ng/rtl8812au), second you have missed something there.. try

airmon-ng check kill
ifconfig wlan0 down
iwconfig wlan0 mode monitor

That's enough.. it does not support virtual Interface very well.
And another thing.. your Interface is "wlan0" thats why "airodump-ng wlan0mon" won't work.

Inyotef, 2017-11-23:
I want wifi attack actually. for example i making reaver im using wlan0 but i getting an error. 0x003. bla bla or 0x004. i try wifite its okay everything but i wait 1 hour not ansver password. wifite making capture i dont understand. i dont want to chance my system :// sorry my english not very well.