astsam/rtl8812au #82: AWUS036ACH 2017.3 + upgrades correct fully functional driver

sheshes, 2018-01-08:
Guys any updates for the driver for Alfa AWUS036ACH to be installed on kali linux 2017.3 + upgrades.
I have tried all available dkms packages from all developers on git, but the adapter doesn't work properly. I am in desperate need of a driver which can:

1. Control full capabilities of the adapter (txpower, range). This is supposed to be a very powerful adapter but I only get very few networks, and the power is much less than the previous versions of the adapter.
2. USB3 proper support upto 900mA as the manufacturer states
3. Work natively with dual bands, without having to switch the frequency option to 5GHz from 2.4GHz as it now.

Please let me know. I am available to do ANY kind of testings any developer will want to fix the above issues.

Thanks

kimocoder, 2018-01-08:
We've done some work over [here](https://github.com/aircrack-ng/rtl8812au)
But lots still remains. Check the issue reports om the aircrack-ng repo om the USB3 power issue, very interesting issue.

hyp3ar, 2018-01-08:
@sheshes ..Please how did you install the AWU036ACH on the kali linux 3.0?
have ben having issues for days installing it..have gone through alot of online tut but no head way..please i would appreciate if you can show me step to step how you installed yours...when i plug on my windows 10 it works but when connected to kali it doesn't show at all...i attached some screen shot to let you see some outputs ..

Thanks

[image]
[image]
[image]
[image]
[image]
[image]
[image]