astsam/rtl8812au #83: Problem with Airmon-ng Monitor Mode kali Linux Issue

m-imran79, 2018-01-20:
Hello all,

I am new to learning Kali Linux and facing a small issue. I hope someone can guide me what I am doing wrong. I am using Kali 2017.03 Virtual Box Version, and Wifi Adaptor is ALFA AWUS036ACH. When I run following command I get a warning to run some another command first. I run this command and then re-run the original command, but it gives the same error.  : 

root@kali:~# airmon-ng start wlan0 

PHY	Interface	Driver		Chipset

phy0	wlan0		8812au		Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac WLAN Adapter
cat: /sys/class/ieee80211/phy0/device/net/wlan0mon/type: No such file or directory

Newly created monitor mode interface wlan0mon is *NOT* in monitor mode.
Removing non-monitor wlan0mon interface...

WARNING: unable to start monitor mode, please run "airmon-ng check kill"
root@kali:~# airmon-ng check kill

root@kali:~# airmon-ng start wlan0

PHY	Interface	Driver		Chipset

phy0	wlan0		8812au		Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac WLAN Adapter
cat: /sys/class/ieee80211/phy0/device/net/wlan0mon/type: No such file or directory

Newly created monitor mode interface wlan0mon is *NOT* in monitor mode.
Removing non-monitor wlan0mon interface...

WARNING: unable to start monitor mode, please run "airmon-ng check kill"

kimocoder, 2018-01-20:
driver lack virtual Interface support, do it manually with:

ifconfig wlan0 down
iwconfig wlan0 mode monitor

m-imran79, 2018-01-21:
Thank you for the reply and suggested commands. I am using which you suggested, but I like to know why airmon-ng command does not work. 

Secondly, as soon when I run airmon-ng, it disconnects the Wifi and ethernet connection. It shows all devices are connected but the internet does not work. I then have no option but to restart Kali and then Wifi & Ether net work.

closed 2018-02-07 by m-imran79

TheSmurterGuy, 2018-08-05:
The fix for your problem m-imran79 is that you have to restart the network manager using the following command: 
service network-manager restart

Anil077, 2020-07-20:
hello everyone,

anyone help me please

i bought a alfa ac1900 wifi adapter in past November it worked fine up-to February later it giving troubles now from below

when i try airmon-ng it gives error like this

`Newly created monitor mode interface wlan0mon is *NOT* in monitor mode.
Removing non-monitor wlan0mon interface...

WARNING: unable to start monitor mode, please run "airmon-ng check kill"`

and later i tried
 `ifconfig wlan0 down
iwconfig wlan0 mode monitor`

and i am getting this error
root@kali:~# iwconfig wlan0 mode monitor
Error for wireless request "Set Mode" (8B06) :
    SET failed on device wlan0 ; Operation not permitted.

Ludrique, 2020-08-09:
hi i am facing the same problem, when i do it manually it disconnects my wif and when i restart the network manager it goes back to managed mode

Snow-tear, 2020-08-17:
> hello everyone,
> 
> anyone help me please
> 
> i bought a alfa ac1900 wifi adapter in past November it worked fine up-to February later it giving troubles now from below
> 
> when i try airmon-ng it gives error like this
> 
> `Newly created monitor mode interface wlan0mon is _NOT_ in monitor mode.
> Removing non-monitor wlan0mon interface...
> Segmentation fault
> 
> WARNING: unable to start monitor mode, please run "airmon-ng check kill"`
> 
> and later i tried
> `ifconfig wlan0 down`
> `iwconfig wlan0 mode monitor`
> 
> and i am getting this error
> root@kali:~# iwconfig wlan0 mode monitor
> Error for wireless request "Set Mode" (8B06) :
> SET failed on device wlan0 ; Operation not permitted.

Hello, I'm facing the same problem too. Have you got the solution?