astsam/rtl8812au #86: Problem Installing realtek-rtl88xxau-5.1.5~20180108

wynscott09, 2018-02-01:
Hi, I'm new to Kali I'm trying to install the driver #realtek-rtl88xxau-5.1.5~20180108  but I get the error log shown below:

ERROR:
root@Kali:~/Desktop/realtek-rtl88xxau-5.1.5~20180108# make
make -j 4 ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.13.0-kali1-amd64/build M=/root/Desktop/realtek-rtl88xxau-5.1.5~20180108  modules
make[1]: *** /lib/modules/4.13.0-kali1-amd64/build: No such file or directory.  Stop.
Makefile:1871: recipe for target 'modules' failed
make: *** [modules] Error 2
root@Kali:~/Desktop/realtek-rtl88xxau-5.1.5~20180108# 

I just followed the instruction given by kali, but nothing happen after i enter the command 

LINK : https://www.kali.org/news/kali-linux-20171-release/
COMMAND
apt-get update
apt install realtek-rtl88xxau-dkms

requesting for assistance , thanks

kimocoder, 2018-02-01:
you are missing the correct "linux-headers" package. you will need to install to one for kernel v4.13 before building the driver.

wynscott09, 2018-02-01:
@kimocoder  Hi, 
requesting for assistance on how to install "linux-headers" package kernel v4.13 
I'm using kali linux 2017.3

thanks in advance.

wynscott09, 2018-02-02:
also checked the folder /lib/modules/4.13.0-kali1-amd64/ prompted on terminal. (see attachment)

[image]

closed 2020-03-31 by wynscott09