astsam/rtl8812au #90: Build errors on kali 2018.1

sebdooris, 2018-04-21:
# uname -a
Linux error 4.15.0-kali2-amd64 #1 SMP Debian 4.15.11-1kali1 (2018-03-21) x86_64 GNU/Linux

receiving the following build errors.

/root/SHD/WIFI/rtl8812au/include/osdep_service_linux.h:277:8: error: ‘_timer {aka struct timer_list}’ has no member named ‘data’
  ptimer->data = (unsigned long)cntx;
        ^~
/root/SHD/WIFI/rtl8812au/include/osdep_service_linux.h:278:2: error: implicit declaration of function ‘init_timer’; did you mean ‘_init_timer’? [-Werror=implicit-function-declaration]
  init_timer(ptimer);
  ^~~~~~~~~~
  _init_timer

Full make log below

# make DEBUG=1
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.15.0-kali2-amd64/build M=/root/SHD/WIFI/rtl8812au  modules
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-kali2-amd64'
  CC [M]  /root/SHD/WIFI/rtl8812au/core/rtw_cmd.o
In file included from /root/SHD/WIFI/rtl8812au/include/osdep_service.h:41:0,
                 from /root/SHD/WIFI/rtl8812au/include/drv_types.h:32,
                 from /root/SHD/WIFI/rtl8812au/core/rtw_cmd.c:22:
/root/SHD/WIFI/rtl8812au/include/osdep_service_linux.h: In function ‘_init_timer’:
/root/SHD/WIFI/rtl8812au/include/osdep_service_linux.h:277:8: error: ‘_timer {aka struct timer_list}’ has no member named ‘data’
  ptimer->data = (unsigned long)cntx;
        ^~
/root/SHD/WIFI/rtl8812au/include/osdep_service_linux.h:278:2: error: implicit declaration of function ‘init_timer’; did you mean ‘_init_timer’? [-Werror=implicit-function-declaration]
  init_timer(ptimer);
  ^~~~~~~~~~
  _init_timer
cc1: some warnings being treated as errors
/usr/src/linux-headers-4.15.0-kali2-common/scripts/Makefile.build:329: recipe for target '/root/SHD/WIFI/rtl8812au/core/rtw_cmd.o' failed
make[4]: *** [/root/SHD/WIFI/rtl8812au/core/rtw_cmd.o] Error 1
/usr/src/linux-headers-4.15.0-kali2-common/Makefile:1528: recipe for target '_module_/root/SHD/WIFI/rtl8812au' failed
make[3]: *** [_module_/root/SHD/WIFI/rtl8812au] Error 2
Makefile:146: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-kali2-amd64'
Makefile:1717: recipe for target 'modules' failed
make: *** [modules] Error 2

danieleperera, 2018-04-23:
i'm getting the same error !!! hope they fix it

kimocoder, 2018-04-23:
It's due to timers api changed in kernel v4.15. anyway, we've got a working fork over [here](https://github.com/aircrack-ng/rtl8812au)