astsam/rtl8812au #92: raspberry pi 3 + monitor mode isn't available.

danieleperera, 2018-05-09:
I'm a newbie to linux. I'm trying to install AWUS036ACH to my Raspberry pi 4.14.34-v7+ 
i run these commands 

$ sudo apt update
$ sudo apt upgrade
$ sudo apt --purge autoremove
$ sudo reboot
$ sudo apt install raspberrypi-kernel-headers
$ sudo wget https://raw.githubusercontent.com/notro/rpi-source/master/rpi-source -O /usr/bin/rpi-source && sudo chmod +x /usr/bin/rpi-source && /usr/bin/rpi-source -q --tag-update
$ sudo apt-get install bc git -y
$ git clone https://github.com/astsam/rtl8812au
$ cd rtl8812au
$ sed -i -e 's/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile
$ sed -i -e 's/CONFIG_PLATFORM_ARM_RPI = n/CONFIG_PLATFORM_ARM_RPI = y/g' Makefile
$ make
$ sudo insmod 8812au.ko
$ sudo cp 8812au.ko /lib/modules/$(uname -r)/kernel/drivers/net/wireless
$ sudo depmod
$ sudo reboot

when i run iwconfig 

wlan0     unassociated  Nickname:"<WIFI@REALTEK>"
              Mode:Auto  Frequency=2.412 GHz  Access Point: Not-Associated
              Sensitivity:0/0
              Retry:off   RTS thr:off   Fragment thr:off
              Encryption key:off
              Power Management:off
              Link Quality:0  Signal level:0  Noise level:0
              Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
              Tx excessive retries:0  Invalid misc:0   Missed beacon:0

when i give commands 

$ ip link set wlan0 down
$ iwconfig wlan0 mode monitor //doesn't give me any errors
$ ip link set wlan0 up 

i checked if i could do packet injections with 

$ aireplay-ng -9 wlan0
it isn't successful. Have you got any suggestions?  Thanks

MrDrache333, 2018-08-09:
Exactly the Same here :(

danieleperera, 2018-08-13:
Try this:

$ sudo apt-get update
$ sudo apt-get dist-upgrade
$ sudo apt — purge autoremove
$ sudo reboot
$ sudo apt install raspberrypi-kernel-headers
$ ls /lib/modules/$(uname -r)
$ sudo apt-get install build-essential bc git dkms airmon-ng -y
$ git clone https://github.com/aircrack-ng/rtl8812au/ — branch v5.1.5
$ cd rtl88*
$ sed -i -e ‘s/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g’ Makefile
$ sed -i -e ‘s/CONFIG_PLATFORM_ARM_RPI = n/CONFIG_PLATFORM_ARM_RPI = y/g’ Makefile
$ sudo ./dkms-install.sh
$ sudo reboot

i wrote a blog post on this but its in italian if you want you can check it out @MrDrache333  [here](https://medium.com/@danieleperera/hai-detto-hacking-pi-per-caso-1abb1a3287c1)