astsam/rtl8812au #94: macchanger -r wlan0 error

danieleperera, 2018-05-16:
I'm using alfa network awus036ach and is working well but i want to change my mac address and when i give this command it gives me an error.

sudo macchanger -r wlan1

Current MAC:   xx:xx:xx:xx:xx:xx (ALFA, INC.)
Permanent MAC: xx:xx:xx:xx:xx:xx (ALFA, INC.)
[ERROR] Could not change MAC: interface up or insufficient permissions: Operation not permitted

jeffro87, 2018-06-25:
a little late, try this
iw wlan1 down
macchanger -r wlan1
iw wlan1 up

qkum, 2019-07-04:
Possible fixes:
1#It pops up  if the interface (wlan0) is already in use in another window. - stop the other program/process using the wireless interface.
2# try with "sudo" in front - as I now see you tried.
3# do as the comment above said, take down the interface and change the mac address manually:
ifconfig wlan1 down
macchanger --random --bia wlan1 
ifconfig wlan1 up

Explanation:
--random = random maccaddress,   
--bia = pretends to be a real burned in mac address.