astsam/rtl8812au #96: Can't enable monitor mode on the external wifi adapter

spaul13, 2018-08-19:
I installed the driver for rtl8812au and using lsmod I can see that driver also. I also created a monitor mode interface using airmon-ng but somehow I can't enable the monitor mode operation for the external wifi usb adapter (ALFA AC1200 USB adapter).

Can anyone please tell me why? What I am missing?

astsam, 2018-08-20:
Do not use the airmon-ng to enable the monitor mode. See README.md for instructions.

spaul13, 2018-08-20:
I have also used the instructions given in the README. But Nothing works.

astsam, 2018-08-24:
Will you provide a little more information about the issue, at list output of the commands (replace wlan0 with the corresponding interface name):
$ sudo iwconfig wlan0 mode monitor
$ sudo iw wlan0 info
and
dmesg output, related to the driver